Koenkk/zigbee2mqtt #6425: Ikea devices always reporting 60% battery

bummzulu, 2021-02-23:
# Bug Report
## What happened
I have 5 IKEA Tradfri E1525/E1745 presence sensors installed. This has been working fine now for many months. I noticed that all devices constantly report "battery: 60" although for 2 of them the battery has already been completely empty.

## What did you expect to happen
Battery information reflecting the actual state so I can replace the battery before the system stops working.

## How to reproduce it (minimal and precise)
Reading the battery information of IKEA Tradfri E1525/E1745

## Debug Info
Zigbee2MQTT version: 1.17.1
Adapter hardware: CC2652R
Adapter firmware version: 20200805

Is anybody else seeing similar issues?

m0wlheld, 2021-02-23:
> Is anybody else seeing similar issues?

I have 3, 1 month old. They are currently reporting 60%, but they have been at 37% when it was cold. 

I don‘t think it‘s a reliable value.

sjorge, 2021-02-23:
Mine fluctuate also, one of mine was < 10% and for a week but still worked and now its at 37% again. It's probably based on the battery's voltage which can fluctuate with temperature and such. I don't have an official gateway to compare if it also fluctuates there.

bummzulu, 2021-02-23:
Thanks for your inputs. Mine always stays at 60 and it never fluctuated so far. What adapters do you use?

sjorge, 2021-02-23:
I'm using a prototype stick that has a CC2652P on it.

m0wlheld, 2021-02-23:
Raspee II

pavax, 2021-02-23:
Hi there, I just started using Z2M a couple of days ago. 

[image]

I did not yet replace a cr2032-battery yet, but I did charge the lithium battery of the KADRILJ roller blind (was < 20%) and it did go up to 100% and is now at 99%

Specs:
Zigbee2MQTT version: 1.17.1
Adapter hardware: Conbee II
Integration: Home Assistant Addon
Adapter firmware version: 0x26680700

closed 2021-04-02 by github-actions[bot]