Koenkk/zigbee2mqtt #6428: Neo NAS-AB02B0 (alarm/siren) missing power indication

70MM13, 2021-02-23:
## What happened
While fooling around with my new device I think I found a missing feature.
It appears this device is very similar to the [Wifi version](https://templates.blakadder.com/neo_coolcam_NAS-AB02W.html)
Not only has that page a nice list of the alarm sounds, I also noticed the "power type" status:
- dpId 101 - power type (1 - battery high; 2 - medium; 3 - low; 4 - usb port)

In my debug log I have correct reportings of that id:
```
Received Zigbee message from 'NEO_siren_alarm', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":101,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID null
Received Zigbee message from 'NEO_siren_alarm', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[4],"type":"Buffer"},"datatype":4,"dp":101,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID null
```
and can confirm the data:0 is on battery and data:4 is on usb power.

So my request is, can we have a reporting on this?

Koenkk, 2021-02-24:
You can check if it work by replacing https://github.com/Koenkk/zigbee-herdsman-converters/blob/9162491ef881debca91e365cebe2ed61b2f63d61/converters/fromZigbee.js#L1859
 with 

```js
            case tuya.dataPoints.neoUnknown1: // 0x0465 [4]
                const lookup = {1: 'battery_high', 2: 'battery_medium', 3: 'battery_low', 4: 'usb'}
                return {power_type: lookup[value]};
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

70MM13, 2021-02-24:
Thanks, tried it and works almost correct.
_4_ for usb is ok but in contrary to the give values I get _0_ when on battery power. So it's not clear if _1,2,3_ are used or maybe not.
I now have 
```
const lookup = {0: 'battery', 1: 'battery_high', 2: 'battery_medium', 3: 'battery_low', 4: 'usb'}
```
and am looking for a way to simulate low battery or something :)

Koenkk, 2021-02-24:
I'm fine with it, does it work correctly? Then I can add it

70MM13, 2021-02-24:
yes, thank you

Koenkk, 2021-02-24:
Added, also added `battery_low` `true`/`false`.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-24 by Koenkk

70MM13, 2021-02-25:
> Added, also added `battery_low` `true`/`false`.
> 
> Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Nice, thanks a lot