Koenkk/zigbee2mqtt #6432: Moes Smart light switch module+RF (1 gang)

asiersan, 2021-02-24:
Itried this: 
https://github.com/Koenkk/zigbee2mqtt/issues/5202
with no succees or my error.

I tried dev branch too

## Information about the device + link
https://es.aliexpress.com/item/1005001830062515.html?spm=a2g0s.9042311.0.0.24aa63c0zpXp97

## data/database.db entry of the device
{"id":10,"type":"Router","ieeeAddr":"0x60a423fffed2d05a","nwkAddr":52849,"manufId":4098,"manufName":"_TZ3000_bkfe0bab","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_bkfe0bab","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1614119397725}

Koenkk, 2021-02-24:
Added.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-24 by Koenkk

asiersan, 2021-02-24:
Many thanks, is ok.