Koenkk/zigbee2mqtt #6433: Device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_mvn6jl7x'

trannamhung, 2021-02-24:
Hi,

I have this log in mqtt add-on. Could you support my device? Thank you.

Zigbee2MQTT:warn  2021-02-24 17:55:22: Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_mvn6jl7x

db file.
{"id":7,"type":"Router","ieeeAddr":"0xbc33acfffee93a85","nwkAddr":8028,"manufId":4098,"manufName":"_TZ3000_mvn6jl7x","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_mvn6jl7x","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001fc64bfe","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"2":{"profId":260,"epId":2,"devId":9,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001fc64bfe","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1614157992578}

[image]
](url)

Koenkk, 2021-02-24:
Can you provide a link to this device? (aliexpress?)

trannamhung, 2021-02-25:
Sorry, I purchase it from my local seller, on Facebook. Could you inquiry other information? Hope that I will have...

trannamhung, 2021-02-25:
That's Tuya platform.

https://www.lazada.vn/products/o-cam-am-tuong-thong-minh-tuya-zigbee-i865266713-s2458248426.html?spm=a2o4n.searchlist.list.4.16d11c5dG2hjoo&search=1

Koenkk, 2021-02-25:
Added.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-25 by Koenkk