Koenkk/zigbee2mqtt #6437: C2652RB (Slaesh) not working with Ser2Net

ludamus, 2021-02-24:
Hi all,

I hope you can help as im at the end of my tether.

**Background:**
I have recently bought [Slaesh CC2652 Dongle](https://slae.sh/projects/cc2652/)  moving away from an older CC2531 dongle which i flashed myself and all has been working quite well (even managed to bodge a small aerial on..).

The dongle was bought pre-flashed with Koenkk coordinator software, so hoping to just plug and pray, I was wrong.  I've spent many hours googling around and lurking on forums with people that have had similar issues but none of them with a resolution, the closest I got was someone saying there was an issue with DTR/RTS of which se2net isn't releasing (and apparently no way to stop it).

I should say, that in desperation i installed a docker instance of zigbee2mqtt on the Pi presenting the local /ttyUSB0 to the instance, do this locally was able to connect to the serial interface, so its defiantly something with Ser2Net related.  (what stopped me was MQTT not reaching the VM / Docker Mosquitto broker IP) I digress...

Is this a known issue ? is there an easy fix? I would like to carry on using Ser2Net (being simple) 

**Similar hit**
The following link is exactly the same issue that I'm getting. 
https://www.gitmemory.com/issue/Koenkk/zigbee2mqtt/4473/751516722
https://github.com/Koenkk/zigbee2mqtt/issues/4473

**Versions**
Ser2Net Version 3.5
Zigbee2MQTT version 1.17.1-dev

 ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Feb 24 14:23 usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_37_81-if00-port0 -> ../../ttyUSB0

154601.397286] usb 1-1.3: new full-speed USB device number 7 using xhci_hcd
[154601.542587] usb 1-1.3: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[154601.542597] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[154601.542605] usb 1-1.3: Product: slae.sh cc2652rb stick - slaesh's iot stuff
[154601.542613] usb 1-1.3: Manufacturer: Silicon Labs
[154601.542620] usb 1-1.3: SerialNumber: 00_12_4B_00_21_CC_37_81

Its seen !!

**Setup**
I have the stick plugged into a raspberry pi which is running Ser2net  and HA in a remote VM

I have tried with /dev/ttyUSB0 and also the full name as in the below config.

20109:raw:0:/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_37_81-if00-port0:115200 8DATABITS NONE 1STOPBIT

**Firmware:**

I recently updated the firmware to run [Slaesh version](http://slae.sh/downloads/znp_CC2652RB_20200715.bin) just in case there was an issue with the firmware that was loaded, same issue.

**Test Script**

Shows that the stick is recognized. 

 python3 ~root/test.py
Opening port /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_37_81-if00-port0, baud 115200

--> Bootloader is working!
--> ZigBee coordinator is working!
`---> transportrev 2, product 1, majorrel 2, minorrel 7, maintrel 1, revision: 20200715

**Zigbee2mqtt Logs**

Zigbee2MQTT:debug 2021-02-24 17:37:26: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2021-02-24 17:37:26: Logging to console and directory: '/config/zigbee2mqtt/log/2021-02-24.17-37-26' filename: log.txt
Zigbee2MQTT:debug 2021-02-24 17:37:26: Removing old log directory '/config/zigbee2mqtt/log/2021-02-24.17-37-12'
Zigbee2MQTT:info  2021-02-24 17:37:26: Starting Zigbee2MQTT version 1.17.1-dev (commit #d5b2486)
Zigbee2MQTT:info  2021-02-24 17:37:26: Starting zigbee-herdsman (0.13.71)
Zigbee2MQTT:debug 2021-02-24 17:37:26: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6758},"serialPort":{"path":"172.21.16.59:20109"}}'
Zigbee2MQTT:error 2021-02-24 17:37:27: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-02-24 17:37:27: Failed to start zigbee
Zigbee2MQTT:error 2021-02-24 17:37:27: Exiting...
Zigbee2MQTT:error 2021-02-24 17:37:27: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open 172.21.16.59:20109'
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)
    at Generator.next (<anonymous>)
    at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:134:49)
    at SerialPort._error (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.17.1-dev start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.17.1-dev start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-02-24T17_37_27_070Z-debug.log
[17:37:27] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.17.1-dev start /app
> node index.js

Hope this is enough information , 

regards

sjorge, 2021-02-24:
Yeah for anything that needs the bootloader rts dtr dance ser2net won’t work. 

On the linked issue is diff for ser2nodenet that makes it work. Tim also created a new repo that has the patch applied.

ludamus, 2021-02-24:
Hi, thanks for the quick response. 

do you know where i can get hold of ser2nodenet, and is it easy to install ? I failed miserable trying to get the latest Ser2Net working as had to compile... gensio issues galore... :(

ludamus, 2021-02-24:
Hi Jorge , found one of your threads, https://www.gitmemory.com/issue/Koenkk/zigbee2mqtt/4473/751504465

sounds like you had similar issues, not sure i understand what you did to get it to work though ;)

lordraiden, 2021-03-02:
I have a similar problem, it was working before and it stoped a few days ago. I haven changed anything but I have the dockers set on autoupdated every night. Any idea how can I fix it?

This is the error log

```
> zigbee2mqtt@1.18.0 start /app
> node index.js

[32mZigbee2MQTT:info [39m 2021-03-02 20:35:02: Logging to console and directory: '/app/data/log/2021-03-02.20-35-02' filename: log.txt
[32mZigbee2MQTT:info [39m 2021-03-02 20:35:02: Starting Zigbee2MQTT version 1.18.0 (commit #3fd6619)
[32mZigbee2MQTT:info [39m 2021-03-02 20:35:02: Starting zigbee-herdsman (0.13.72)
[31mZigbee2MQTT:error[39m 2021-03-02 20:35:22: Error while starting zigbee-herdsman
[31mZigbee2MQTT:error[39m 2021-03-02 20:35:22: Failed to start zigbee
[31mZigbee2MQTT:error[39m 2021-03-02 20:35:22: Exiting...
[31mZigbee2MQTT:error[39m 2021-03-02 20:35:22: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
at Generator.throw (<anonymous>)
at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.18.0 start: `node index.js`
npm ERR! Exit status 1

npm ERR! Failed at the zigbee2mqtt@1.18.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR! /root/.npm/_logs/2021-03-02T19_35_22_308Z-debug.log
```

This is usb
```root@Unraid:~# ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Mar  2 20:34 usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_46_CD-if00-port0 -> ../../ttyUSB0
```

And this is a portion of my configuration.yaml

```
homeassistant: true
permit_join: true
mqtt:
  discovery: true
  base_topic: zigbee2mqtt
  server: 'mqtt://10.10.10.101:1883'
  user: mosquitto
  password: contrasena.fuerte2021
  client_id: zigbee2mqtt
serial:
  port: /dev/ttyUSB0
```

zombielinux, 2021-03-03:
https://github.com/cminyard/ser2net/issues/46

Take a look at the latest version of ser2net. Should have a fix for our DTR/RTS problem.

ludamus, 2021-03-03:
@lordraiden , somethings odd.

In the end I gave up with Ser2Net and moved the Docker Ziggbee2Mqtt instance from my VM onto the Pi so that it had direct access to //dev/ttyUSB0, and with the addition of the web GUI ive been quite impressed and happy to leave it with this architecture. 

However.... I tinkered (_if ain't broke don't fix...._),   and decided to upgrade  to "**koenkk/zigbee2mqtt:latest-dev**" image (thinking that I needed this version for OTA updates to support upgrading Ikea stuff), like you had really odd issues connecting slaesh C2652RB dongle afterwards. 

What I noticed is that the blue light would just go of and continually cycle.

So, thinking this may be an issue I tried the latest (  **koenkk/zigbee2mqtt:latest** ) stable up to date version, this didn't fix it.

Rebooted the Pi, did a full upgrade of all components , kernel, Pi  firmware etc. Same issue.

"
Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
at Generator.throw (<anonymous>)
"

I even reflashed the C2652RB !!, no luck,

Out of exasperation, I downloaded the original version  (1.17.1) which I knew worked "**koenkk/zigbee2mqtt:1.17.1"** , guess what all sprang back into life.

Not sure what went on..

Lastly , does any one know what the LED lights mean ? a blue light vs a purple light (or perhaps red and blue) ...

Ludamus

sjorge, 2021-03-03:
> at ZStackAdapter. (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)

There was a follow up release of z2m, an upgrade of node-serialport broke some sticks.

~ sjorge

dtmf, 2021-03-08:
> [cminyard/ser2net#46](https://github.com/cminyard/ser2net/issues/46)
> 
> Take a look at the latest version of ser2net. Should have a fix for our DTR/RTS problem.

I can confirm the latest ser2net fixes the DTR/RTS problem.

I turned of DTR and RTS in the ser2net.yaml:

`
connection: &con01
  accepter: tcp,20108
  connector: serialdev,/dev/ttyUSB0,115200n81,local,dtr=off,rts=off
  options:
    kickolduser: true
`

zombielinux, 2021-03-08:
@dtmf 

Is your zigbee2mqtt dockerized? because this is the error I'm getting with the latest everything. 

```
Zigbee2MQTT:error 2021-03-08 16:16:46: Exiting...,
Zigbee2MQTT:error 2021-03-08 16:16:46: Failed to start zigbee,
Zigbee2MQTT:error 2021-03-08 16:16:46: Error while starting zigbee-herdsman,
Zigbee2MQTT:error 2021-03-08 16:16:46: Error: Error while opening socket,
    at Socket.emit (events.js:327:22),
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:185:28),
    at emitErrorNT (internal/streams/destroy.js:106:8),
    at emitErrorCloseNT (internal/streams/destroy.js:74:3),
    at processTicksAndRejections (internal/process/task_queues.js:80:21),
npm ERR! code ELIFECYCLE,
npm ERR! zigbee2mqtt@1.18.1 start: `node index.js`,
npm ERR! errno 1,
npm ERR! Exit status 1
```

dtmf, 2021-03-08:
I'm running the Zigbee2mqtt Edge (version 1.18.0-dev) installed from the Home Assistant GUI.
I think mine is dockerized, but I am not sure.
On the HA console I can see a bunch of docker stuff running including:
  root       32965  0.0  0.0    196     4 ?        S    18:41   0:00 s6-supervise zigbee2mqtt

Here is my log:
> node index.js
Zigbee2MQTT:info  2021-03-08 12:41:52: Logging to console and directory: '/share/zigbee2mqtt/log/2021-03-08.12-41-52' filename: log.txt
Zigbee2MQTT:info  2021-03-08 12:41:52: Starting Zigbee2MQTT version 1.18.0-dev (commit #b9c69f9)
Zigbee2MQTT:info  2021-03-08 12:41:52: Starting zigbee-herdsman (0.13.74)
Zigbee2MQTT:info  2021-03-08 12:42:03: zigbee-herdsman started
Zigbee2MQTT:info  2021-03-08 12:42:03: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2021-03-08 12:42:03: Currently 6 devices are joined:

zombielinux, 2021-03-08:
Hmmm, same error from the latest-dev version. Must be a permissions error somewhere?

dtmf, 2021-03-08:
In your zigbee2mqtt config do you have something like:

serial:
  port: 'tcp://10.2.1.25:20108'

And on the remote host running ser2net do you have "dtr=off,rts=off" included in your ser2net.yaml ?

zombielinux, 2021-03-08:
Yep. I actually wrote that documentation and contacted the ser2net author
to get the RTS/DTS options added.

dtmf, 2021-03-08:
> Yep. I actually wrote that documentation and contacted the ser2net author to get the RTS/DTS options added.

Very cool.

In that case it's funny that it is working for me and not you.

I may be doing something wrong.

Let me know if I can provide any other info.

dtmf, 2021-03-08:
@zombielinux 
On the host running ser2net can you see network packets on the ser2net port with something like tcpdump?
Ex:
sudo tcpdump -n -s0  port 20108 or icmp

dtmf, 2021-03-08:
> EDIT: I did a stupid and didn't check that my package manager included the correct git revision of ser2net.

I just saw this edit.  Did that fix your error?

zombielinux, 2021-03-08:
Kinda, I have to muck with the pkgbuild and get a git version going.

zombielinux, 2021-04-09:
The final solution for the ser2net.yaml:

dtr=off,rts=off != rts=off,dtr=off

Order of the two arguments matters. The latest git versions of gensio and
ser2net support operation on arm, arm64, and x86. See the latest
documentation, it has been updated and accurately reflects the state of
slaesh adapters on a remote host.

closed 2021-05-18 by github-actions[bot]

sjorge, 2021-08-01:
I couldn't get the node wrapper to work on my new server, given I am using FreeBSD the ser2net version is still 3.x because there is no gensai support on FreeBSD (AFAIK).

I gave up with the node one and hacked together something in python, it required python 3.8 or newer and a newish pyserial. The ones available on FreeBSD 12 and 13 work fine.

```python
#!/usr/local/bin/python3.8
"""
cc2tcp - Zigbee Adaptor to TCP
"""

import sys
import select
import socket
import ipaddress
import argparse
import logging
import serial
from time import sleep

class Cc2TcpConnection():
    def __init__(self, connection, serial, logger=None):
        self._socket, self._addr = connection
        self._serial = serial
        self._logger = logger if logger else logging.Logger("Cc2TcpConnection")
        self._logger.info(f"Client connected: {self._addr[0]}:{self._addr[1]}")

    def __del__(self):
        self.close()

    @property
    def socket(self):
        """Return reference to socket"""
        return self._socket

    def close(self):
        """Close connection"""
        if self._socket:
            self._socket.close()
            self._socket = None
            self._logger.info(f"Client disconnected: {self._addr[0]}:{self._addr[1]}")

    def fileno(self):
        """emulate fileno method of socket"""
        return self._socket.fileno() if self._socket else None

    def send(self, data):
        """Send data to client"""
        raw_data = []
        for dat in data:
            if dat == 255:
                raw_data.append(255)
            raw_data.append(dat)
        self._socket.send(bytearray(data))

    @staticmethod
    def list_pull_first(data):
        """get first entry from array"""
        dat = data[0]
        del data[0]
        return dat

    def on_received(self, data):
        """Received data from client"""
        data = list(data)
        if data:
            self._serial.write(data)

class Cc2TcpServer():
    """TCP server"""

[... 183 lines cut ...]
    parser.add_argument(
        'baudrate', type=int,
        help="Serial baudrate")
    args = parser.parse_args()

    logging.basicConfig(format='%(asctime)-15s %(message)s')
    logger = logging.getLogger('cc2tcp')
    logger.setLevel(10 if args.verbose else 0)

    cc2tcp = Cc2TcpServer(
        bind_address=args.bind,
        serial_port=[args.port, args.baudrate, args.skip_bootloader],
        logger=logger,
    )
    try:
        while True:
            cc2tcp.process()
    except Exception as err:
        logger.error(f"{err}")
        raise err
    except KeyboardInterrupt:
        pass
    finally:
        cc2tcp.close()
```

airmonitor, 2021-12-28:
@sjorge If you don't mind I will create a small container with your python version.
I need to run this in K8s, hopefully sharing device to k8s pod will not cause any issue, but let see... :)