Koenkk/zigbee2mqtt #6438: Unable to create binding between IKEA motion sensor and bulb

limes007, 2021-02-24:
## What happened
I use an IKEA bulb [LED1836G9](https://www.zigbee2mqtt.io/devices/LED1836G9.html#ikea-led1836g9) with two IKEA motion sensors [E1525/E1745](https://www.zigbee2mqtt.io/devices/E1525_E1745.html#ikea-e1525%252Fe1745). 
I tried to create a binding between on of the motion sensors with the bulb with the command:
`zigbee2mqtt/bridge/request/device/bind` with payload `{"from": "MotionSensor1", "to": "Bulb"}`

But I got the messages:

`Failed to bind cluster 'genOnOff' from 'MotionSensor1' to 'Bulb' (Error: Bind 0x588e81fffe260f93/1 genOnOff from '0x588e81fffe42c532/1' failed (AREQ - ZDO - bindRsp after 10000ms))`
`info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genOnOff","from":"MotionSensor1","to":"Bulb"},"type":"device_bind_failed"}'`
`error Failed to bind cluster 'genLevelCtrl' from 'MotionSensor1' to 'Bulb' (Error: Bind 0x588e81fffe260f93/1 genLevelCtrl from '0x588e81fffe42c532/1' failed (AREQ - ZDO - bindRsp after 10000ms))`
`info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genLevelCtrl","from":"MotionSensor1","to":"Bulb"},"type":"device_bind_failed"}'`
`info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":[],"failed":["genOnOff","genLevelCtrl"],"from":"MotionSensor1","to":"Bulb"},"error":"Failed to bind","status":"error"}'`
`error Failed to bind`

Am I doing something wrong?

## What did you expect to happen
Binding would be created successfully

## How to reproduce it (minimal and precise)
See above

## Debug info
Zigbee2MQTT version: 1.17.1
Adapter hardware: CC2531
Adapter firmware version: zStack12 20190619

nsheridan, 2021-02-25:
You cannot bind IKEA remotes or motion sensors directly to bulbs, only to groups. You need to create a group, add your light or lights to it, and bind the motion sensor to the group.
IKEA remotes tend to only allow binding to a single endpoint, so you may also need to unbind the motion sensor's OnOff from the coordinator

limes007, 2021-03-06:
@nsheridan Thanks for your reply. Unfortunately I get the same error messages using a group instead of the bulb directly.

How can I unbind the motion sensor from the coordinator? Using zigbee2mqtt/bridge/request/device/unbind? What would be the target name?

closed 2021-04-13 by github-actions[bot]

milvert, 2021-04-13:
I have also problem with bind E1747

I think it default group is 0 and I want to move it to group 1.

`Zigbee2MQTT:debug 2021-04-13 21:09:43: Received Zigbee message from 'Motion_3', type 'commandOnWithTimedOff', cluster 'genOnOff', data '{"ctrlbits":0,"offwaittime":0,"ontime":1800}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-04-13 21:09:43: MQTT publish: topic 'zigbee_home_2/Motion_3', payload '{"battery":100,"linkquality":90,"occupancy":true,"requested_brightness_level":76,"requested_brightness_percent":30,"update":{"state":"idle"},"update_available":false}'
`

When I tries to bind to group

` mosquitto_pub -t "zigbee_home_2/bridge/request/device/bind" -h xxx.xx. -p yyyy -m '{"from": "Motion_3", "to": "1"}'`

I receive:

`
Zigbee2MQTT:debug 2021-04-13 21:13:23: Received MQTT message on 'zigbee_home_2/bridge/request/device/bind' with data '{"from": "Motion_3", "to": "1"}'
Zigbee2MQTT:debug 2021-04-13 21:13:23: binding cluster 'genOnOff' from 'Motion_3' to '1'
Zigbee2MQTT:error 2021-04-13 21:13:43: Failed to bind cluster 'genLevelCtrl' from 'Motion_3' to '1' (Error: Bind 0x847127fffebe9e0d/1 genLevelCtrl from '1' failed (AREQ - ZDO - bindRsp after 10000ms))
Zigbee2MQTT:info  2021-04-13 21:13:43: MQTT publish: topic 'zigbee_home_2/bridge/log', payload '{"message":{"cluster":"genLevelCtrl","from":"Motion_3","to":"1"},"type":"device_bind_failed"}'
Zigbee2MQTT:info  2021-04-13 21:13:43: MQTT publish: topic 'zigbee_home_2/bridge/response/device/bind', payload '{"data":{"clusters":[],"failed":["genOnOff","genLevelCtrl"],"from":"Motion_3","to":"1"},"error":"Failed to bind","status":"error"}'
Zigbee2MQTT:error 2021-04-13 21:13:43: Failed to bind
`

Have tried with unbind first from group 0. 
Does E1745 need to be in some state? Have tried with both Occupied and Not Occupied

tcostermans, 2021-08-15:
> You cannot bind IKEA remotes or motion sensors directly to bulbs, only to groups. You need to create a group, add your light or lights to it, and bind the motion sensor to the group.
> IKEA remotes [...]

🙏  Thx @nsheridan , you just made my day. After spending hours on this issue you handed me the solution. Thx a lot!

heikkis, 2023-02-05:
@tcostermans I try to bind Ikea motion sensor with Ikea bulb. I have made a group for the bulb but all binding actions are failing. Can you share some details how to setup? Not sure how to unbind the action from coordinator eg.

I would like to set bulb on when motion is detected and after a while off when no motion is detected. Currently Im doing this with NodeRED but binding would be much nicer way to do the same.

Thanks in advance.

xtrmenl, 2023-12-17:
IKEA peoducts can be linked via touchlink
Turn light on and keep the motion detector near(5cm) and hold the connect button on the motion detector

Using z2m
Add both devices
Create group
Add IKEA bulb

Make sure that motion is in undected mode. By covering it or close the door 
And link the group in motion detector 
And the moment after you press connect, take the cover of the motion detector so it detects 
Thenn you will get the green bind successfully 
If red redo