Koenkk/zigbee2mqtt #6439: IKEA Symfonisk E1744: no "brightness" changes when "legacy: false"

bartplessers, 2021-02-24:
## What happened
- paired IKEA Symfonisk E1744 to zigbee2mqtt
- in "devices.yaml", added "legacy: false"
- rebooted hass (with zigbee2mqtt addon v1.17.1-4)

After reboot and turning the button, following mqtt messages appear:

```
Client mosqsub|30408-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x680ae2fffe1022ae', ... (161 bytes))
{"action":"brightness_move_down","action_rate":195,"battery":74,"brightness":152,"linkquality":129,"rate":195,"update":{"state":"idle"},"update_available":false}
Client mosqsub|30408-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x680ae2fffe1022ae', ... (123 bytes))
{"action":"","battery":74,"brightness":152,"linkquality":129,"rate":195,"update":{"state":"idle"},"update_available":false}
Client mosqsub|30408-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x680ae2fffe1022ae/action', ... (20 bytes))
brightness_move_down
Client mosqsub|30408-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x680ae2fffe1022ae', ... (138 bytes))
{"action":"brightness_stop","battery":74,"brightness":152,"linkquality":126,"rate":195,"update":{"state":"idle"},"update_available":false}
Client mosqsub|30408-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x680ae2fffe1022ae', ... (123 bytes))
{"action":"","battery":74,"brightness":152,"linkquality":126,"rate":195,"update":{"state":"idle"},"update_available":false}
Client mosqsub|30408-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x680ae2fffe1022ae/action', ... (15 bytes))
brightness_stop
Client mosqsub|30408-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x680ae2fffe1022ae', ... (161 bytes))
{"action":"brightness_move_down","action_rate":195,"battery":74,"brightness":152,"linkquality":129,"rate":195,"update":{"state":"idle"},"update_available":false}
Client mosqsub|30408-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x680ae2fffe1022ae', ... (123 bytes))
{"action":"","battery":74,"brightness":152,"linkquality":129,"rate":195,"update":{"state":"idle"},"update_available":false}

```

notice that with several "brightness_move_down", the value remains "brightness":152

Without  "legacy: false", this becomes
```
Client mosqsub|30408-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x680ae2fffe1022ae', ... (134 bytes))
{"action":"rotate_right","battery":74,"brightness":85,"linkquality":153,"rate":195,"update":{"state":"idle"},"update_available":false}
Client mosqsub|30408-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x680ae2fffe1022ae', ... (122 bytes))
{"action":"","battery":74,"brightness":85,"linkquality":153,"rate":195,"update":{"state":"idle"},"update_available":false}
Client mosqsub|30408-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x680ae2fffe1022ae/action', ... (12 bytes))
rotate_right
Client mosqsub|30408-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x680ae2fffe1022ae', ... (135 bytes))
{"action":"rotate_right","battery":74,"brightness":105,"linkquality":153,"rate":195,"update":{"state":"idle"},"update_available":false}
Client mosqsub|30408-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x680ae2fffe1022ae', ... (123 bytes))
{"action":"","battery":74,"brightness":105,"linkquality":153,"rate":195,"update":{"state":"idle"},"update_available":false}
Client mosqsub|30408-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x680ae2fffe1022ae/action', ... (12 bytes))
rotate_right
Client mosqsub|30408-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x680ae2fffe1022ae', ... (135 bytes))
{"action":"rotate_right","battery":74,"brightness":125,"linkquality":153,"rate":195,"update":{"state":"idle"},"update_available":false}
Client mosqsub|30408-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/0x680ae2fffe1022ae', ... (123 bytes))
{"action":"","battery":74,"brightness":125,"linkquality":153,"rate":195,"update":{"state":"idle"},"update_available":false}

```

notice here that "brightness": *** changes

## What did you expect to happen
The value of "brightness" changes also with setting "legacy: false"

Koenkk, 2021-02-24:
I believe the `simulated_brightness` also works for this device (https://www.zigbee2mqtt.io/devices/ICTC-G-1.html#device-type-specific-configuration).

bartplessers, 2021-02-24:
yep, that did the trick.

current config
```
'0x680ae2fffe1022ae':
  friendly_name: '0x680ae2fffe1022ae'
  legacy: false
  debounce: 0.1
  simulated_brightness:
    delta: 25 # delta per interval, default = 20
    interval: 100 # interval in milliseconds, default = 200
```

seems to work smooth at first sight.

kind regards,
B

Koenkk, 2021-02-25:
Good, docs will be updated with this with the next release (1 march)

closed 2021-02-25 by Koenkk

createcandle, 2021-03-25:
I'm having a similar issue, but in my case the simulated brightness is not changing.

This is the config used by the Zigbee2MQTT addon for the Webthings Gateway:
```
homeassistant: false
permit_join: false
devices: devices.yaml
groups: groups.yaml
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM0
advanced:
  availability_timeout: 10
  legacy_api: false
  log_level: error
  log_directory: /tmp
  log_file: Zigbee2MQTT-adapter-%TIMESTAMP%.txt
device_options:
  simulated_brightness:
    delta: 2
    interval: 100
  legacy: false
map_options:
  graphviz:
    colors:
      fill:
        enddevice: '#CCCCCC'
        coordinator: '#333333'
        router: '#666666'
      line:
        active: '#5d9bc7'
        inactive: '#554444'
```