Koenkk/zigbee2mqtt #6441: Problem adding a Blitzwolf BW-SHP13

felix63, 2021-02-24:
## What happened
I included a BlitzWolf BW-SHP13 but it is not recognized although it is on the list of supported devices.

## Debug info
Zigbee2MQTT version: v.1.17.0 (zStack3x0 20201026)
Adapter hardware: Electrolama zig-a-zig-ah!
Adapter firmware version:

In the database:
{"id":80,"type":"Router","ieeeAddr":"0x60a423fffe621547","nwkAddr":25465,"manufId":4098,"manufName":"_TZ3000_g5xawfcq","powerSource":"Mains (single phase)","modelId":"TS0121","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,6,1794,2820],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0121”,"manufacturerName":"_TZ3000_g5xawfcq","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0}}},"binds":[],,"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1614179454052}

In the log:
info  2021-02-24 16:14:49: Successfully interviewed '0x60a423fffe621547', device has successfully been paired
warn  2021-02-24 16:14:49: Device '0x60a423fffe621547' with Zigbee model 'undefined' and manufacturer name 'undefined' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
info  2021-02-24 16:14:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x60a423fffe621547","ieee_address":"0x60a423fffe621547","status":"successful","supported":false},"type":"device_interview"}'

Koenkk, 2021-02-25:
Try updating to Zigbee2MQTT 1.17.1

m0wlheld, 2021-02-26:
I just successfully paired 2 Blitzwolf BW-SHP13 with Zigbee2MQTT 1.17.1. They are identified as "10A UK or 16A EU smart plug (TS0121_plug) by TuYa" in Home Assistant.

felix63, 2021-02-26:
So did I (after updating to 1.7.1) 👍

closed 2021-02-26 by felix63