Koenkk/zigbee2mqtt #6452: Battery power level never changes

davidnite, 2021-02-25:
Battery stats are not updating for any devices in my networking.  I currently have the following two device models with battery level integration:

Aqara RTCGQ11LM
Tuya TS0044

I have 4-5 of each, and every device interviewed correctly and shows a battery status of 100%.  None of the devices ever updates its battery status, and I have already had one battery failure that was never reported, the sensor still showed 100% capacity.

## Debug info
Zigbee2MQTT version: 1.17.1
Adapter hardware: CC2652RB (Slaesh)
Adapter firmware version: 20201026

kremik, 2021-02-28:
Same here with IKEA E1743 switch. Reports low battery level which doesn't change event the battery is replaced.

akunia, 2021-04-02:
Same for my with my 3 RTCGQ11LM , the battery shows 100%

closed 2021-05-12 by github-actions[bot]

tetsuo55, 2021-08-17:
Any updates on this?

davidnite, 2021-08-17:
I've updated z2m several times since opening this issue, no change as of yet.

akunia, 2021-08-17:
I've switched to ZHA integration. More stable for me with Xiaomi Gateway 3 ant the battery is reported correctly.

Minims, 2022-09-01:
I Have the same issue, battery is never updated for this device. It's always displayed @ 100%.
Any solution  ? 

Reconfigure divide does not help.

@Koenkk

Koenkk, 2022-09-01:
What is the voltage level? Anything above 3000mv + will give you 100%,

Minims, 2022-09-01:
My battery has now a voltage of 2.79v. 
Not enough to start the device. 
But the battery has always report 100% during the last months.