Koenkk/zigbee2mqtt #6460: Certaon Components are failing to comunica

oneofthemany, 2021-02-26:
## What happened

New container installation which works fine for around 6 hours then crashes repeatedly, so much so that HA also stops (also in a container)

## What did you expect to happen

Both HA and zigbee2mqtt to work in tandem without issues

## How to reproduce it (minimal and precise)

Juts having zigbee2mqtt on creates the issue

## Debug info
Zigbee2MQTT version: 1.17.1
Adapter hardware: CC2531
Adapter firmware version:
Log:
1 - Pub issue 
Publish 'set' 'state' to 'TV 1' failed: 'Error: Command 0x0017880104537570/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 54277 - 11 - 28 - 6 - 11 after 10000ms)'
Debug Error: Command 0x0017880104537570/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 54277 - 11 - 28 - 6 - 11 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35) at listOnTimeout (internal/timers.js:554:17) at processTimers (internal/timers.js:497:7)
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'TV 1' failed: 'Error: Command 0x0017880104537570/11 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 54277 - 11 - 28 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"TV 1"},"type":"zigbee_publish_error"}'

kremik, 2021-03-01:
Having similar problems:

`Mar 01 08:21:22 raspberrypi4 npm[15244]: Zigbee2MQTT:error 2021-03-01 08:21:22: Publish 'set' 'system_mode' to 'valve6bedroom' failed: 'Error: Command 0x5c0272fffedfb9b3/1 manuSpecificTuya.setData({"status":0,"transid":2,"dp":106,"datatype":4,"length_hi":0,"length_lo":1,"data":[2]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC transaction expired' (240))'`

`Mar 01 08:21:22 raspberrypi4 npm[15244]: Zigbee2MQTT:info  2021-03-01 08:21:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'system_mode' to 'valve6bedroom' failed: 'Error: Command 0x5c0272fffedfb9b3/1 manuSpecificTuya.setData({\"status\":0,\"transid\":2,\"dp\":106,\"datatype\":4,\"length_hi\":0,\"length_lo\":1,\"data\":[2]}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC transaction expired' (240))'","meta":{"friendly_name":"valve6bedroom"},"type":"zigbee_publish_error"}'`

oneofthemany, 2021-04-01:
I have solved the issue by using the electrolama zstack stick. The issue seems to be directly linked to the stick I was using cc2531- note I had 3 sticks all with the same issue.

closed 2021-04-01 by oneofthemany