Koenkk/zigbee2mqtt #6461: NEW Devicesupport! BHT-002-GALZBW (Water Floor Heating)

RBroks, 2021-02-26:
Hello!

Add please floor thermostat, it bean recognized as GCLZBW model, for gas boiler, help pls.

https://www.moeshouse.com/collections/zigbee-thermostat/products/white-zigbee-smart-thermostat-programmable-temperature-controller-zigbee-hub-required-for-water-floor-heating

Robert

insipiens, 2021-03-06:
The gas boiler version also handles the other types as its just a change in which sensor it is set to use. The sensor selection can be changed via an MQTT message. The Z2M device page describes how to do that.

insipiens, 2021-03-06:
I checked and the page now doesn't mention that you can set the source of the external temperature.

From the home assistant MQTT integrations you can publish a message to the topic with payload sensor = either 0,1,or 2 where they correspond to IN/AL/OU choices, or using any MQTT message utility.

My choice is internal so I published 0 to it:

[image]

RBroks, 2021-03-10:
Ou i will try it out!

RBroks, 2021-05-06:
Is there any chance to get  reading in one message to get values of In (room sensor) and Ou (floor sensor)?

closed 2021-05-14 by github-actions[bot]