Koenkk/zigbee2mqtt #6469: zigbee2mqtt docker container keeps alternating between ttyUSB0 and ttyUSB1

MarsWarrior, 2021-02-26:
# Bug Report
## What happened
On every reboot, the container alternates between ttyUSB0 and ttyUSB1 for the zigbee adapter.
This means I have to change the configuration.yaml file on every reboot of my little NUC server.

In the container definition, a /dev/serial/by-id device is mapped to ttyUSB0, but that seems to be ignored.

## What did you expect to happen
ttyUSB0 stays ttyUSB0.

## How to reproduce it (minimal and precise)
On reboot.

## Debug Info
Zigbee2MQTT version: 1.17.0, 1.17.1, 1.16.2
Adapter hardware: CC26X2R1
Adapter firmware version: One from 2020.

m0wlheld, 2021-02-27:
Assignment of ttyUSBx is done on host-level, not inside the container. 

> In the container definition, a /dev/serial/by-id device is mapped to ttyUSB0, but that seems to be ignored.

Please show how you do this, I will try to help.

lux73, 2021-02-27:
so why don't you use /dev/serial/by-id/[yourDevice] in your Config?
this prevents the anytime change between ttyUSB0 & ttyUSB1 on your Host...

MarsWarrior, 2021-02-27:
The Portainer definition uses the following mapping:
[image]

I have changed the device now to ttyUSB3, as USB0 and USB1 are always present in the Alpine container. The Alpine container does not have a serial/by-id mapping.

The configuration.yaml also contains the port ttyUSB3 now.
```yaml
serial:
  port: /dev/ttyUSB3
  disable_led: false
  adapter: zstack
```

I must check if USB3 solves the problem. Using USB0 or USB1 had problems, as I had to change the port number in the configuration.yaml (the docker mapping remained the same?!?!?!?!?!) after EVERY reboot / restart of Docker.

m0wlheld, 2021-02-27:
I makes absolutely no sense to change the serial path *inside* the container multiple times, if the serial path *outside* the container is stable. 

My recommendation is to map /dev/serial/... to `/dev/ttyACM0` and adjust the serial port in configuration.yaml for the last time.

MarsWarrior, 2021-02-27:
I will try that.

The weird thing is, is that I'm also using zwave2mqtt, ie another usb stick - also using the by-id mapping, and that that container has never had any problems after a reboot or upgrade of docker under Ubuntu 20.04 LTS.

m0wlheld, 2021-02-27:
> So it might be caused - in my case - by the different container image.

Maybe. But /dev/ttyACM0 seems safe for both.

MarsWarrior, 2021-02-27:
Indeed!

Will report back if I have done some testing...

luismalddonado, 2021-04-03:
> # Bug Report
> ## What happened
> On every reboot, the container alternates between ttyUSB0 and ttyUSB1 for the zigbee adapter.
> This means I have to change the configuration.yaml file on every reboot of [...]

I had similar problems; finally I had to bind the USB devices to static names and then use those names for the binding
https://www.domoticz.com/wiki/PersistentUSBDevices

MarsWarrior, 2021-05-05:
Well, by using an undefined USB number - USB3 in my case - the problem is solved. I can reboot my NUC without problems: zigbee2mqtt can find the stick!

closed 2021-05-05 by MarsWarrior