Koenkk/zigbee2mqtt #6470: TZE200_nogaemzt Curtain Motor / TuYa TS0601_cover

mich1978, 2021-02-27:
## What happened
[image]
[image]

[https://www.zigbee2mqtt.io/devices/TS0601_cover.html#tuya-ts0601_cover](url)

I was able to successfully pair my curtain motor to zigbee2mqtt, however the state isn't reflected correctly.  When the device is **closed**, the state shows **opened** and vice versa.  The motor responds correct to mqtt publish commands for open, close, reverse_direction, etc.  It does not respond to the invert_cover command.  I wonder if there is something simple I am forgetting or doing incorrectly.
****
## What did you expect to happen
I hoped that the state would be reported correctly.

## How to reproduce it (minimal and precise)
I have two of these identical curtain motors and they both have the same issue.

## Debug info
**Zigbee2MQTT version:**
I am using the EDGE version.

**Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II**

CC2531

**Adapter firmware version:**

I am sorry, I can't figure out where this is.  I just installed it the other day from the instructions on the web:
[https://www.zigbee2mqtt.io/information/flashing_the_cc2531.html](url)

Koenkk, 2021-02-27:
@DomiStyle what would be wise to do here? If I understand correctly the cover can be reverted with a remote? Or should it be added to https://github.com/Koenkk/zigbee-herdsman-converters/blob/d8071c6adac067a64be6acb93f303c5b1385da86/lib/tuya.js#L128

DomiStyle, 2021-02-27:
@Keonkk Since this is the first time this motor is mentioned I would just reverse it by default in the array you linked. Not every motor can be inverted via remote (mine can't) so I wouldn't count on it.

Koenkk, 2021-02-27:
Done! @mich1978 let me know if it works

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

mich1978, 2021-02-28:
I am amazed at how fast you guys responded! 👍

I wonder if there is something I am doing wrong.  Maybe I am impatient.  :(. 

I am running the edge vs in Home Assistant.  It doesn't show any updates for zigbee2mqtt.  I erased the curtain motor and re-added it to zigbee2mqtt a couple minutes ago.  Unfortunately it is still the same issue. Did I not wait long enough? Am I forgetting something stupid?

Thanks again!

Koenkk, 2021-02-28:
> I think the fabric is causing it not to reach the exact close position sometimes. I believe this issue is with the individual curtain.

I don't know if it's possible, but can this curtain be calibrated? @DomiStyle 

> Will this curtain motor always be in the edge version or will it get pushed to the normal version?

It will be in the normal version with the next release (which is on 1 March)

DomiStyle, 2021-02-28:
@mich1978 Can you post the device state after closing a few times and see if it flactuates? Can you also try rerunning the initial calibration as per the device documentation and check if that changes anything?

@Koenkk I'm not aware of any further calibration for the curtain motors except the initial end point calibration.

This is probably the same issue as in https://github.com/Koenkk/zigbee2mqtt/issues/6267. A calibration parameter would be needed in z2m, but it needs the withState first before it would work. An alternative would be to pull the position to 0 if it falls below the calibration parameter.

mich1978, 2021-03-01:
[image]

I reset the curtain a couple times yesterday already.  I was hoping it would fix it.  

The problem yesterday was it would close but only get to position 2 or 3 and it would still say open.  Ideally if it could be set so "1, 2, or 3" was closed.  

Is that possible?

Thanks again for all your help.  I am overwhelmed by your kindness. 😀

DomiStyle, 2021-03-03:
> Ideally if it could be set so "0, 1, 2, 3, or 4" was closed.

This is dependent on https://github.com/Koenkk/zigbee2mqtt/issues/6267, but it is planned.

mich1978, 2021-03-04:
> 
> 
> > Ideally if it could be set so "0, 1, 2, 3, or 4" was closed.
> 
> This is dependent on #6267, but it is planned.

Thanks!~ 🤗

closed 2021-04-11 by github-actions[bot]