Koenkk/zigbee2mqtt #6477: Unbind Error

srtxema, 2021-02-27:
I have a conbee2 coordinator with a lot of device.
i have 2 Xiamit switch https://www.zigbee2mqtt.io/devices/WXCJKG12LM.html#xiaomi-wxcjkg12lm connect with a shelly roller and in the kitchen 2 https://www.zigbee2mqtt.io/devices/TS0121_plug.html#tuya-ts0121_plug
buttons and switch are not connected but from last week when i down roller the tuya switch is on and when i up is off

i have looking for information and i have found bind option.
in web interface i see it

Tuya Bind
[image]

Button Bind
[image]

i havent connected it but it is like this

i have try to unbind pressing heart broken button but it show error

[image]

[image]

any help please?
i want that both devices will be unbind
Thanks!

sjorge, 2021-02-27:
You might need to press a button on the device immediately after hitting the bind/unbind button. If it is sleeping you get a timeout like shown above.

srtxema, 2021-02-27:
it´s with reset button affet unbind! thanks!!

closed 2021-04-06 by github-actions[bot]