Koenkk/zigbee2mqtt #6480: new 2 way MOES switch TS0601 not recognized

EddyGitH, 2021-02-27:
Hi,

I just recently received a 2 way switch from MoesHouse.
Pairing it correctly through domoticz was not working.
I found out that the manufacturername was different in node red so i added it to the devices file fingerprint of zigbeeModel  ['qnazj70', 'kjintbl'],

{modelID: 'TS0601', manufacturerName: '_TZE200_g1ib5ldv'},

now it works, please add this to the branch.

Kind regards
Edwin

Koenkk, 2021-02-27:
Added!
Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-02-27 by Koenkk

EddyGitH, 2021-02-27:
Hi!, Thanks!! 

I was a bit too hasty that all works fine. Just found out that the GET function is not working and also the Toggle function isn't working. Could also be that i am doing something wrong though (I am quite a noob).

I am not using these functions so I do not mind, but my claim was a bit too enthousiastic hahaha
Can I read something how to fix this or we just leave it at this for now.

regards

Edwin