Koenkk/zigbee2mqtt #6481: Lidl HG06492B (Livarno Lux E14 candle CCT) error Timeout

rmickael62, 2021-02-27:
## What happened
It works the first 5 to 10 seconds after pairing or restarting Z2M then you have to wait 20 seconds between each command (until the error message appears) 

## What did you expect to happen
have no error message and be able to use commands faster than every 20 seconds.

## How to reproduce it (minimal and precise)
Use any command in the example of the "ON" log 

## Debug info
Zigbee2MQTT version: Zigbee2mqtt 1.17.1-4 or Zigbee2mqtt Edge (Home Assistant)
Adapter hardware: CC2530+CC2591
Adapter firmware version: 20190109

Logs : https://pastebin.com/mfq4kAXT

rmickael62, 2021-02-27:
#5839

rmickael62, 2021-02-28:
I carried out tests under ZHA and there is no problem. (except CCT which does not work) so it is not a material problem with the bulb

Koenkk, 2021-03-01:
Could you sniff the traffic of the error? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

rmickael62, 2021-03-01:
filter WIRESHARK : (zbee_nwk.src == 0x1ec4) or (zbee_nwk.dst == 0x1ec4)
Traffic Z2M Pairing : [Livarno_Lux_E14_pairing.zip](https://github.com/Koenkk/zigbee2mqtt/files/6064731/Livarno_Lux_E14_pairing.zip)
Traffic Z2M Command "on" : [Livarno_Lux_E14_on.zip](https://github.com/Koenkk/zigbee2mqtt/files/6064732/Livarno_Lux_E14_on.zip)

Traffic ZHA Command "on" (0x35ed) : [Livarno_Lux_E14_on_ZHA.zip](https://github.com/Koenkk/zigbee2mqtt/files/6064733/Livarno_Lux_E14_on_ZHA.zip)

Koenkk, 2021-03-02:
Thanks, can you also provide the herdsman debug log?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

rmickael62, 2021-03-02:
it is already activated on the first post .

debug 2021-03-02 19:31:52: Received MQTT message on 'zigbee2mqtt/0x588e81fffe9c5096/set' with data '{"state":"ON"}'
error 2021-03-02 19:32:15: Publish 'set' 'state' to '0x588e81fffe9c5096' failed: 'Error: Command 0x588e81fffe9c5096/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 8024 - 1 - 3 - 6 - 11 after 10000ms)'
debug 2021-03-02 19:32:15: Error: Command 0x588e81fffe9c5096/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 8024 - 1 - 3 - 6 - 11 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)

Koenkk, 2021-03-02:
It's not the herdsman debug logging but the normal debug logging. The herdsman debug logging is only logged to stdout.

rmickael62, 2021-03-02:
This is correct? 
[log_02_03_2021.txt](https://github.com/Koenkk/zigbee2mqtt/files/6071274/log_02_03_2021.txt)

Koenkk, 2021-03-02:
Can you also provide the network key of your ZHA sniff?

Koenkk, 2021-03-02:
BTW, I also see you are running an old firmware, please update to https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default

rmickael62, 2021-03-02:
Key ZHA : 70:25:bd:76:90:d3:c8:24:e7:28:f7:41:49:7b:de:c5

rmickael62, 2021-03-02:
I just updated the coordinator but nothing works suddenly 
all the connections have disappeared 
[image]
🤕

Koenkk, 2021-03-02:
Reflashing clears the network table, that doesn't mean devices are not connected anymore.

One difference I see with the ZHA setup that there is a direct connection between the coordinator and bulb. With Z2M it's goes via `0xbdf6`. Can you repair the bulb close to the coordinator so that it also has a direct connection?

rmickael62, 2021-03-02:
Everything begins to return to order but unfortunately the problem is still present even with the bulb to pair next to the coordinator .
I think the bulb will soon land in the trash !!
A big thank you for your help Koenkk

kbuc, 2021-04-01:
Same problem here with Lidl HG06106C, soon after switching on or restart of zigbee2mqtt:

`Publish 'set' 'color_temp' to 'RGB_1' failed: 'Error: Command 0x588e81fffeaf6feb/1 lightingColorCtrl.moveToColorTemp({"colortemp":370,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51510 - 1 - 20 - 768 - 11 after 10000ms)'`

Running version 1.18.1-1, latest coordinator firmware on zzh.

closed 2021-05-10 by github-actions[bot]