Koenkk/zigbee2mqtt #6482: Provide hardware for OSRAM Smart+ Plug AB32570

Larsen-Locke, 2021-02-27:
This is a re-opening of a closed issue:
The state of the OSRAM Smart+ Plug ist updated when device is switched with the hardware button.
The log does not show any entries when the button is triggered.

I would be happy to contribute by donating a device to Koenkk (I've got quite a few). If you're interessted please let me know where to send it.

Koenkk, 2021-02-27:
Can you provide the data/databse.db entry of your device?

Larsen-Locke, 2021-02-27:
Of course. Here it is:

{"id":7,"type":"Router","ieeeAddr":"0x7cb03eaa0a0a600b","nwkAddr":5352,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Plug 01","epList":[3],"endpoints":{"3":{"profId":49246,"epId":3,"devId":16,"inClusterList":[4096,0,3,4,5,6,2820,64527],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Plug 01","manufacturerName":"OSRAM","powerSource":1,"zclVersion":1,"appVersion":9,"stackVersion":2,"hwVersion":1,"dateCode":"20140331DEOS****","swBuildId":"V1.05.11"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b77e74","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}}},"appVersion":9,"stackVersion":2,"hwVersion":1,"dateCode":"20140331DEOS****","swBuildId":"V1.05.09","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1614462993640}

 

Cheers

Carsten

 

PS: my great respect and admiration for all the work and knowledge that you put in zigbee2mqtt. Great fun to play around and so happy to get rid of all the chinese smart home accounts.

Koenkk, 2021-02-28:
This look correct to me. You see that zigbee2mqtt successfully setup reporting + binding for the on/off cluster. To debug this further it would be interesting to check if this works with the official gateway. If it does you can make a sniff of it: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Larsen-Locke, 2021-02-28:
Hi,
The plug works with the original gateway, i tested that already.
At the moment I only got a CC2652 and a CC2530 and as far as i have
understood i cannot sniff with them.
I would be happy to send you one plug for free if you're interested but no
worries if that's not on your list.

Respect for your great project, incredible support and Regards
carsten

Koenkk, 2021-03-01:
> I would be happy to send you one plug for free if you're interested but no
worries if that's not on your list.

That won't help much (since I don't have the official gateway). Would be good if you can sniff the traffic with the original gateway.

Larsen-Locke, 2021-03-01:
ok. I ordered a cc2531. Let's see how far i get...

Regards

Larsen-Locke, 2021-03-14:
Hi Koen,

 

sorry, it took me a little to get a stick for sniffing the traffic and then the adventure to get the firmware on the stick by soldering jumper cables…

I guess I finally made it 

 

Attached are two files: 

*	Sniff with original Gateway
*	Sniff with the CC2652-Coordinator.

 

I’m new tot he sniffing-business, so if there is anything I can do with my limited skills please tell me .

 

 

Big thanks for your great work, zigbee2mqtt works great. My Livolo-Switches run much better than with the original gateways, all Aquara and Heiman smoke sensors work perfect. Really a pleasure.

 

Regards, 

Carsten

Koenkk, 2021-03-16:
I don't see the files, can you re-upload them?

Larsen-Locke, 2021-03-16:
Hmm, attachments look ok in the outbox.
Can you read from Google-drive?

https://drive.google.com/folderview?id=1Ub1x6h-18xPqolq2SdWY2R-48mivKykk

Regards
***@***.***

Koenkk, 2021-03-17:
Thanks, for both network I need the network key. Can you also take the sniff starting from pairing the plug till controlling it manually?

Larsen-Locke, 2021-04-02:
After deleting and rebinding the osram plugs they now receive state changes triggered from the button. It has nothing to do with the firmware cause it was already the latest version. I updated zigbee2mqtt in the meantime so maybe there was a change solving the issue.
I think it can be closed now cause the plugs work perfectly. It's been stated that osram plugs are not reliable as routers but i have to compare a little more against my innr plugs.

closed 2021-04-02 by Larsen-Locke

ankorth, 2021-04-14:
I'd like to let you know that I found this ticket because I had also the problem that pressing the on/off button on the device didn't produce any message. After reading this, I removed the device from network as described [MQTT topics and message structure documentation](https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestdeviceremove) and after pairing it again it worked. This was two days after my initial setup. The only noteworthy what changed in that time is that I found an issue on the antenna which caused a massive drop of link quality and that - by accident - two IKEA Tradfri LED drivers have been added.