Koenkk/zigbee2mqtt #6484: Zemismart ZM25TQ Roller Shade Motor (Zigbee version)

antti365, 2021-02-27:
Hi, I bought a curtain motor Zemismart ZM25TQ, I can't add it to z2m. zigbee2mqtt version 1.17.1 commit: 99274c7. coordinator type zStack30x.
In the log files, when adding, I observe the following:
`info  2021-02-27 18:02:39: Device '0x5c0272fffea4d1ae' joined
info  2021-02-27 18:02:39: MQTT publish: topic 'Zigbee/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffea4d1ae","ieee_address":"0x5c0272fffea4d1ae"},"type":"device_joined"}'
info  2021-02-27 18:02:39: Starting interview of '0x5c0272fffea4d1ae'
info  2021-02-27 18:02:39: MQTT publish: topic 'Zigbee/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffea4d1ae","ieee_address":"0x5c0272fffea4d1ae","status":"started"},"type":"device_interview"}'
info  2021-02-27 18:02:39: MQTT publish: topic 'Zigbee/bridge/log', payload '{"message":{"friendly_name":"0x5c0272fffea4d1ae"},"type":"device_connected"}'
info  2021-02-27 18:02:39: MQTT publish: topic 'Zigbee/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x5c0272fffea4d1ae"},"type":"pairing"}'
error 2021-02-27 18:02:49: Failed to interview '0x5c0272fffea4d1ae', device has not successfully been paired
info  2021-02-27 18:02:49: MQTT publish: topic 'Zigbee/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffea4d1ae","ieee_address":"0x5c0272fffea4d1ae","status":"failed"},"type":"device_interview"}'
info  2021-02-27 18:02:49: MQTT publish: topic 'Zigbee/bridge/devices', payload '[{"definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]}},"friendly_name":"0x5c0272fffea4d1ae","ieee_address":"0x5c0272fffea4d1ae","interview_completed":false,"interviewing":false,"network_address":29125,"supported":false,"type":"Router"}]'
info  2021-02-27 18:02:49: MQTT publish: topic 'Zigbee/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x5c0272fffea4d1ae"},"type":"pairing"}'`

antti365, 2021-02-28:
In the database.db file, the device is registered as:
`{"id":14,"type":"Router","ieeeAddr":"0x5c0272fffea4d1ae","nwkAddr":29125,"manufId":4098,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1614441818082}`

antti365, 2021-02-28:
After upgrading to version 1.17.1-dev commit: 12c34a8 the motor was successfully added to z2m as TS0601_cover, but there are some inaccuracies... is it possible to eliminate them?
1. Option motor_speed in device exposes does not work correctly, or rather does not work at all

`Info Setting motor speed to 255
Warning TuYa_cover_control: Unhandled DP #12 for _TZE200_fzo2pocs: {"status":0,"transid":1,"dp":12,"datatype":5,"fn":0,"data":{"type":"Buffer","data":[1]}}
Warning TuYa_cover_control: Unhandled DP #103 for _TZE200_fzo2pocs: {"status":0,"transid":1,"dp":103,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}}
Warning TuYa_cover_control: Unhandled DP #104 for _TZE200_fzo2pocs: {"status":0,"transid":1,"dp":104,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}}
Info MQTT publish: topic 'Zigbee/0x5c0272fffea4d1ae', payload '{"last_seen":1614519924906,"linkquality":21,"motor_speed":false,"position":0,"running":false,"state":"CLOSE"}'
Warning TuYa_cover_control: Unhandled DP #106 for _TZE200_fzo2pocs: {"status":0,"transid":1,"dp":106,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[0]}}`

2. When controlling the motor from the remote control, the status (close/open) in z2m changes only when the up button is pressed; when the down button is pressed, no changes occur.
3. The picture of the device does not correspond to reality - do you use the same picture for a bunch of different devices? Well, this is not the most important thing ...

EddieDSuza, 2021-03-31:
[image]
I can see the device in the frontend but cannot control the device in HomeKit via Homebridge

antti365, 2021-04-02:
@Koenkk, please tell me if it is possible to teach this motor to understand the "stop" command so that it can be controlled using, for example, this switch - IKEA E1766 (https://www.zigbee2mqtt.io/devices/E1766.html). So that it stops when the key is released and the switch sends a stop command. At the moment, it is impossible to stop the motor between 0 and 100 positions using the switch, which would be very convenient for users in the family.
And the questions above - how to remove the "motor speed" option, this model does not support it and replace the icon with your own. Thanks.

Koenkk, 2021-04-04:
@antti365  have you try sending `zigbee2mqtt/[FRIENDLY_NAME]/set` `{"state": "STOP"}` to the device? This device can definitely not be controlled with the E1766 since this TuYa device talks non standard zigbee.

antti365, 2021-04-04:
@Koenkk Yes, indeed, this command works, and as I did not think to try it myself ... thanks!
Now how would you remove the option “motor speed”, which is not supported by this motor?

Koenkk, 2021-04-05:
I cannot, the current definition contains a lot of similar devices, some support it some don't (and I don't want to split up: https://github.com/Koenkk/zigbee-herdsman-converters/blob/49139afa6a05ded0a43ebc81ab80889ea9f0e59e/devices.js#L1609

antti365, 2021-04-10:
@Koenkk Need help! Today three more of the same motors have come and it is not possible to connect them, did zemismart change something or what happened?
[image]

closed 2021-04-10 by antti365