Koenkk/zigbee2mqtt #6497: Incorrect state message to zigbee2mqtt with ptvo.switch and group

yurganov, 2021-02-28:
## What happened

Using ptvo.switch DIY device. If I switch one relay, correct message send to mqtt:
{"linkquality":52,"state":"ON","state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF"}
{"linkquality":49,"state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF"}

But if I add 3 of 5 (1-3) relay to group in zigbee2mqtt and click it, incorrect message send:

{"linkquality":52,"state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF"}
{"linkquality":52,"state":"OFF","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF"}

## What did you expect to happen

{"linkquality":52,"state":"ON","state_l1":"ON","state_l2":"ON","state_l3":"ON","state_l4":"OFF","state_l5":"OFF"}

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.17.1-dev commit: 9d835ff
Adapter hardware: CC2531
Adapter firmware version:

Koenkk, 2021-02-28:
@ptvoinfo can you check this?

ptvoinfo, 2021-02-28:
@yurganov, Could you please attach a screenshot with your firmware settings? I'll try to repeat this issue.

yurganov, 2021-02-28:
@ptvoinfo, I’ve bought them from one guy, he will attach screenshot as he can, he is not near the PC nearest time. 

Maybe I can attach some another information to figure it out? If not, we need to pause it. 

Thank you for reply!

closed 2021-04-07 by github-actions[bot]