Koenkk/zigbee2mqtt #6499: Moes BHT-002 desn't accept non-integer setpoint

MiloshCZ, 2021-02-28:
Hi
# Bug Report
## What happened
I would like set "current_heating_setpoint" to non-integer value with 0.5°C steps. But when I try set something different than integer value, setpoint is set to 0.0°C
## What did you expect to happen
Setpoint is set to right value, or to some supported value. But right value is preffered.
## How to reproduce it (minimal and precise)
Publish _{"current_heating_setpoint": "15.5"}_ to the _zigbee2mqtt/FRIENDLY_NAME/set_.
Current answer is 
`{"child_lock":"UNLOCKED","current_heating_setpoint":0,"heat":"OFF","last_seen":1614524818395,"linkquality":21,"local_temperature":20.5,"preset":"hold","preset_mode":"hold","program":[{"p1":"6h:0m 40°C"},{"p2":"11h:30m 42°C"},{"p3":"13h:30m 44°C"},{"p4":"17h:30m 46°C"},{"sa1":"6h:0m 48°C"},{"sa2":"12h:0m 46°C"},{"sa3":"14h:30m 44°C"},{"sa4":"17h:30m 42°C"},{"su1":"6h:0m 38°C"},{"su2":"12h:30m 40°C"},{"su3":"14h:30m 42°C"},{"su4":"18h:30m 40°C"}],"system_mode":"heat"}`
## Debug Info
Zigbee2MQTT version: 1.17.1 commit: 99274c7
Adapter hardware: CC1352P-2
Adapter firmware version: 20201010
coordinator type: zStack30x

1300371, 2021-03-02:
Yep
If I remember well, it can be achieved due to the way Tuya implements their stuff. :(
https://github.com/Koenkk/zigbee2mqtt/issues/4610

MiloshCZ, 2021-03-03:
Hi,
thanks for your reply. Is it possible to implement some workaroud, that setpoint will be set to supported integer value?

closed 2021-04-10 by github-actions[bot]