Koenkk/zigbee2mqtt #6500: Xiaomi ZNLDP12LM UNSUPPORTED_ATTRIBUTE

SargonofAssyria, 2021-02-28:
Zigbee2MQTT version 1.17.1-dev commit: 1f60c17
Coordinator type zStack3x0
Coordinator revision 20210120
Frontend version 0.3.72## Debug info
Adapter hardware: CC1352P-2

In the UI it exposes 'color_temp_startup' but when I try to set in to neutral or any other color temperature is gives the error
`error 2021-02-28 17:00:15: Publish 'set' 'color_temp_startup' to 'Overlooplamp' failed: 'Error: Write 0x00158d0002b7d689/1 lightingColorCtrl({"startUpColorTemperature":150}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
`
All those bulbs give the same error.

Koenkk, 2021-03-01:
Devices don't support it, fixed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/2310

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-03-01 by Koenkk

SargonofAssyria, 2021-03-02:
Tnx, Solved.