Koenkk/zigbee2mqtt #6502: Samsung Smartthings Multipurpose sensor (2018 model) (IM6001-MPP01) - Tilt/accelerometer feature

w0lf3h, 2021-02-28:
Saw that [support](https://github.com/Koenkk/zigbee-herdsman-converters/commit/8e9cc3fd8c0b2f82e0755884ee6be35416948861) was added for a previous model of this sensor ([F-MLT-US-2](https://github.com/Koenkk/zigbee-herdsman-converters/pull/2260)) and wanted to see how to get this added for the IM6001-MPP01 model as well.

Debug logging looks similar when tilt is activated.
`Zigbee2MQTT:debug 2021-02-28 15:32:54: Received Zigbee message from '0x286d9700010d57a6', type 'attributeReport', cluster 'manuSpecificSamsungAccelerometer', data '{"acceleration":1}' from endpoint 1 with groupID 0`

Zigbee2mqtt version: 1.17.1-dev commit: 57165e3

w0lf3h, 2021-03-01:
Not sure if I was supposed to open an issue or just post a discussion. I'm not running docker version (running addon zigbee2mqtt so unable to actually edit the files, and I think the device and converter needs to be edited rather than just adding an external converter. At least I'm having difficulties when attempting to add a new device with those parameters.

orrpan, 2021-03-01:
Working on it:
[moving/acceleration](https://github.com/Koenkk/zigbee-herdsman-converters/pull/2295)
[xyz](https://github.com/Koenkk/zigbee-herdsman-converters/pull/2295)

w0lf3h, 2021-03-01:
https://github.com/Koenkk/zigbee-herdsman-converters/commit/628a06204e96fce0899288a222dfa2e9c494482c

closed 2021-03-01 by w0lf3h