Koenkk/zigbee2mqtt #6503: Latest dev version breaks "switch as light" functionality

brokenmass, 2021-03-01:
## What happened
The latest dev version of zigbee2mqtt breaks the option to redefine the homeassistant type of the device, typically used to redefine switch as lights as introduced in https://github.com/Koenkk/zigbee2mqtt/pull/4593

## What did you expect to happen
A switch redefined as light should appear as light in homeassistant

## How to reproduce it (minimal and precise)
See https://github.com/Koenkk/zigbee2mqtt/pull/4593 examples

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

avvalex, 2021-03-19:
I can confirm, if i'm not wrong, that latest 1.18.1 has this feature broken. MQTT keeps publishing (and HA discovering) the entity as a switch.
A temporary workaround is ---> https://www.home-assistant.io/integrations/light.switch/
Thanks for your great job!

closed 2021-04-26 by github-actions[bot]

brokenmass, 2021-05-13:
@Koenkk can you look into this , please ?