Koenkk/zigbee2mqtt #6506: TuYa Valve: Zigbee_error (MAC transaction expired)

kremik, 2021-03-01:
## What happened

After my RPi was updated like 1w ago, I'm getting these errors and like everyday I have to restart zigbee2mqtt because it is not responding. In the log I found following errors with MAC transaction expired and zigbee_publish_error. These are related to TuYa Thermostatic Valve

`
Mar 01 09:02:53 raspberrypi4 npm[26972]: Zigbee2MQTT:error 2021-03-01 09:02:53: Publish 'set' 'system_mode' to 'valve1livingroom' failed: 'Error: Command 0x5c0272fffec9d6c1/1 manuSpecificTuya.setData({"status":0,"transid":7,"dp":106,"datatype":4,"length_hi":0,"length_lo":1,"data":[0]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC transaction expired' (240))'
`

`Mar 01 09:02:53 raspberrypi4 npm[26972]: Zigbee2MQTT:info  2021-03-01 09:02:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'system_mode' to 'valve1livingroom' failed: 'Error: Command 0x5c0272fffec9d6c1/1 manuSpecificTuya.setData({\"status\":0,\"transid\":7,\"dp\":106,\"datatype\":4,\"length_hi\":0,\"length_lo\":1,\"data\":[0]}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC transaction expired' (240))'","meta":{"friendly_name":"valve1livingroom"},"type":"zigbee_publish_error"}'`

## What did you expect to happen
Not to see these errors ;-)

## How to reproduce it (minimal and precise)
I get this error again when I try to change system_mode of the valve

## Debug info
Zigbee2MQTT version: 1.17.1 commit: 99274c7
Adapter hardware: CC2531
Adapter firmware version:
zStack12
coordinator revision
20201127

Koenkk, 2021-03-01:
Are these errors fixed after repairing?

kremik, 2021-03-18:
No still the same error occures.

Koenkk, 2021-03-18:
Can you create a networkmap and post it here? https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestnetworkmap

kremik, 2021-03-18:
> Can you create a networkmap and post it here? https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequestnetworkmap

Network maps produces strange results every time a try to generate it. Sometimes devices without any connections, etc
Here it is. In the mean time I had to replace the coordinator with the second one I wasnt able to start Z2M, this one uses older zstack:
https://pastebin.com/DbXKKit4

Koenkk, 2021-03-19:
Can you also post the picture (I'm interested in where valve1livingroom is connected to)

kremik, 2021-03-22:
Sorry I think it won't be relevant, because my Z2M didn't want to start hersman with my C2531 updated to the latest firmware (20201127) version. So I had to disconnect it and replace it with other one running the older firmware version (20190608) and now the error won't show any more. Also with the oher coordinator the network map looks better and problems with IKEA light bulb disappeared. So I don't know what was the problem if the firmware of the coordinator or hardware itself :(

closed 2021-04-30 by github-actions[bot]