Koenkk/zigbee2mqtt #6509: Lidl FB20-002 the "on" button is only evaluated once

makoONE, 2021-03-01:
The status of the "on" button is only evaluated for a single actuation.
Pressing the "on" button again will be ignored. The other buttons ("off", "brightness up", "brightness down)" generate an event each time they are pressed, as expected.
The correct function is required for a multiple assignment of the button.

OMotornyi, 2021-11-10:
Hello. Were you able to figure this one out?

OMotornyi, 2022-08-31:
yeah... It's a shame though really, basically renders this remote less useful as we can't bind a completely independent actions on every button

chris400, 2023-10-08:
I'm so glad that I found that issue. I was starting to question myself bc I thought I made a mistake in my Home Assistant configuration.

So here is my though on this:
The first "on" command looks like this
`Debug 2023-10-08 22:10:54Received Zigbee message from 'ZigBee Remote 003', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0`
the next 4 times it looks like this
`Debug 2023-10-08 22:11:01Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,79,253,0],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:02Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,79,253,0],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:02Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,79,253,0],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:03Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,79,253,0],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:04Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,79,253,0],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:13Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,80,253,1],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:13Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,80,253,1],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:14Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,80,253,1],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:15Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,80,253,1],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:15Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,80,253,1],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:27Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,81,253,2],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:28Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,81,253,2],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:29Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,81,253,2],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:30Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,81,253,2],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:30Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,81,253,2],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:41Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,82,253,3],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:41Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,82,253,3],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:42Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,82,253,3],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:43Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,82,253,3],"type":"Buffer"}' from endpoint 1 with groupID null
Debug 2023-10-08 22:11:44Received Zigbee message from 'ZigBee Remote 003', type 'raw', cluster 'genOnOff', data '{"data":[1,82,253,3],"type":"Buffer"}' from endpoint 1 with groupID null`

I would like to request, that zigbee2mqtt interprets these commands as "On" as well.

Kaskadeur111, 2023-11-05:
same here, find it less usefull, if that ON Button only works once

chris400, 2023-12-10:
@makoONE I just came back here. Would you mind re-opening this issue and removing staklle lable?

Bastelritius, 2023-12-28:
Would be nice when this can be solved and the RC sending always the on command when pushing the on button

jnswb, 2024-02-08:
I second this proposal.

chris400, 2024-02-08:
Bc @makoONE is not responding, would you @Koenkk mind re-opening this issue?

Virusmater, 2024-06-15:
Same issue there, but I don't even see any RAW commands in logs. Is it because my remote was never connected to Tuya?

chris400, 2024-06-16:
> Same issue there, but I don't even see any RAW commands in logs. Is it because my remote was never connected to Tuya?

I don't think so. I have never used anything else than Zigbee2MQTT with my device and I did see these commands.
Did you set the right log level?

Virusmater, 2024-06-16:
I don't think so, logs are on the debug level and I can see codes of other button presses in logs.
Also my remote looks slightly different from the one on the z2m page:
[image]

akula3, 2024-07-04:
Is here a solution or workaround?

Koenkk, 2024-07-06:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/d524d84f7e5e3b9d20559dd296d005b2): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Virusmater, 2024-07-06:
Hello @Koenkk , it worked after the small change: `type: 'raw'` to `type: 'commandTuyaAction'`
Full gist:
https://gist.github.com/Virusmater/13a39d8d2fed6d2b110d1cf8968decb1

Logs before:
```
[2024-07-06 15:35:50] debug: 	z2m: Received Zigbee message from 'Guest Blinds Remote', type 'commandTuyaAction', cluster 'genOnOff', data '{"data":{"data":[],"type":"Buffer"},"value":2}' from endpoint 1 with groupID 0
[2024-07-06 15:35:50] debug: 	z2m: No converter available for 'FB20-002' with cluster 'genOnOff' and type 'commandTuyaAction' and data '{"data":{"data":[],"type":"Buffer"},"value":2}'
[2024-07-06 15:35:50] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-07-06 15:35:50] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-07-06 15:35:50] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-07-06 15:35:50] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-07-06 15:35:50] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-07-06 15:35:50] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,33,231]
[2024-07-06 15:35:50] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,33,231]
[2024-07-06 15:35:50] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,33] - 231
[2024-07-06 15:35:50] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":33}
[2024-07-06 15:35:50] debug: 	zh:zstack:unpi:parser: --- parseNext []
```

Logs now:
```
[2024-07-06 15:44:17] debug: 	z2m: Received Zigbee message from 'Guest Blinds Remote', type 'commandTuyaAction', cluster 'genOnOff', data '{"data":{"data":[],"type":"Buffer"},"value":0}' from endpoint 1 with groupID 0
[2024-07-06 15:44:17] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Guest Blinds Remote', payload '{"action":"on","linkquality":84}'
[2024-07-06 15:44:17] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Guest Blinds Remote', payload '{"action":"","linkquality":84}'
[2024-07-06 15:44:17] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Guest Blinds Remote/action', payload 'on'
[2024-07-06 15:44:17] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-07-06 15:44:17] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-07-06 15:44:17] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-07-06 15:44:17] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-07-06 15:44:17] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-07-06 15:44:17] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,22,208]
[2024-07-06 15:44:17] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,22,208]
[2024-07-06 15:44:17] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,22] - 208
[2024-07-06 15:44:17] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":22}
[2024-07-06 15:44:17] debug: 	zh:zstack:unpi:parser: --- parseNext []
```

 
Thank you!

akula3, 2024-07-06:
Hi @Koenkk ,
the proposal from @Virusmater works!

Koenkk, 2024-07-08:
Great, integrated the fix. You can remove the ext converter after updating to the next release.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-07-08 by Koenkk

stefan131, 2024-07-17:
I was about to jump into this issue, all of my ceiling lights came with this remote and they were useless before
Thanks for the great work! Looking forward to the next release :)

stefan131, 2024-08-14:
After some testing I found out the first 'on' button press is not registered anymore after this update. @Koenkk could the issue be opened again?

stefan131, 2024-08-14:
> Do you have the steps to reproduce? I have some issues when the first button press doesn't get registered, but I think it is not related to the update, I had it before. The remote drops out from the Zigbee network and on the first step, it just reconnects to it.
> 
> Cannot debug it, because it happens only after I haven't used the remote for a week

My experience is that Zigbee remotes start blinking when they are reconnecting.
For me it just shows the normal green light, and the second press immediately works.
I'll try to debug it some more when I get home.

Reproducing is easy: don't touch your remote for about 30 minutes and push the on button.
I'll do some more testing tonight, for example if pressing off and then on gives the same result. I vaguely remember that it did