Koenkk/zigbee2mqtt #6511: Error after update 1.18.0

tricky16, 2021-03-01:
After update to versión 1.18.0 I have a error to start the addon.
The message is:
Failed to start addon - configuration validation failed!
Missing option ota in root.

¿Any idea?

Thanks

OlivierAMG, 2021-03-01:
I had the same issue 5 minutes ago after the update, fixed by setting following attributes in the add-on configuration... here the default values copied from the documentation (https://www.zigbee2mqtt.io/information/configuration.html):

```
ota:
  update_check_interval: 10
  disable_automatic_update_check: false
```

m0wlheld, 2021-03-01:
Is this about Zigbee2MQTT or the Zigbee2MQTT add-on for Home Assistant? Just upgraded from 1.17.1 to 1.18.0 (using koenkk/zigbee2mqtt container image) and no issues, though I do not have an `ota` section in my configuration.yaml.

m0wlheld, 2021-03-01:
Okay, I was still on 1.17.1, my fault. 1.18.0 does not start for another reason.

Koenkk, 2021-03-02:
@ciotlosm I added `ota` in https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/commit/d041a3ea3dd34ece605c50bd334bede123def4b0 , but if I understand this correctly this breaks existing setups?

boulete31, 2021-03-04:
Hello,
Same here but i'm on windows 10 64 with node.js for windows (works great since 2 years ) .
If i stay on 1.17.1 or lower everything run fine but if I try to run 1.18 or higher nothing works (doesn't start)    .

ciotlosm, 2021-03-05:
@Koenkk if that parameter is required, you should add a default. It should not break anything if it's an optional parameter.

ciotlosm, 2021-03-05:
@Koenkk I checked and the problem is with how configs are validated by the Supervisor. If you added in the schema:
```
    "ota": {
      "update_check_interval": "int?",
      "disable_automatic_update_check": "bool?"
    },
```

You should have added in the default config at minimum:
```
"ota":{},
```

You can't make 1st level optional unfortunately :(

Koenkk, 2021-03-06:
> You can't make 1st level optional unfortunately :(

OTA section is option, so I removed it in the 1.18.1-1 release.

ciotlosm, 2021-03-08:
Yes, I think this can be closed now. There is a separate discussion if we can minimise the add-on options passed, to allow the new built in UI to take over. See: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/41

closed 2021-03-08 by Koenkk