Koenkk/zigbee2mqtt #6513: zigbee2mqt not starting after upgrade to version 1.18.0

ruttromp, 2021-03-01:
Running zigbee2mqt as docker image on Raspberry pi 4
Updated to docker image to version 1.18.0

Z2M is refusing to start.  Full error log:
`
info  2021-03-01 20:52:45: Starting Zigbee2MQTT version 1.18.0 (commit #3fd6619)
info  2021-03-01 20:52:45: Starting zigbee-herdsman (0.13.72)
error 2021-03-01 20:52:55: Failed to call 'OnEvent' 'stop' (TypeError: Cannot read property 'getEntries' of null
    at Function.loadFromDatabaseIfNecessary (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:220:55)
    at Function.all (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:240:16)
    at Controller.getDevices (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:273:31)
    at Zigbee.getClients (/app/lib/zigbee.js:149:30)
    at OnEvent.stop (/app/lib/extension/onEvent.js:23:42)
    at Controller.callExtensionMethod (/app/lib/controller.js:380:44)
    at async Controller.stop (/app/lib/controller.js:187:9)
    at async stop (/app/index.js:46:5)
    at async process.handleQuit (/app/index.js:52:9))
error 2021-03-01 20:52:55: Not connected to MQTT server!
error 2021-03-01 20:52:55: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
info  2021-03-01 20:52:55: Disconnecting from MQTT server
error 2021-03-01 20:53:05: Error while starting zigbee-herdsman
`

Same after rebooting the RPi

Zigbee2MQTT version: 1.18.0
Adapter hardware: slae.sh cc2652rb stick
Adapter firmware version: 20210120 (zStack3)

Rolled back to 1.17.1 and it's running without issues. Any idea what's wrong?

mcgeri, 2021-03-01:
Put his into your config:
# Optional: OTA update settings
ota:
    # Minimum time between OTA update checks, see https://www.zigbee2mqtt.io/information/ota_updates.html for more info
    update_check_interval: 10
    # Disable automatic update checks, see https://www.zigbee2mqtt.io/information/ota_updates.html for more info
    disable_automatic_update_check: false

This resolved my problem.

aesedepece, 2021-03-01:
@mcgeri's solution worked as well for me

ruttromp, 2021-03-02:
tried @mcgeri solution. I get the same error after adding the OTA configuration settings. This issue is duplicate with #6517

Koenkk, 2021-03-02:
@ruttromp can you provide your data/database.db?

ruttromp, 2021-03-02:
Closing this issue. Will watch and contribute to #6517.

closed 2021-03-02 by ruttromp