Koenkk/zigbee2mqtt #6517: 1.18.0 won't start: "Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)"

m0wlheld, 2021-03-01:
## What happened
Just switched from Zigbee2MQTT 1.17.1 (koenkk/zigbee2mqtt:1.17.1) to 1.18.0 (koenkk/zigbee2mqtt:latest) and now Zigbee2MQTT won't start any more (see below). Configuration remained unchanged.

```
Zigbee2MQTT:info  2021-03-01 22:42:12: Logging to console only'
Zigbee2MQTT:info  2021-03-01 22:42:12: Starting Zigbee2MQTT version 1.18.0 (commit #3fd6619)
Zigbee2MQTT:info  2021-03-01 22:42:12: Starting zigbee-herdsman (0.13.72)
Zigbee2MQTT:error 2021-03-01 22:42:33: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-03-01 22:42:33: Failed to start zigbee
Zigbee2MQTT:error 2021-03-01 22:42:33: Exiting...
Zigbee2MQTT:error 2021-03-01 22:42:33: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.18.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.18.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-03-01T21_42_33_282Z-debug.log
```

## What did you expect to happen
I expected Z2M to start.

## How to reproduce it (minimal and precise)

My configuration (network key changed):
```
advanced:
  channel: 11
  homeassistant_legacy_triggers: false
  legacy_api: false
  log_level: info
  log_output:
    - console
  network_key: GENERATE
  pan_id: GENERATE
homeassistant: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://mqtt'
  user: zigbee2mqtt
  password: *************
permit_join: false
serial:
  port: /dev/ttyACM0
devices: devices.yaml
groups: groups.yaml
```` 

Service definition in docker-compose file:

```
version: '2'
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    depends_on:
      - mqtt
    devices:
      - /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_44_12-if00-port0:/dev/ttyACM0
    environment:
      - TZ=Europe/Berlin
    image: koenkk/zigbee2mqtt:latest
    restart: always
    volumes:
      - /srv/zigbee2mqtt/zigbee2mqtt-volume:/app/data
```` 
## Debug info
Zigbee2MQTT version: 1.18.0 (official Docker image)
Adapter hardware: CC26X2R1 (slae'sh CC2652)
Adapter firmware version: 20210120

nbostrom, 2021-03-01:
I'm having the same problem with the same zigbee stick and docker-compose setup

CSchlipp, 2021-03-01:
Thanks for creating a new issue for this! 

There are several others reporting the on an older (closed) issue: https://github.com/Koenkk/zigbee2mqtt/issues/2997#issuecomment-788190244

m0wlheld, 2021-03-01:
Additional info: The reported error has been observed on a Raspberry Pi 3B+ with Slae'sh CC2652RB USB stick (on extension cable, of course). However, version 1.18.0 works fine on my 2nd system, a Raspberry Pi 4 with Raspbee II add-on card.

```
zigbee2mqtt   | Zigbee2MQTT:info  2021-03-01 23:18:39: Logging to console only'
zigbee2mqtt   | Zigbee2MQTT:info  2021-03-01 23:18:39: Starting Zigbee2MQTT version 1.18.0 (commit #3fd6619)
zigbee2mqtt   | Zigbee2MQTT:info  2021-03-01 23:18:39: Starting zigbee-herdsman (0.13.72)
zigbee2mqtt   | CREATED DECONZ ADAPTER
zigbee2mqtt   | Zigbee2MQTT:info  2021-03-01 23:18:39: zigbee-herdsman started
zigbee2mqtt   | Zigbee2MQTT:info  2021-03-01 23:18:39: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":105,"product":0,"revision":"0x26690700","transportrev":0},"type":"ConBee2/RaspBee2"}'
zigbee2mqtt   | Zigbee2MQTT:info  2021-03-01 23:18:39: Currently 40 devices are joined:
```` 

So it could be limited to CC2562RB, the zStack-Firmware or USB devices. Infact, most (if not all) of the older issue reports are related to CC2652R devices.

CSchlipp, 2021-03-01:
I'm on the same setting. 

For me, it broke with yesterdays release of the dev image, same for todays 1.18 release.
Rolling back to 1.17.1 always brought z2m back to life.

CSchlipp, 2021-03-01:
> @CSchlipp Did it work with -dev images before yesterday and you have the commit ID of the last working version?

Yes, it did. The last dev-image running successfully was: Zigbee2MQTT version 1.17.1-dev (commit #1ddd2ac) (latest-dev as of 2021-02-28 04:20:07)

The first version that threw this error was Zigbee2MQTT version 1.17.1-dev (commit #57165e3) (latest-dev as of 2021-03-01 04:18:20)

nbostrom, 2021-03-01:
Looks like zigbee-herdsman was updated which updated a dependency "serialport" (sounds like it could have something to do with this) one patch version. Looking into the changes in "serialport" if I can find anything that could have caused this...

shirou93, 2021-03-01:
same issue slae'sh CC2652

nbostrom, 2021-03-01:
I'm going to sleep. One thing to try would be to downgrade zigbee-herdsman to see if that solves the issue.

m0wlheld, 2021-03-02:
> Yes, it did. The last dev-image running successfully was: Zigbee2MQTT version 1.17.1-dev (commit #1ddd2ac) (latest-dev as of 2021-02-28 04:20:07)
>
> The first version that threw this error was [...]

That‘s 

> Looks like zigbee-herdsman was updated which updated a dependency "serialport" (sounds like it could have something to do with this) one patch version. Looking into the changes in "serialport" if I can find anything that could have caused this...

Can you link the commit that introduced the change?

Dylantje, 2021-03-02:
here also the same error..[ stupid to upgrade withouth looking here ]

Is there a way to downgrade?
Or hope @Koenkk  there will be a patch.

liambrugaro, 2021-03-02:
Hello everybody,

Same issue for me. I've downgraded with those commands :

cd /opt/zigbee2mqtt
git checkout 99274c7 (find version wanted here : https://github.com/Koenkk/zigbee2mqtt/tags)
npm ci --production
Npm start
And then z2m starts 😊

m0wlheld, 2021-03-02:
> git checkout [99274c7](https://github.com/Koenkk/zigbee2mqtt/commit/99274c72c26ba21347908af8238fa8dfdb4fb99d)

Could you check, if it still works with commit 1ddd2ac (the last working version as reported by @CSchlipp)?

d-EScape, 2021-03-02:
Had the same issue after upgrading tot 1.18. Using slae.sh CC2652 stick. Linux on intel. Works after reverting to previous version from backup.

f-bader, 2021-03-02:
Same issue with `zigbee2mqtt@1.18.0`.
Updated the coordinator to `CC2652RB_coordinator_20210120` firmware with no luck.
Rollback to `1.17.1` fixed the issue

liambrugaro, 2021-03-02:
> Could you check, if it still works with commit [1ddd2ac](https://github.com/Koenkk/zigbee2mqtt/commit/1ddd2ac30a8fde5e30cceab54924c4101c7a5e61) (the last working version as reported by @CSchlipp)?

I'm at work, i can check at the end of the day :)

oxivanisher, 2021-03-02:
I have the exact same problem. But only dockervavailable. @liambrugaro Can I test something with a docker based setup?

mrTuomoK, 2021-03-02:
> Could you check, if it still works with commit [1ddd2ac](https://github.com/Koenkk/zigbee2mqtt/commit/1ddd2ac30a8fde5e30cceab54924c4101c7a5e61) (the last working version as reported by @CSchlipp)?

I can confirm this works. @m0wlheld & @liambrugaro (Just tested)

wimb0, 2021-03-02:
I'm also having the same issue with 1.18.0 via Docker Compose.
Using Slaesh's CC2652RB stick on a Dell Optiplex 3060 Micro with Ubuntu 18.04LTS.

m0wlheld, 2021-03-02:
> > Could you check, if it still works with commit [1ddd2ac](https://github.com/Koenkk/zigbee2mqtt/commit/1ddd2ac30a8fde5e30cceab54924c4101c7a5e61) (the last working version as reported by @CSchlipp)?
> 
> I can confirm this DOES NOT work. @m0wlheld & @liambrugaro (Just tested)

Are you sure? In my case, it does work with commit 1ddd2ac. Here's what I used to test:

```bash
$ docker run --device /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_44_12-if00-port0:/dev/ttyACM0 --rm -ti node:latest /bin/bash
root@e1523ad5a16b:/# git clone https://github.com/Koenkk/zigbee2mqtt.git
root@e1523ad5a16b:/# cd zigbee2mqtt
root@e1523ad5a16b:/# npm ci
root@e1523ad5a16b:/# npm start
```

Result after last command:

```
> zigbee2mqtt@1.18.0 start
> node index.js

Zigbee2MQTT:info  2021-03-02 08:25:51: Logging to console and directory: '/zigbee2mqtt/data/log/2021-03-02.08-25-49' filename: log.txt
Zigbee2MQTT:info  2021-03-02 08:25:51: Starting Zigbee2MQTT version 1.18.0 (commit #3fd6619)
Zigbee2MQTT:info  2021-03-02 08:25:51: Starting zigbee-herdsman (0.13.72)
Zigbee2MQTT:error 2021-03-02 08:26:11: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-03-02 08:26:11: Failed to start zigbee
Zigbee2MQTT:error 2021-03-02 08:26:11: Exiting...
Zigbee2MQTT:error 2021-03-02 08:26:11: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code 1
npm ERR! path /zigbee2mqtt
npm ERR! command failed
npm ERR! command sh -c node index.js

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-03-02T08_26_11_751Z-debug.
```

Then I did a checkout on 1ddd2ac:

```bash
root@e1523ad5a16b:/zigbee2mqtt# git checkout 1ddd2ac
root@e1523ad5a16b:/zigbee2mqtt# npm ci
root@e1523ad5a16b:/zigbee2mqtt# npm start
```
Result after last command:

```
Zigbee2MQTT:info  2021-03-02 08:31:01: Logging to console and directory: '/zigbee2mqtt/data/log/2021-03-02.08-30-59' filename: log.txt
Zigbee2MQTT:info  2021-03-02 08:31:01: Starting Zigbee2MQTT version 1.17.1-dev (commit #1ddd2ac)
Zigbee2MQTT:info  2021-03-02 08:31:01: Starting zigbee-herdsman (0.13.71)
Zigbee2MQTT:info  2021-03-02 08:31:17: zigbee-herdsman started
Zigbee2MQTT:info  2021-03-02 08:31:17: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2021-03-02 08:31:17: Currently 0 devices are joined:
Zigbee2MQTT:warn  2021-03-02 08:31:17: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2021-03-02 08:31:17: Allowing new devices to join.
Zigbee2MQTT:warn  2021-03-02 08:31:17: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2021-03-02 08:31:17: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2021-03-02 08:31:18: Connecting to MQTT server at mqtt://localhost
```

mrTuomoK, 2021-03-02:
> > > Could you check, if it still works with commit [1ddd2ac](https://github.com/Koenkk/zigbee2mqtt/commit/1ddd2ac30a8fde5e30cceab54924c4101c7a5e61) (the last working version as reported by @CSchlipp)?
> > 
> > 
> > I can confirm this DOES NOT work. @m0wlheld & @liambrugaro (Just tested)
> 
> Are you sure? In my case, it does work with commit [1ddd2ac](https://github.com/Koenkk/zigbee2mqtt/commit/1ddd2ac30a8fde5e30cceab54924c4101c7a5e61). Here's what I used to test:
> 
> ```shell
> $ docker run --device /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_44_12-if00-port0:/dev/ttyACM0 --rm -ti node:latest /bin/bash
> root@e1523ad5a16b:/# git clone https://github.com/Koenkk/zigbee2mqtt.git
> root@e1523ad5a16b:/# cd zigbee2mqtt
> root@e1523ad5a16b:/# npm ci
> root@e1523ad5a16b:/# npm start
> ```
> 
> Result after last command:
> 
> ```
> > zigbee2mqtt@1.18.0 start
> > node index.js
> 
> Zigbee2MQTT:info  2021-03-02 08:25:51: Logging to console and directory: '/zigbee2mqtt/data/log/2021-03-02.08-25-49' filename: log.txt
> Zigbee2MQTT:info  2021-03-02 08:25:51: Starting Zigbee2MQTT version 1.18.0 (commit #3fd6619)
> Zigbee2MQTT:info  2021-03-02 08:25:51: Starting zigbee-herdsman (0.13.72)
> Zigbee2MQTT:error 2021-03-02 08:26:11: Error while starting zigbee-herdsman
> Zigbee2MQTT:error 2021-03-02 08:26:11: Failed to start zigbee
> Zigbee2MQTT:error 2021-03-02 08:26:11: Exiting...
> Zigbee2MQTT:error 2021-03-02 08:26:11: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
>     at ZStackAdapter.<anonymous> (/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
>     at Generator.throw (<anonymous>)
>     at rejected (/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
> npm ERR! code 1
> npm ERR! path /zigbee2mqtt
> npm ERR! command failed
> npm ERR! command sh -c node index.js
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     /root/.npm/_logs/2021-03-02T08_26_11_751Z-debug.
> ```
> 
> Then I did a checkout on [1ddd2ac](https://github.com/Koenkk/zigbee2mqtt/commit/1ddd2ac30a8fde5e30cceab54924c4101c7a5e61):
> 
> ```shell
> root@e1523ad5a16b:/zigbee2mqtt# git checkout 1ddd2ac
> root@e1523ad5a16b:/zigbee2mqtt# npm ci
> root@e1523ad5a16b:/zigbee2mqtt# npm start
> ```
> 
> Result after last command:
> 
> ```
> Zigbee2MQTT:info  2021-03-02 08:31:01: Logging to console and directory: '/zigbee2mqtt/data/log/2021-03-02.08-30-59' filename: log.txt
> Zigbee2MQTT:info  2021-03-02 08:31:01: Starting Zigbee2MQTT version 1.17.1-dev (commit #1ddd2ac)
> Zigbee2MQTT:info  2021-03-02 08:31:01: Starting zigbee-herdsman (0.13.71)
> Zigbee2MQTT:info  2021-03-02 08:31:17: zigbee-herdsman started
> Zigbee2MQTT:info  2021-03-02 08:31:17: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'
> Zigbee2MQTT:info  2021-03-02 08:31:17: Currently 0 devices are joined:
> Zigbee2MQTT:warn  2021-03-02 08:31:17: `permit_join` set to  `true` in configuration.yaml.
> Zigbee2MQTT:warn  2021-03-02 08:31:17: Allowing new devices to join.
> Zigbee2MQTT:warn  2021-03-02 08:31:17: Set `permit_join` to `false` once you joined all devices.
> Zigbee2MQTT:info  2021-03-02 08:31:17: Zigbee: allowing new devices to join.
> Zigbee2MQTT:info  2021-03-02 08:31:18: Connecting to MQTT server at mqtt://localhost
> ```

It does startup but when I tried to control my devices they did not respond. (Can't debug more right now, working)

CSchlipp, 2021-03-02:
> > > git checkout [99274c7](https://github.com/Koenkk/zigbee2mqtt/commit/99274c72c26ba21347908af8238fa8dfdb4fb99d)
> > 
> > 
> > Could you check, if it still works with commit [1ddd2ac](https://github.com/Koenkk/zigbee2mqtt/commit/1ddd2ac30a8fde5e30cceab54924c4101c7a5e61) (the last working version as reported by @CSchlipp)?
> 
> I can confirm this DOES NOT work. @m0wlheld & @liambrugaro (Just tested)
> 
> It does startup but things don't work properly
> 
> If it's for any help I'm also using Slaesh's CC2652RB --> usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_41_34-if00-port0

I've double checked the logs again, but the docker image build from this commit was definitely running smoothly for 24h, until the next update broke it.

CSchlipp, 2021-03-02:
> It does startup but when I tried to control my devices they did not respond. (Can't debug more right now, working)
Note the log line "Currently 0 devices are joined", that's probably the reason why none of your devices responded ;)

mrTuomoK, 2021-03-02:
> > It does startup but when I tried to control my devices they did not respond. (Can't debug more right now, working)
> 
> Note the log line "Currently 0 devices are joined", that's probably the reason why none of your devices responded ;)

Yeah in this case, dunno if it was like that on my case :D

m0wlheld, 2021-03-02:
Did more checks:
* 9d835ff does not work
* 744dd47 works
* fa5a7b1 works
* 1ddd2ac works

So the issue has been introduced with commit 9d835ff and therefore with update of zigbee-herdsman to 0.13.72

arnaud-gouder, 2021-03-02:
I have the same problem (accidentally created a duplicate issue #6522)

m0wlheld, 2021-03-02:
Not a Node developer, but the only [change introduced](https://github.com/Koenkk/zigbee-herdsman/compare/v0.13.71...v0.13.72#diff-7ae45ad102eab3b6d7e7896acd08c427a9b25b346470d7bc6507b6481575d519) by zigbee-herdsman-0.13.72 was a dependency update from [serialport](https://github.com/serialport/node-serialport) 9.0.6 to serialport 9.0.7

Cannot identify *the* change that is relevant, though.

Mark-Sellwood, 2021-03-02:
@slaesh Do you have any idea what the isue is with your stick & serialport 9.0.7?

nbostrom, 2021-03-02:
I'm not able to test this atm, but a theory is that since slaesh recommends using the by-id path to get the serial device we all seem to have that setup in common. This is a good thing to do generally, but I would try changing the path to `/dev/ttyUSBx` (where x is an integer). Just to see if it solves the problem.

oxivanisher, 2021-03-02:
@nbostrom I the following settings in my docker-compose without any luck:
* /dev/ttyUSB0:/dev/ttyUSB0
* /dev/ttyACM0:/dev/ttyUSB0
* /dev/ttyUSB0:/dev/ttyACM0
* /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_01_2D-if00-port0:/dev/ttyUSB0                                   
* /dev/ttyUSB0:/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_01_2D-if00-port0

nbostrom, 2021-03-02:
@oxivanisher Alright. Thanks!

nbostrom, 2021-03-02:
These are the changes made in "serialport" for 9.0.7 btw:
https://github.com/serialport/node-serialport/compare/@serialport/bindings@9.0.7...master

goodwin, 2021-03-02:
@nbostrom these are changes made after the 9.0.7 release. difference between 9.06 and 9.0.7 is this:
https://github.com/serialport/node-serialport/compare/v9.0.6...v9.0.7

nbostrom, 2021-03-02:
@goodwin Thanks! I noticed my error as well, but you were quicker with the edits ;)

goodwin, 2021-03-02:
From noticeable change, I see only the addition of low-latency mode... which is disabled by default, so unsure if that may be a reason (on other hand - there is additional code execution even with async=off, so who knows...)

m0wlheld, 2021-03-02:
Testing bricked my CC2652RB stick completly, it's giving me the "Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel" message now. 

No way to revive, neither reflash nor reset button did help.

goodwin, 2021-03-02:
what is even more strange - I can't find where is ASYNC_LOW_LATENCY definition is coming from... it is used in new method linuxSetLowLatencyMode (that is executed always), but on other place there is check if this definition exists at all (but not inside of linuxSetLowLatencyMode). it is not defined in the project itself, and it is not in the linux/serial.h ... I'm not expert, but that looks suspicious...

goodwin, 2021-03-02:
btw, using the latest z2m on mac with 2652P - works fine. with the same dongle connected to my prod box with Linux - doesn't work. So, I'm almost sure that breaking change is related to that low-latency mode in the serialport (it is linux-specific change)

goodwin, 2021-03-02:
@Koenkk  as of now, I think the best option would be to roll-back dependency change in the Herdsman for SerialPort, as this breaks a lot of setups for ppl with 2652 + linux

nbostrom, 2021-03-02:
> Testing bricked my CC2652RB stick completly, it's giving me the "Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel" message now.
> 
> No way to revive, neither reflash nor reset button did help.

Oh no! :(
What exactly did you test when the stick became bricked?

shirou93, 2021-03-02:
> Testing bricked my CC2652RB stick completly, it's giving me the "Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel" message now.
> 
> No way to revive, neither reflash nor reset button did help.

Try disconnect antenna and start without it.

m0wlheld, 2021-03-02:
> Try disconnect antenna and start without it.

You're killing me, that worked! What kind of sorcery is this?

fdaemon, 2021-03-02:
> Testing bricked my CC2652RB stick completly, it's giving me the "Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel" message now.
> 
> No way to revive, neither reflash nor reset button did help.

Try turning off router devices. In my setup the IKEA bulbs took over the channel/panID while the coordinator was offline. Had to turn off all IKEA bulbs, then start the coordinator.

m0wlheld, 2021-03-02:
> You're killing me, that worked! What kind of sorcery is this?

Okay, worked once and only, while the coordinator was away from next routing device. Going to power down routers next and try again.

m0wlheld, 2021-03-02:
> Going to power down routers next and try again.

After powering down all router devices within reach of the antenna-less coordinator, Zigbee2MQTT (1.17.1 for now) starts again. 

Having re-attached the antenna and power back the routers, no devices are addressable. Am I expected to re-pair them?

shirou93, 2021-03-02:
> > Going to power down routers next and try again.
> 
> After powering down all router devices within reach of the antenna-less coordinator, Zigbee2MQTT (1.17.1 for now) starts again.
> 
> Having re-attached the antenna and power back the routers, no devices are addressable. Am I expected to re-pair them?

Devices should be appear.

frogale, 2021-03-02:
Same here.. Slaesh and RPI4 with docker image 1.18.0...

station7, 2021-03-02:
> > Going to power down routers next and try again.
> 
> After powering down all router devices within reach of the antenna-less coordinator, Zigbee2MQTT (1.17.1 for now) starts again.
> 
> Having re-attached the antenna and power back the routers, no devices are addressable. Am I expected to re-pair them?

@m0wlheld If I understood your current idea correctly, you are trying to use a spare device (CC2652RB) instead of slaesh's stick, which currently is not working due to that serialport-upgrade. I do not think that this is possible because of the PAN ID of the zigbee-network. Replacing a coordinator needs a new PAN ID according to my understanding, which needs repairing of all devices. But this is a different problem and not resulting form the 1.18.0/serialport-upgrade.

m0wlheld, 2021-03-02:
> @m0wlheld If I understood your current idea correctly, you are trying to use a spare device (CC2652RB) instead of slaesh's stick, which currently is not working due to that serialport-upgrade

No. I tested various Z2M versions / commit with my only coordinator device (slaesh's CC2652RB) to identify the "faulty" [commit](https://github.com/Koenkk/zigbee2mqtt/commit/9d835ff9dcc763406d4b409bd77e5083865c1bcf). This made my slaesh's stick unresponsive, even after returning to 1.17.1 . In desperate need to fix this I asked for help in this issue thread, that was indeed unrelated. 

The situation that Zigbee2MQTT versions after commit 9d835ff do not work with slaesh's stick remains unresolved.

(Btw: Got mine back to life by starting with router devices powered off, antenna unscrewed and database backup, thanks for helping)

Dylantje, 2021-03-02:
> Hello everybody,
>
> Same issue for me. I've downgraded with those commands :
>
> cd /opt/zigbee2mqtt
> git checkout [...]

Thanks works again.. only no Npm just npm

goodwin, 2021-03-02:
> Hello everybody,
>
> Same issue for me. I've downgraded with those commands :
>
> cd /opt/zigbee2mqtt
> git checkout [...]

Unfortunately, not an option for Hassio users... Supervisor does not have any option to downgrade, only upgrade )))

IngoHenning, 2021-03-02:
Same issue for me.
Docker with Slaesh on an NUC with Ubuntu. 
I will revert to 1.17.1 and wait for a fix

Burki24, 2021-03-02:
Hi all,

also the same Issue. Slae-Stick Z2M1.18 RPI4.
Reverted back to 1.17.1 all runs fine again.
Thanks for the tips.

Koenkk, 2021-03-02:
Please check if this is fixed in the latest dev (will be available in 1 hour from now). If yes I will create a hotfix release (1.18.1)

m0wlheld, 2021-03-02:
> Please check if this is fixed in the latest dev (will be available in 1 hour from now). If yes I will create a hotfix release (1.18.1)

Thanks, will do. Do you have a CC2652RB stick for tests? Otherwise I would love to sponsor one or two ...

paulz-gh, 2021-03-02:
Same issue using zzh stick.
After rollback to 1.17.1 still the same error so stick seems to be bricked.
Any idea on how to get the stick back to life ?

m0wlheld, 2021-03-02:
>  Any idea on how to get the stick back to life ?

In my case:
* unscrewed the antenna
* powered of any routing device (bulbs, socket) in close range
* restart
* screw antenna back on
* power on devices

PedroQ, 2021-03-02:
Tried the latest edge version of the addon (checked the commit, it matches the latest in dev ([d77231b](https://github.com/Koenkk/zigbee2mqtt/commit/d77231b99fee15ddc5cb403bb4fbc5ab4ee1db16))) and I still get the error.

> Zigbee2MQTT:info  2021-03-02 19:25:23: Logging to console and directory: '/config/zigbee2mqtt-edge/log/2021-03-02.19-25-20' filename: log.txt
> Zigbee2MQTT:info  2021-03-02 19:25:23: Starting Zigbee2MQTT version 1.17.1-dev (commit #d77231b)
> Zigbee2MQTT:info  2021-03-02 19:25:23: Starting zigbee-herdsman (0.13.73)
> Zigbee2MQTT:error 2021-03-02 19:25:44: Error while starting zigbee-herdsman
> Zigbee2MQTT:error 2021-03-02 19:25:44: Failed to start zigbee
> Zigbee2MQTT:error 2021-03-02 19:25:44: Exiting...
> Zigbee2MQTT:error 2021-03-02 19:25:44: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
>     at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
>     at Generator.throw (<anonymous>)
>     at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)

Koenkk, 2021-03-02:
@PedroQ you are not on the latest edge (`Zigbee2MQTT:info 2021-03-02 19:25:23: Starting Zigbee2MQTT version 1.17.1-dev (commit #d77231b)`). Should say `1.18.0-dev`. Try uninstalling and installing

m0wlheld, 2021-03-02:
But commit id d77231b matches latest dev commit?

Koenkk, 2021-03-02:
Ah I see, error on my side indeed. Can others also confirm the fix does not work?

[33 more replies]

closed 2021-03-02 by Koenkk