Koenkk/zigbee2mqtt #6522: Version 1.18.0 not working with Slaesh's CC2652RB stick

arnaud-gouder, 2021-03-02:
## What happened

The upgrade to 1.18.0 broke the Slaesh's CC2652RB functionality. I already explained this in https://github.com/Koenkk/zigbee2mqtt/issues/5854, but think it might be better to have a separate issue for this specific issue since #5854 is about all kind of different problems (like people using the wrong serial device etc.)

## What did you expect to happen

Zigbee2mqtt should start/work

## How to reproduce it (minimal and precise)

Upgrade from 1.17.1 to 1.18.0

## Debug info
Zigbee2MQTT version: 1.18.0
Adapter hardware: CC2652RB
Adapter firmware version: does not matter (tried both Koen's and Slaesh's)

Log:
'''
Zigbee2MQTT:info  2021-02-28 19:44:41: Logging to console and directory: '/app/data/log/2021-02-28.19-44-41' filename: log.txt,
Zigbee2MQTT:debug 2021-02-28 19:44:41: Removing old log directory '/app/data/log/2021-02-28.19-31-19',
Zigbee2MQTT:info  2021-02-28 19:44:41: Starting Zigbee2MQTT version 1.17.1 (commit #99274c7),
Zigbee2MQTT:info  2021-02-28 19:44:41: Starting zigbee-herdsman (0.13.59),
Zigbee2MQTT:debug 2021-02-28 19:44:41: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6757},"serialPort":{"path":"/dev/zigbee2mqtt"}}',
2021-02-28T18:44:41.721Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[182,210,165,214,37,2,179,116,214,90,240,168,82,145,32,66],"panID":6757,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/zigbee2mqtt"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}',
2021-02-28T18:44:41.722Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/zigbee2mqtt and {"baudRate":115200,"rtscts":false,"autoOpen":false},
2021-02-28T18:44:41.724Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened,
2021-02-28T18:44:41.733Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0],
2021-02-28T18:44:41.733Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0],
2021-02-28T18:44:42.037Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0],
2021-02-28T18:44:42.038Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0,0],
2021-02-28T18:44:42.179Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload,
2021-02-28T18:44:42.180Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239],
2021-02-28T18:44:43.181Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1},
2021-02-28T18:44:43.182Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32],
2021-02-28T18:44:49.186Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1},
2021-02-28T18:44:49.186Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32],
2021-02-28T18:44:55.190Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1},
2021-02-28T18:44:55.190Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32],
Zigbee2MQTT:error 2021-02-28 19:45:01: Error while starting zigbee-herdsman,
Zigbee2MQTT:error 2021-02-28 19:45:01: Failed to start zigbee,
Zigbee2MQTT:error 2021-02-28 19:45:01: Exiting...,
Zigbee2MQTT:error 2021-02-28 19:45:01: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms),
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31),
    at Generator.throw (<anonymous>),
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65),
npm ERR! code ELIFECYCLE,
npm ERR! errno 1,
npm ERR! zigbee2mqtt@1.17.1 start: `node index.js`,
npm ERR! Exit status 1,
'''

m0wlheld, 2021-03-02:
Duplicate of #6517. Please close, so we can keep the issue in one place.

arnaud-gouder, 2021-03-02:
Indeed a duplicate of #6517.

closed 2021-03-02 by arnaud-gouder