Koenkk/zigbee2mqtt #6524: Support for Aqara smart wall switch (no neutral, double rocker) - [lumi.switch.l2aeu1]

tavalin, 2021-03-02:
## Information about the device + link

[Link to the switch](https://smarterhome.sk/en/xiaomi-aqara/zigbee-wallswitch-with-double-relay-aqara-smart-wall-switch-h1-eu-no-neutral-double-rocker-ws-euk02-1001.html)

This device is an EU version of [this US one](https://www.aqara.com/en/smart_switch_no_neutral.html) but has a `left` and `right` button instead of `top` and `bottom`

I copied the device section from the US version (lumi.switch.b2laus01) and changed the model and zigbee model accordingly but I've noticed that although I can control the left and right buttons from zigbee2mqtt, the reporting doesn't work i.e. if the switch is manually toggled the state doesn't get updated in zigbee2mqtt.

See section from `devices.js` below - any help appreciated

```
    {
        zigbeeModel: ['lumi.switch.l2aeu1'],
        model: 'WS-EUK02',
        vendor: 'Xiaomi',
        description: 'Aqara smart wall switch (no neutral, double rocker)',
  	extend: preset.switch(),
	exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right')],
        meta: {multiEndpoint: true, configureKey: 1},
        endpoint: (device) => {
            return {'left': 1, 'right': 2};
        },
	configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint1);
            const endpoint2 = device.getEndpoint(2);
            await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint2);
        },
    }
```

## data/database.db entry of the device
`{"id":12,"type":"EndDevice","ieeeAddr":"0x54ef4410000d0438","nwkAddr":21323,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.l2aeu1","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,9],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.l2aeu1","manufacturerName":"LUMI","powerSource":1,"zclVersion":3,"appVersion":11,"stackVersion":2,"hwVersion":1,"dateCode":"01-12-2021"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002160067e","endpointID":1},{"cluster":2,"type":"endpoint","deviceIeeeAddress":"0x00124b002160067e","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b002160067e","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002160067e","endpointID":1},{"cluster":2,"type":"endpoint","deviceIeeeAddress":"0x00124b002160067e","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002160067e","endpointID":1},{"cluster":2,"type":"endpoint","deviceIeeeAddress":"0x00124b002160067e","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":11,"stackVersion":2,"hwVersion":1,"dateCode":"01-12-2021","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1614682069845}`

piotrbielaska, 2021-03-02:
Hi. I'm looking for a way to support similar device but with neutral wire. It is lumi.switch.n2aeu1. Could you explain where did you find and coppied configuration for us version? I'm using Home Assistant with Z2MQTT as addon.

tavalin, 2021-03-02:
@piotrbielaska [this link](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html) explains about adding a new device.  The file you need to edit is called `devices.js`

tavalin, 2021-03-04:
@Koenkk are you able to provide any direction please? Is that enough information?

piotrbielaska, 2021-03-04:
I had problems using converters and decided to wait until this device will be supported. Description on how to add new device in Home Assistant Add-On is very vague.

Koenkk, 2021-03-04:
Do you also not get updates of the left switch?

tavalin, 2021-03-04:
> Do you also not get updates of the left switch?

I did get some reports from the left switch but didn't test thoroughly yet. Definitely none from the right.  

What does the SOFTWARE_FAILURE mean when I manually tried to set up the bind/report on the right switch?

Koenkk, 2021-03-06:
I don't know what it exactly means, but this is reported by the device itself (doesn't come from z2m). I suggest sniffing the traffic with the Xiaomi gateway to see how it has to be configured: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

tavalin, 2021-03-06:
@Koenkk I've flashed my spare CC2531 as a sniffer and I've got zboss and Wireshark running on Windows 10.  I've added the Trust Center key as well as my z2m network key (obtained from Wireshark by re-adding a device) but Wireshark still can't decode/decrypt the payloads.  

Am I missing something?

Koenkk, 2021-03-06:
You need to obtain the Xiaomi gateway key, if you can provide a sniff of the pairing with the Xiaomi gateway I can grab the key from it.

tavalin, 2021-03-06:
Is there a known channel I should be sniffing for the gateway?

Koenkk, 2021-03-07:
The gateway is probably on channel 11, 15, 20 or 25. If Z2M is also on that channel I recommend to turn of all devices on the z2m network.

tavalin, 2021-06-04:
Hi @Koenkk 

Can you look through this wireshark capture and see if you can help with supporting this device?

PS I am also beta testing a new Aqara Roller Shade cord controller and am having similiar issues.

[l2aeu1.zip](https://github.com/Koenkk/zigbee2mqtt/files/6599919/l2aeu1.zip)

Koenkk, 2021-06-04:
@tavalin could you provide the network key, this should be revealed when starting the sniffing before pairing the device. And when used with the original gateway, can you click the right button, does the state on the xiaomi gateway update immediately?

blakadder, 2021-06-04:
I've reported the missing status report to Aqara. 

The status on the gateway has a noticeable delay for the right rocker compared to the left one. The assumption is Aqara is polling states.

Koenkk, 2021-06-04:
@blakadder 

> The status on the gateway has a noticeable delay for the right rocker compared to the left one. The assumption is Aqara is polling states.

Can you confirm this by taking a sniff? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

tavalin, 2021-06-04:
> @tavalin could you provide the network key, this should be revealed when starting the sniffing before pairing the device. And when used with the original gateway, can you click the right button, does the state on the xiaomi gateway update immediately?

I am using the new [Aqara E1 Hub](https://homekitnews.com/2021/05/20/aqara-e1-hub-first-look/) as I'm testing it also.  I no longer have the old Xiaomi hub anymore - i'm not sure if that makes a difference but the new E1 Hub is also Zigbee 3.0.

Sorry if i'm being stupid but can you explain how I can get the network key from the Aqara hub? Is it a just a case of running the sniffer for a while or do I need to do something apart from pairing a device? The [how to guide](https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html) describes pairing a device with z2m to get the network key

Koenkk, 2021-06-05:
@tavalin do you have another device to pair? The key can be found in "Transport key" message and should come after the "Association response", I don't know why but maybe because the device has already been paired before.

blakadder, 2021-06-06:
> @blakadder
> 
> > The status on the gateway has a noticeable delay for the right rocker compared to the left one. The assumption is Aqara is polling states.
> 
> Can you confirm this by taking a sniff? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Sadly I managed to brick my G2H gateway in the meantime and I don't have any alternatives that support that device

tavalin, 2021-06-09:
@blakadder have you had a response from Aqara? How did you 

> I've reported the missing status report to Aqara.
> 
> The status on the gateway has a noticeable delay for the right rocker compared to the left one. The assumption is Aqara is polling states.

This is consistent with my experience using the Aqara hub. @blakadder how did you raise it with Aqara? Did you get a response yet? 

If you don't get a response I have a contact I try. 

@Koenkk I will try do more sniffing over the coming weekend with some new devices that haven't been paired before.

Koenkk, 2021-06-11:
@tavalin 

> I've also captured another Wireshark and it seems the right button is sending a packet when manually toggled at the switch. my new hypothesis is that z2m doesn't understand the ZigBee packet 

Can you check if this same message is send when the switch is paired with z2m?

tavalin, 2021-06-11:
> @tavalin
> 
> > I've also captured another Wireshark and it seems the right button is sending a packet when manually toggled at the switch. my new hypothesis is that z2m doesn't understand the ZigBee packet
> 
> Can you check if this same message is send when the switch is paired with z2m?

I'm away from the device again at the moment but the packet I'm referring to is a "Report attributes" that's triggered for both - however the structure of this is different for the left and right buttons

tavalin, 2021-06-13:
@Koenkk I've had a further look into the logs when the device is paired with the Aqara hub and as far as I can see, the Aqara hub does not send a command to configure reporting on the switch. The Aqara Hub does send a manufacturer specific command which I can't seem to replicate through the z2m UI at the moment.

Koenkk, 2021-06-14:
> The Aqara Hub does send a manufacturer specific command which I can't seem to replicate through the z2m UI at the moment.

Which command do you mean, I cannot find this in "aqara hub transport key.zip"

tavalin, 2021-06-14:
@Koenkk here is another log from the Aqara hub - the transport key has changed as I had to factory reset the hub - it's now `96:87:76:ad:fb:c5:de:64:0b:f5:16:e1:bf:fe:8e:0e`

The following log shows the switch joining the Aqara hub network and me toggling the right switch manually, resulting in reports back to the Aqara hub.

[l2aeu1-join and toggle.zip](https://github.com/Koenkk/zigbee2mqtt/files/6651466/l2aeu1-join.and.toggle.zip)

What I am unable to see if the logs is what exactly causes the switch to send these reports. During the join phase the hub sends a manufacturer specific `write attributes` to the switch but the payload seems to change every time I reset the network.  I also can't see any setting up of the binding/reporting  in the logs- is this encrypted with another key?

Here's an example of the `write attributes` from the hub to the switch

[image]

I'm running out of ideas on how to work out exactly what's happening between the switch and the Aqara hub

Koenkk, 2021-06-15:
I've seen this exact message also for other Xiaomi devices and it indeed changes everytime. Looks that they somehow encrypted it, but I currently don't know how to reverse engineer this.

antst, 2021-07-01:
Thus one is nearly the same as QBKG39LM
Copying configuration from QBKG39LM gives full functionality.

Koenkk, 2021-07-01:
@antst cool, could you make a pr?

k-szabi, 2021-07-01:
Hi @antst,
Thanks for the info. The status reporting works fine for both button now.

antst, 2021-07-01:
Decoupling also works.

antst, 2021-07-01:
> @antst cool, could you make a pr?

I will try. But can not make guarantee it is complete and fully working. Particularly, I think that there might be required change for "power_outage_memory" and unsure how properly multi-button presses are handled.

closed 2021-08-08 by github-actions[bot]

hoclengchung1991, 2021-11-19:
@antst 
Still no PR?

adipierro, 2022-02-19:
Uncoupled button actions stopped working for some reason more than a month ago – I have auto-update of z2m on hassos, so I don't know exact zigbee2mqtt version on which the issue started. Double rocker and single rocker models are both affected.

No lines in log appear on button click, no state changes are published.

HemiBob, 2022-05-25:
I'm new to zigbee2mqtt and found none of my Aqara wall switches is supported.
Since these are now being sold by Amazon UK can I look forward to this thread and others ending up with a working solution?
Following the "Support new devices" is beyond me.

adynis, 2022-09-08:
I have WS-EUK04 (Aqara smart wall switch H1 EU (with neutral, double rocker)) (lumi.switch.n2aeu1) and I have the same issue: it does NOT report anything on mqtt. I'm looking also with mqtt explorer and the physical buttons do not report anything :( 
Is there any solution as of today ? Why people do not complain... do other aqara switches work and only some of them do not work ?

adynis, 2022-09-08:
I have WS-EUK04 (Aqara smart wall switch H1 EU (with neutral, double rocker)) (lumi.switch.n2aeu1) and I was having the same issue: It was not reporting anything to my Home Assistant with zigbee2mqtt.
I just wrote a comment hear 2 hours ago, then deleted because something happend:
After Adding the devices to ZHA (stopped my zigbee2mqtt, started ZHA, it asked me something about the network of the zigbee, I selected "keep the same"), after that I observed that the device IS WORKING with ZHA, then I tried one more time in zigbee2mqtt (disabled ZHA, re-started zigbee2mqtt) and now it works.... 
I have no idea why .... but hopefully I will not have this issue anymore. (Funny is that I had EXACTLY the same thing with a gledopto led controller, and after adding it to ZHA and then back to zigbe2mqtt it worked .....)

pipiche38, 2023-02-06:
> e something about the network of the zigbee, I selected "keep the same"), after that I observed that the device IS WORKING with ZHA, then I tried one more time in zigbee2mqtt (disabled ZHA, re-started zigbee2mqtt) and now it works....

Most-likely zigbee2mqtt does some magic that ZHA is not