Koenkk/zigbee2mqtt #6525: After Zigbee2MQTT Update 1.18.0-1 does not start AND configuration cannot be changed.

attkri, 2021-03-02:
Hi,

I have a similar problem as #6513 but its solution does not help me. 

According to #6513 and the minimal requirements (https://www.zigbee2mqtt.io/information/configuration.html) my configuration should look like this:

```yaml
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost:1883'
serial:
  port: /dev/ttyACM0
ota:
  update_check_interval: 10
  disable_automatic_update_check: false
```
I get the following error:

Failed to save addon configuration, not a valid value for dictionary value @ data['options']. Got {'mqtt': {'base_topic': 'zigbee2mqtt', 'server': 'mqtt://localhost:1883'}, 'serial': {'port': '/dev/ttyACM0'}, 'ota': {'update_check_interval': 10, 'disable_automatic_update_check': False}}

This also happens with the 'Reset to Default' setting and also with my previous ones. Basically I can't enter ota because I always get this error when saving.

Anyone have an idea?

m0wlheld, 2021-03-02:
> Failed to save addon configuration

This is an issue with the Zigbee2MQTT add-on of Home Assistant, you should file an issue [there](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt). Actually, multiple people did already ...

attkri, 2021-03-02:
ok, I will look for a solution there...

closed 2021-03-02 by attkri