Koenkk/zigbee2mqtt #6526: TRADFRI shortcut button Exposes Action N/A

chr55, 2021-03-02:
## What happened
I have purchased the new IKEA Tradfri Shortcut button. I was using deconz/phoscon however this new shortcut device was not recognized. Since it is supported in zigbee2mqtt I decided to try this Home Assistant add-on. I installed the add-on and it is running. I include the IKEA Shortcut button and it appears to be added successfully. I am not able to see an action event for the push button when I watch the MQTT messages. 

## What did you expect to happen
After the interview is completed the Devices / Exposes tab shows "N/A" for the action. I am able to see the battery level and link quality in MQTT message. I don't see action.

## How to reproduce it (minimal and precise)
I can remove the device and try again with the same result. 

## Debug info
Zigbee2MQTT version: 1.18.0-1
Adapter hardware: Conbee
Adapter firmware version: Unknown

[image]

[image]

[image]

chr55, 2021-03-02:
[image]

chr55, 2021-03-02:
[image]

chr55, 2021-03-02:
[image]

chr55, 2021-03-02:
[image]

Koenkk, 2021-03-02:
Can you provide the debug log when pressing the button?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

chr55, 2021-03-02:
I updated the log level to debug and restarted. When I push the Ikea button nothing comes in the log. 

I paste below the last three messages in the debug log from when it was restarted. Again, nothing in log when I push the button. Thanks for the help!

```Zigbee2MQTT:debug 2021-03-02 13:26:13: Received MQTT message on 'homeassistant/sensor/0x588e81fffe3ef20b/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x588e81fffe3ef20b"],"manufacturer":"IKEA","model":"TRADFRI shortcut button (E1812)","name":"0x588e81fffe3ef20b","sw_version":"Zigbee2MQTT 1.18.0"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x588e81fffe3ef20b","name":"0x588e81fffe3ef20b linkquality","state_topic":"zigbee2mqtt/0x588e81fffe3ef20b","unique_id":"0x588e81fffe3ef20b_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:debug 2021-03-02 13:26:13: Received MQTT message on 'homeassistant/sensor/0x588e81fffe3ef20b/update_state/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x588e81fffe3ef20b"],"manufacturer":"IKEA","model":"TRADFRI shortcut button (E1812)","name":"0x588e81fffe3ef20b","sw_version":"Zigbee2MQTT 1.18.0"},"icon":"mdi:update","json_attributes_topic":"zigbee2mqtt/0x588e81fffe3ef20b","name":"0x588e81fffe3ef20b update state","state_topic":"zigbee2mqtt/0x588e81fffe3ef20b","unique_id":"0x588e81fffe3ef20b_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
Zigbee2MQTT:debug 2021-03-02 13:26:13: Received MQTT message on 'homeassistant/binary_sensor/0x588e81fffe3ef20b/update_available/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x588e81fffe3ef20b"],"manufacturer":"IKEA","model":"TRADFRI shortcut button (E1812)","name":"0x588e81fffe3ef20b","sw_version":"Zigbee2MQTT 1.18.0"},"json_attributes_topic":"zigbee2mqtt/0x588e81fffe3ef20b","name":"0x588e81fffe3ef20b update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x588e81fffe3ef20b","unique_id":"0x588e81fffe3ef20b_update_available_zigbee2mqtt","value_template":"{{ value_json.update_available}}"}'```

Koenkk, 2021-03-02:
Could be a conbee specific thing. Do you maybe have a CC2531 sniffer to debug this further? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

chr55, 2021-03-02:
I don't have a sniffer I'm afraid. My production Home Assistant is using a Conbee2 with deconz/phoscon. Since the Ikea shortcut buttons were not working there I decided to try zigbee2mqtt using my old Conbee (1) usb stick. I am happy to purchase a new usb device. Would this be suitable?

[image]

Koenkk, 2021-03-02:
You also need other hardware (https://www.zigbee2mqtt.io/information/flashing_the_cc2531.html#required-hardware) for sniffing. If you want to use a new adapter for zigbee2mqtt see https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html#supported-zigbee-adapter

chr55, 2021-03-02:
I have decided to return the Tradfri Shortcut Buttons to Ikea and purchase more of the Wireless dimmer instead. They are the same price and I already have some of the Wireless dimmers working in deconz/phoscon. Even though there is an on and an off button I have them coded that both sides perform the same action, basically like the shortcut button. 

I do appreciate the assistance. 

[image]

ttopholm, 2021-03-03:
I have the exact same issue with the dimmer. It pairs fine, but I don't get any log entries when I click the buttons.

chr55, 2021-03-03:
I had the actual Tradfri Gateway in storage which I brought out for testing. I can confirm that the real Ikea Tradfri Gateway is able to pair to the shortcut button and I was able to trigger a scene with the shortcut button. I was happy to confirm that the shortcut button was not broken. I still plan to return the shortcut buttons to Ikea and get the other kind. It is not worth me getting frustrated with this. I'll monitor this thread and revist these devices in the future if these initial problems are resolved.

Koenkk, 2021-03-03:
@ttopholm what adapter are you using?

ttopholm, 2021-03-03:
@Koenkk the cc2531 from Sonoff, with default firmware.

Koenkk, 2021-03-03:
@ttopholm what firmware version? https://www.zigbee2mqtt.io/information/flashing_the_cc2531.html#how-to-check-the-installed-firmware-version

ttopholm, 2021-03-03:
@Koenkk 

Zigbee2MQTT:info  2021-03-02 16:33:46: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20180507,"transportrev":2},"type":"zStack12"}' 

So 20180507.

I have ordered the cc-debugger and cable (coming from alixpress, so it will take a while) .. so I'll properly update the firmware and see if it helps...

Koenkk, 2021-03-04:
@ttopholm I expect the outdated firmware to be the issue indeed.

I've also ordered this device, will be delivered in 2 weeks from now.

ioanfesteu, 2021-03-17:
I have recently flashed the latest coordinator firmware onto a CC2530 module and I can confirm that the Ikea dimmer button is working as expected showing it's action status in log section of z2m:
`Received Zigbee message from 'Dimmer', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 901
Info MQTT publish: topic 'zigbee2mqtt/Dimmer', payload '{"action":"on","battery":60,"click":"on","elapsed":1989,"last_seen":"2021-03-17T10:25:11.947Z","linkquality":115,"update":{"state":"idle"},"update_available":false}'`
Otherwise if I pair it back in HASS at which I have connected a CC2531 usb stick is showing N/A again. I have to update its firmware too.

Koenkk, 2021-03-26:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/6526

closed 2021-03-26 by Koenkk

ttopholm, 2021-03-26:
It helped flashing it with the newest firmware.

gerardlouw, 2023-02-26:
I had the same problem using a [E1743 TRADFRI ON/OFF switch](https://www.zigbee2mqtt.io/devices/E1743.html#ikea-e1743) on the latest firmware version 2.3.079 with a [SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P](https://www.zigbee2mqtt.io/guide/adapters/#recommended:~:text=SONOFF%20Zigbee%203.0%20USB%20Dongle%20Plus%20ZBDongle%2DP) on an older Z-Stack 3.x.0 firmware version 20210708.

I don't have a debugger, so I couldn't flash newer firmware, but I found an alternative solution so I'm sharing it here in case anyone in the same situation stumbles upon this issue like I did.

What worked for me, was to explicitly bind the switch with the coordinator on clusters `OnOff` and `LevelCtrl` (by default, it was only bound on `genPollCtrl` and `PowerCfg` after interviewing). After doing so, it immediately started publishing action events.