Koenkk/zigbee2mqtt #6536: TS0219 siren whitelabel WOOX R7051

AntonioDiGennaro, 2021-03-02:
## Information about the device + link
https://wooxhome.com/woox-r7051-smart-indoor-siren-p49

## data/database.db entry of the device
{"id":4,"type":"Router","ieeeAddr":"0xbc33acfffe17d8c4","nwkAddr":49510,"manufId":0,"manufName":"_TYZB01_ynsiasng","powerSource":"Emergency mains and transfer switch","modelId":"TS0219","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,1,3,4,5,1280,1282,61183],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0219","manufacturerName":"_TYZB01_ynsiasng","powerSource":6,"zclVersion":3,"appVersion":74,"stackVersion":2,"hwVersion":2,"dateCode":"20190430"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b001ccea402","zoneState":1,"zoneStatus":16}},"genPowerCfg":{"attributes":{"batteryVoltage":36,"batteryPercentageRemaining":7}},"ssIasWd":{"attributes":{"1":50,"2":50,"3":0,"maxDuration":240}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":2,"hwVersion":2,"dateCode":"20190430","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1614724445947}

Entries in the log
Zigbee2MQTT:debug 2021-03-02 23:34:47: Received Zigbee message from '0xbc33acfffe17d8c4', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":7,"batteryVoltage":36}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2021-03-02 23:34:47: Received message from unsupported device with Zigbee model 'TS0219' and manufacturer name '_TYZB01_ynsiasng'
Zigbee2MQTT:warn  2021-03-02 23:34:47: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:debug 2021-03-02 23:34:47: Received Zigbee message from '0xbc33acfffe17d8c4', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":7,"batteryVoltage":36}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-02 23:34:47: Received Zigbee message from '0xbc33acfffe17d8c4', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":7,"batteryVoltage":36}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-02 23:34:47: Received Zigbee message from '0xbc33acfffe17d8c4', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":7,"batteryVoltage":36}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-02 23:34:47: Received Zigbee message from '0xbc33acfffe17d8c4', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":7,"batteryVoltage":36}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-02 23:34:47: Received Zigbee message from '0xbc33acfffe17d8c4', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":7,"batteryVoltage":36}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-02 23:34:47: Received Zigbee message from '0xbc33acfffe17d8c4', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":7,"batteryVoltage":36}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2021-03-02 23:34:47: Received message from unsupported device with Zigbee model 'TS0219' and manufacturer name '_TYZB01_ynsiasng'
Zigbee2MQTT:warn  2021-03-02 23:34:47: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2021-03-02 23:34:47: Received message from unsupported device with Zigbee model 'TS0219' and manufacturer name '_TYZB01_ynsiasng'
Zigbee2MQTT:warn  2021-03-02 23:34:47: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2021-03-02 23:34:47: Received message from unsupported device with Zigbee model 'TS0219' and manufacturer name '_TYZB01_ynsiasng'
Zigbee2MQTT:warn  2021-03-02 23:34:47: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2021-03-02 23:34:47: Received message from unsupported device with Zigbee model 'TS0219' and manufacturer name '_TYZB01_ynsiasng'
Zigbee2MQTT:warn  2021-03-02 23:34:47: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2021-03-02 23:34:47: Received message from unsupported device with Zigbee model 'TS0219' and manufacturer name '_TYZB01_ynsiasng'
Zigbee2MQTT:warn  2021-03-02 23:34:47: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2021-03-02 23:34:47: Received message from unsupported device with Zigbee model 'TS0219' and manufacturer name '_TYZB01_ynsiasng'
Zigbee2MQTT:warn  2021-03-02 23:34:47: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:debug 2021-03-02 23:34:47: Received Zigbee message from '0xbc33acfffe17d8c4', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":16}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2021-03-02 23:34:47: Received message from unsupported device with Zigbee model 'TS0219' and manufacturer name '_TYZB01_ynsiasng'
Zigbee2MQTT:warn  2021-03-02 23:34:47: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:debug 2021-03-02 23:34:47: Received Zigbee message from '0xbc33acfffe17d8c4', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":16}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2021-03-02 23:34:47: Received message from unsupported device with Zigbee model 'TS0219' and manufacturer name '_TYZB01_ynsiasng'
Zigbee2MQTT:warn  2021-03-02 23:34:47: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:debug 2021-03-02 23:37:25: Saving state to file /config/zigbee2mqtt/state.json

AntonioDiGennaro, 2021-03-03:
Adding more info. Running 1.18.0-dev hassio-zigbee2mqtt addon on supervised ha on debian 10
I am trying to use external converters as I am reading devices.js is now deprecated.
adding this in my configuration

external_converters: 
  - wooxR7051.js

wooxR7051.js looks now like this

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;

const device = {
//    zigbeeModel: ['myZigbeeModel'],
    fingerprint: [{modelID: 'TS0219', manufacturerName: '_TYZB01_ynsiasng'}]
    model: 'TS0219_siren',
    vendor: 'TuYa',
    description: 'My super lamp!',
    fromZigbee: [],
    toZigbee: [],
    exposes: [],
};

module.exports = device;

when zigbee2mqtt starts, I get this error

[22:57:37] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.18.0-dev start /app
> node index.js
(node:3376) UnhandledPromiseRejectionWarning: SyntaxError: Unexpected identifier
(Use `node --trace-warnings ...` to show where the warning was created)
(node:3376) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:3376) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
[22:57:42] INFO: Handing over control to Zigbee2mqtt Core ...

Koenkk, 2021-03-04:
Try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const device = {
// zigbeeModel: ['myZigbeeModel'],
fingerprint: [{modelID: 'TS0219', manufacturerName: '_TYZB01_ynsiasng'}]
model: 'TS0219_siren',
vendor: 'TuYa',
description: 'My super lamp!',
fromZigbee: [fz.ts0216_siren, fz.battery],
exposes: [e.battery(), exposes.binary('alarm', ea.STATE_SET, true, false),
            exposes.numeric('volume', ea.ALL).withValueMin(0).withValueMax(100).withDescription('Volume of siren')],
toZigbee: [tz.ts0216_alarm, tz.ts0216_duration, tz.ts0216_volume],
meta: {configureKey: 1},
configure: async (device, coordinatorEndpoint, logger) => {
    const endpoint = device.getEndpoint(1);
    await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
},
};

module.exports = device;
```

AntonioDiGennaro, 2021-03-04:
thank you Koen!
now using the following with some good progress

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const device = {
// zigbeeModel: ['myZigbeeModel'],
fingerprint: [{modelID: 'TS0219', manufacturerName: '_TYZB01_ynsiasng'}],
model: 'TS0219_siren',
vendor: 'TuYa',
description: 'Sound and flash siren',
whiteLabel: [{vendor: 'Woox', model: 'R7051'}],
fromZigbee: [fz.ts0216_siren, fz.battery, fz.command_status_change_notification_action, fz.ias_no_alarm],
exposes: [e.battery(), exposes.binary('alarm', ea.STATE_SET, true, false),
            exposes.numeric('volume', ea.ALL).withValueMin(0).withValueMax(100).withDescription('Volume of siren')],
toZigbee: [tz.ts0216_alarm, tz.ts0216_duration, tz.ts0216_volume],
meta: {configureKey: 1},
configure: async (device, coordinatorEndpoint, logger) => {
    const endpoint = device.getEndpoint(1);
    await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
    },
};

module.exports = device;

I am not getting any more message 'No converter.....'

I need to play with the device to understand now how to make flash and play sound.
two more things are power source is not identified and the whitelabel info doesn't seem to be displayed anywhere

[image]
[image]

Koenkk, 2021-03-06:
White label isn't displayed indeed (only used in docs). Can you turn of the alarm via the exposes tab?

AntonioDiGennaro, 2021-03-06:
using the exposes tab I was able to adjust the siren volume and turn on the alarm. It seems instead an error is returned when clicking on the 'false' in an attempt to turn the alarm off. Eventually it turns off but maybe because the duration has elapsed. It does not turn off just after clicking 'false'

`Debug Received MQTT message on 'zigbee2mqtt/0xbc33acfffe17d8f2/set' with data '{"alarm":true}'
Debug Publishing 'set' 'alarm' to '0xbc33acfffe17d8f2'
Debug Received Zigbee message from '0xbc33acfffe17d8f2', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":17}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe17d8f2', payload '{"battery":52.5,"battery_low":false,"linkquality":123,"tamper":false,"voltage":3800,"volume":1}'
Debug Received Zigbee message from '0xbc33acfffe17d8f2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":17}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe17d8f2', payload '{"battery":52.5,"battery_low":false,"linkquality":126,"tamper":false,"voltage":3800,"volume":1}'
Info MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe17d8f2', payload '{"action":"","battery":52.5,"battery_low":false,"linkquality":126,"tamper":false,"voltage":3800,"volume":1}'
Debug Received MQTT message on 'zigbee2mqtt/0xbc33acfffe17d8f2/set' with data '{"alarm":false}'
Debug Publishing 'set' 'alarm' to '0xbc33acfffe17d8f2'
Debug Received Zigbee message from '0xbc33acfffe17d8f2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":17}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe17d8f2', payload '{"battery":52.5,"battery_low":false,"linkquality":126,"tamper":false,"voltage":3800,"volume":1}'
Info MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe17d8f2', payload '{"action":"","battery":52.5,"battery_low":false,"linkquality":126,"tamper":false,"voltage":3800,"volume":1}'
Debug Received Zigbee message from '0xbc33acfffe17d8f2', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":17}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe17d8f2', payload '{"battery":52.5,"battery_low":false,"linkquality":126,"tamper":false,"voltage":3800,"volume":1}'
Error Publish 'set' 'alarm' to '0xbc33acfffe17d8f2' failed: 'Error: Command 0xbc33acfffe17d8f2/1 ssIasWd.startWarning({"startwarninginfo":36,"warningduration":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 34641 - 1 - 150 - 1282 - 11 after 10000ms)'
Debug Error: Command 0xbc33acfffe17d8f2/1 ssIasWd.startWarning({"startwarninginfo":36,"warningduration":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 34641 - 1 - 150 - 1282 - 11 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:67:35) at listOnTimeout (internal/timers.js:554:17) at processTimers (internal/timers.js:497:7)
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'alarm' to '0xbc33acfffe17d8f2' failed: 'Error: Command 0xbc33acfffe17d8f2/1 ssIasWd.startWarning({\"startwarninginfo\":36,\"warningduration\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 34641 - 1 - 150 - 1282 - 11 after 10000ms)'","meta":{"friendly_name":"0xbc33acfffe17d8f2"},"type":"zigbee_publish_error"}'
Debug Received Zigbee message from '0xbc33acfffe17d8f2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":16}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe17d8f2', payload '{"battery":52.5,"battery_low":false,"linkquality":126,"tamper":false,"voltage":3800,"volume":1}'`

Koenkk, 2021-03-06:
Try changing `meta: {configureKey: 1},` to `meta: {configureKey: 1, timeout: 30000},`. Probably the device only wakes up e.g. every 15 seconds, that is the reason for the delay.

Koenkk, 2021-03-10:
Hmm looks like the device doesn't respond. Since it's TuYa you need the original gateway and follow: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html

closed 2021-04-10 by Koenkk

comwise-nl, 2022-02-01:
Apologies for commenting on an old issue. However, I had the same issue of not being able to switch off the Woox siren and had not found a working converter so far.

I have taken the TS0216 converter as a basis and made some adjustments which work for me. I can now switch the siren on and off.

By setting the volume to 0 it is also possible to only have the flash light active without sound.

The reason why the TS0216 converter cannot switch off the siren is because it does not correctly obtain the alarm status. 

The converter code I'm currently using is listed below.

@AntonioDiGennaro, if you still have the Woox siren would it be possible for you to check whether this external converter also works for you?

I still do get the same error message you mentioned earlier, but it does not seem to have a negative effect.

```javascript
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const woox = {
    fz: {
        siren_alarm_status: {
            cluster: 'ssIasZone',
            type: ['attributeReport', 'commandStatusChangeNotification'],
            convert: (model, msg, publish, options, meta) => {
                const zoneStatus = msg.data.zoneStatus;
                const dp = msg.data.dp; 
                switch (zoneStatus) {
                    case 16: return {alarm: false};
                    case 24: return {alarm: false};
                    case 17: return {alarm: true};
                    case 25: return {alarm: true};
                }
            },
        },
        siren_duration_volume: {
            cluster: 'ssIasWd',
            type: ['attributeReport', 'readResponse'],
            convert: (model, msg, publish, options, meta) => {
                const result = {};
                if (msg.data.hasOwnProperty('maxDuration')) result['duration'] = msg.data.maxDuration;
                if (msg.data.hasOwnProperty('2')) {
                    result['volume'] = msg.data['2'];
                }
                return result;
            },
        },
    
    },

    tz: {
        ts0219_volume: {
            key: ['volume'],
            convertSet: async (entity, key, value, meta) => {
                await entity.write('ssIasWd', {0x0002: {value: value, type: 0x20}});
            },
            convertGet: async (entity, key, meta) => {
                await entity.read('ssIasWd', [0x0002]);
            },
        },
    },
};

module.exports = [
    {
        fingerprint: [{modelID: 'TS0219', manufacturerName: '_TYZB01_ynsiasng'}],
        model: 'Zigbee Smart Indoor Siren R7051',
        vendor: 'Woox',
        description: 'Woox Zigbee Smart Indoor Siren',
        meta: {configureKey: 1, timeout: 30000},
        
        fromZigbee: [fz.battery, woox.fz.siren_alarm_status, woox.fz.siren_duration_volume],
        toZigbee: [ tz.ts0216_alarm, tz.ias_max_duration, woox.tz.ts0219_volume],

        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic', 'genPowerCfg']);
            await endpoint.read('ssIasZone', ['zoneState', 'iasCieAddr', 'zoneId']);
            await endpoint.read('ssIasWd', ['maxDuration']);
            // Device advertises itself as Router but is an EndDevice
            device.type = 'EndDevice';
            device.save();
        },
        exposes: [e.battery_low(), e.battery(),
        exposes.numeric('max_duration', ea.ALL).withUnit('s').withValueMin(0).withValueMax(600).withDescription('Duration of Siren'),
        exposes.binary('alarm', ea.STATE_SET, true, false).withDescription('Manual start of siren'),
        exposes.numeric('volume', ea.ALL,).withValueMin(0).withValueMax(100).withDescription('Volume of siren')],
    },
];
```

Kumskaka, 2022-02-15:
Hello, I have the siren and found following data points:

Alarm time
code: "18"
Alarm Volume
code: "5"
Power supply state
code: "6"
Alarm switch
code: "13"
Battery level
code: "14"
Brightness Adjustment
code: "17"

Kumskaka, 2022-02-15:
comwise-nl
I get the error too. Maybe the datapoints are helpful.

comwise-nl, 2022-02-15:
I changed the meta line to the following to get rid of the error message:

```javascript
meta: {configureKey: 1, timeout: 30000, disableDefaultResponse: true},
```

I already had the list of data points, but unfortunately could not get them to work following the instructions, as it seems this device does not expose the 'manuSpecificTuya' cluster.

In the cluster listing there is an unknown cluster 61183. Which is probably the cluster to use, but I don't know how.

Kumskaka, 2022-02-17:
> Hmm looks like the device doesn't respond. Since it's TuYa you need the original gateway and follow: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html

Hallo Konekk,
maybe you can have a look on this problem.
The brightness level is missing, too.

AntonioDiGennaro, 2022-02-18:
Hello,
thanks for trying to make this siren work. It took me some time to check as I had left my home assistant project a bit behind.
I have tried your converter and it seems to be working fine for the goal of turning it on and off at a set volume and a given duration.
My main concern with this siren is the battery duration. Maybe I was not lucky. I got 2 units and the battery does not last.
I understand that if the siren goes off it consumes a lot of energy and needs be recharged.
But if it remains silent the battery should last for weeks if not months.
I am afraid this is independent of the converter though.
I'll try once again to fully charge them and see how long they last.
I would be interested to hear what your experience with the battery of this unit.
thanks

comwise-nl, 2022-02-19:
I can't really comment on the battery life. I was under the impression that the battery is rather intended as a backup supply, so I have always used the sirens supplied through the power adapter.

gisi97, 2023-01-11:
Hi, I am writing here because I have a tuya siren "TS0219" that I could not connect, but thanks to the help of @comwise-nl  now almost everything works. The only thing that doesn't appear to me is the entity of the "tamper". Do you have any idea how to add ? [https://github.com/Koenkk/zigbee-herdsman-converters/issues/5240](https://github.com/Koenkk/zigbee-herdsman-converters/issues/5240)
[image]

Mecallie, 2023-11-13:
> Apologies for commenting on an old issue. However, I had the same issue of not being able to switch off the Woox siren and had not found a working converter so far.
> 
> I have taken the TS0216 converter as a basis and made some adjustments which work for me. I can now switch the siren on and off.
> 
> By setting the volume to 0 it is also possible to only have the flash light active without sound.
> 
> The reason why the TS0216 converter cannot switch off the siren is because it does not correctly obtain the alarm status.
> 
> The converter code I'm currently using is listed below.
> 
> @AntonioDiGennaro, if you still have the Woox siren would it be possible for you to check whether this external converter also works for you?
> 
> I still do get the same error message you mentioned earlier, but it does not seem to have a negative effect.
> 
> ```js
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const woox = {
>     fz: {
>         siren_alarm_status: {
>             cluster: 'ssIasZone',
>             type: ['attributeReport', 'commandStatusChangeNotification'],
>             convert: (model, msg, publish, options, meta) => {
>                 const zoneStatus = msg.data.zoneStatus;
>                 const dp = msg.data.dp; 
>                 switch (zoneStatus) {
>                     case 16: return {alarm: false};
>                     case 24: return {alarm: false};
>                     case 17: return {alarm: true};
>                     case 25: return {alarm: true};
>                 }
>             },
>         },
>         siren_duration_volume: {
>             cluster: 'ssIasWd',
>             type: ['attributeReport', 'readResponse'],
>             convert: (model, msg, publish, options, meta) => {
>                 const result = {};
>                 if (msg.data.hasOwnProperty('maxDuration')) result['duration'] = msg.data.maxDuration;
>                 if (msg.data.hasOwnProperty('2')) {
>                     result['volume'] = msg.data['2'];
>                 }
>                 return result;
>             },
>         },
>     
>     },
> 
>     tz: {
>         ts0219_volume: {
>             key: ['volume'],
>             convertSet: async (entity, key, value, meta) => {
>                 await entity.write('ssIasWd', {0x0002: {value: value, type: 0x20}});
>             },
>             convertGet: async (entity, key, meta) => {
>                 await entity.read('ssIasWd', [0x0002]);
>             },
>         },
>     },
> };
> 
> module.exports = [
>     {
>         fingerprint: [{modelID: 'TS0219', manufacturerName: '_TYZB01_ynsiasng'}],
>         model: 'Zigbee Smart Indoor Siren R7051',
>         vendor: 'Woox',
>         description: 'Woox Zigbee Smart Indoor Siren',
>         meta: {configureKey: 1, timeout: 30000},
>         
>         fromZigbee: [fz.battery, woox.fz.siren_alarm_status, woox.fz.siren_duration_volume],
>         toZigbee: [ tz.ts0216_alarm, tz.ias_max_duration, woox.tz.ts0219_volume],
> 
>         configure: async (device, coordinatorEndpoint, logger) => {
>             const endpoint = device.getEndpoint(1);
>             await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic', 'genPowerCfg']);
>             await endpoint.read('ssIasZone', ['zoneState', 'iasCieAddr', 'zoneId']);
>             await endpoint.read('ssIasWd', ['maxDuration']);
>             // Device advertises itself as Router but is an EndDevice
>             device.type = 'EndDevice';
>             device.save();
>         },
>         exposes: [e.battery_low(), e.battery(),
>         exposes.numeric('max_duration', ea.ALL).withUnit('s').withValueMin(0).withValueMax(600).withDescription('Duration of Siren'),
>         exposes.binary('alarm', ea.STATE_SET, true, false).withDescription('Manual start of siren'),
>         exposes.numeric('volume', ea.ALL,).withValueMin(0).withValueMax(100).withDescription('Volume of siren')],
>     },
> ];
> ```

Hi, I also have this siren. Right now all I can do with it is have it beep when I set the volume to a different level...
Is there some kind of manual on how to import this custom converter somehow? I would like to be able to use this siren...

faust93, 2024-01-06:
Hi, I managed to create external converter for this siren with volume, brightness & duration controls exposed

https://gist.github.com/faust93/7c99b63768f3fb376ee129d155752b6d