Koenkk/zigbee2mqtt #6540: Color Wheel not working properly

T-IOOO, 2021-03-03:
When adjusting the temperature wheel to cold in homekit, the dot will jump to another position upon release.
It's as if only 1/4 of the wheel is working, a very narrow adjustment around the warm area of the wheel.
When I adjust the temperature value and at the same time looking at the log, the temp value reaches 250 on the warmer area of the wheel and the rest 3/4 of the temp. wheel is only staying at 250
I have only tested with IKEA light bulbs

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: CC2531
Adapter firmware version: 20190608

sjorge, 2021-03-03:
Trådfri WS bulbs have a minColorTemp of 250 mireds and a maxColorTemp of 454 mireds.

They can't output color temp outside this range.

T-IOOO, 2021-03-03:
I am aware of that, but the temperature wheel is not working as it should. It was working just fine before the two latest updates.
See the areas in the image. The red area is the only area that is chaning the values
[image]

sjorge, 2021-03-03:
The blue area you marked sends values < 250 for me, so it makes sense that they are not making the light cooler.

T-IOOO, 2021-03-03:
The values were spread evenly from cold to warm before.
If I lift my finger anywhere in the blue area the marker will snap back to the area where the red/blue meet.
It's not supposed to act like that and hasn't before.

sjorge, 2021-03-03:
I did some more digging, looks hass.io is not passing the minValue and maxValue correctly to homekit.
[image]

When maxValue=454 and minValue=250 is passed, the color temperature picker looks slightly different, but now the coolest spot returns 250 and the warmest returns 454

T-IOOO, 2021-03-03:
my temperature picker is not changing like that.
Did the marker "snap" to that position?

sjorge, 2021-03-03:
When the range is properly passed to homekit, it does not snap. Because 
all positions on the wheel will are within range of the bulb.

Not sure if the issue is how z2m passes the info to hass.io or how 
hass.io passes it to homekit.

sjorge, 2021-03-03:
https://github.com/Koenkk/zigbee2mqtt/blob/a43d1f3917fee5516429bf1bf80419f1d2e0e7f8/lib/extension/homeassistant.js#L103-L106

Looks like z2m is passing it to ha, I guess ha is ignoring those then or not passing them to homekit.

sjorge, 2021-03-03:
ha is also hardcoding it, https://github.com/home-assistant/core/blob/dev/homeassistant/components/homekit/type_lights.py#L88-L93 should probably use the discovery_payload values if available.

T-IOOO, 2021-03-03:
Maybe I should mention that I am using Homebridge and not Home Assistant

sjorge, 2021-03-03:
Which plugin? It's probably the same issue of it not setting min/max for homekit.

T-IOOO, 2021-03-03:
”Homebridge Zigbee2MQTT”

T-IOOO, 2021-04-03:
Someone who's willing or able to fix this issue? 😊

sjorge, 2021-04-03:
It’s not a z2m issue, I believe it was properly fixed in https://github.com/itavero/homebridge-z2m 

However there is a non finished discussion on the correctness of the fix in the issue there https://github.com/itavero/homebridge-z2m/issues/88 I mention I no longer have a gateway from ikea to verify if the behavior is the same or not. If it is indeed not that same a form of mapping should be done at the home ridge side.

closed 2021-05-12 by github-actions[bot]