Koenkk/zigbee2mqtt #6551: Please add support for EasyAccess door lock-model: Easyfinger

sonybansi, 2021-03-03:
step no 2 Adding your device# I didn't find where is node_modules/zigbee-herdsman--converters/devices. js

## Information about the device + link
I have a easyfinger-v1
https://easyaccess.no/product/easyfinger-v2/

[image]

[image]
[image]

..

## data/database.db entry of the device
[log1.txt](https://github.com/Koenkk/zigbee2mqtt/files/6076886/log1.txt)

..

GD-Dal, 2021-03-03:
Nice.
I've actually made a request for the zigbee module in this lock a few days ago:
https://github.com/Koenkk/zigbee2mqtt/issues/6379

I have since been in contact with the reseller (fingeravtrykk.no), and they told me that they had made a deal with the zigbee module developers that they should create a manual on how to implement this module into home systems, including Zigbee2MQTT.

This should be ready in "about 14 days", so time will tell what they actually meant by this.

sonybansi, 2021-03-03:
Ok, thanks 
Can you upload pictures of your Zigbee modul and door lock. I think i will close the issue and follow your request for add new device.

sonybansi, 2021-03-04:
**I have tried again ,maybe i dont understand documentation.**
[image]
**i have create a file easyaccess.js and add in zigbee2mqtt folder in smb://homeassistant/share/zigbee2mqtt/easyaccess.js**
[image]
 

**Homeassistant addon  zigbee2mqtt configurations:**

`data_path: /share/zigbee2mqtt
external_converters:
  - easyaccess.js
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://core-mosquitto'
  user: mqtt
  password: mqtt
serial:
  port: >-
    /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018E1E88B-if00
advanced:
  log_level: info
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false`

zigbee2mqtt LOG
`> zigbee2mqtt@1.17.1 start /app
> node index.js
evalmachine.<anonymous>:20
};
^
SyntaxError: Unexpected token '}'
    at new Script (vm.js:101:7)
    at createScript (vm.js:262:10)
    at Object.runInNewContext (vm.js:303:10)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.js:159:16)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.js:10:20)
    at new Controller (/app/lib/controller.js:77:34)
    at Object.<anonymous> (/app/index.js:25:20)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.17.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.17.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> zigbee2mqtt@1.17.1 start /app
> node index.js
> zigbee2mqtt@1.17.1 start /app
> node index.js
> zigbee2mqtt@1.17.1 start /app
> node index.js
> zigbee2mqtt@1.17.1 start /app
> node index.js
> zigbee2mqtt@1.17.1 start /app
> node index.js
> zigbee2mqtt@1.17.1 start /app
> node index.js
> zigbee2mqtt@1.17.1 start /app
> node index.js
> zigbee2mqtt@1.17.1 start /app
> node index.js
> zigbee2mqtt@1.17.1 start /app
> node index.js
> zigbee2mqtt@1.17.1 start /app
> node index.js
> zigbee2mqtt@1.17.1 start /app
> node index.js
> zigbee2mqtt@1.17.1 start /app
> node index.js
[14:27:54] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.17.1 start /app
> node index.js
> zigbee2mqtt@1.17.1 start /app
> node index.js
> zigbee2mqtt@1.17.1 start /app
> node index.js
> zigbee2mqtt@1.17.1 start /app
> node index.js
> zigbee2mqtt@1.17.1 start /app
> node index.js
> zigbee2mqtt@1.17.1 start /app
> node index.js`

nununu76, 2021-03-06:
Hei.. Someone knows how to connect easy finger touch to smartthings from Samsung, i got the ziggbee 3.0 module and i connect to the lock but it doesn't connected..

nununu76, 2021-03-06:
I got this" E-Life Zigbee Module " but i don't know how can it help me, i can't attach the address of the file but the name of the file it's enough to get it

stolevegen, 2021-03-23:
> I have since been in contact with the reseller (fingeravtrykk.no), and they told me that they had made a deal with the zigbee module developers that they should create a manual on how to implement [...]

@GD-Dal 
That would be nice, but interesting to see if they go into specific details for different systems. 
I'm working on Z2M integration right now, but my issue has been closed. I've requested it re-opened. If not I can update here

Koenkk, 2021-03-26:
Looking at the pdf in https://github.com/Koenkk/zigbee2mqtt/issues/6379#issue-812608086 it seems they use standard zigbee stuff.

Try with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const device = {
    zigbeeModel: ['EasyCode903G2.1'],
    model: 'EasyCode903G2.1',
    vendor: 'EasyAccess',
    description: 'EasyFinger V2',
    fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery],
    toZigbee: [tz.lock],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
        await reporting.lockState(endpoint);
        await reporting.batteryPercentageRemaining(endpoint);
    },
    exposes: [e.lock(), e.battery()],
};

module.exports = device;
```

Next try to control it via the zigbee2mqtt frontend -> click on your device and go to the exposes tab.

stolevegen, 2021-03-26:
Adding the following to fromZigbee.js gave me an attribute to track lock operation source. 
Then adding "fromZigbee: [fz.easycodetouch_action, fz.lock, fz.lock_operation_event, fz.battery]," to devices.js
`
    easycodetouch_action: {
        cluster: 'closuresDoorLock',
        type: 'raw',
        convert: (model, msg, publish, options, meta) => {
            const lck_lookup = {
                13: 'manual locked',
                14: 'zigbee unlocked',
                 3: 'rfid tag unlocked',
                 0: 'keypad unlocked',
            },
            lok = lck_lookup[msg.data[3]],
            lck = lck_lookup[msg.data[4]];
            if (lck == "manual locked" || lck == "zigbee unlocked") {
              return {"action": lck}; } 
            else {
              return {"action": lok}; } 
        },
    },
`

Koenkk, 2021-03-26:
@stolevegen try applying the changes from https://github.com/Koenkk/zigbee-herdsman-converters/pull/2418/files 

- When NOT having `fz.easycodetouch_action`, can you check removing `fz.lock_operation_event`, does this have any effect? (did you get an `action` before and after removing not?)
- Can you provide the herdsman debug log when receiving the message for `fz.easycodetouch_action`

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

stolevegen, 2021-03-26:
@Koenkk 
Applied changes. 

First ( **no** fz.easycodetouch_action, **with** fz.lock_operation_event) : 

- No action / lock operation source. 

- Sound volume is visible in Home Assistant, but as a sensor. No operation possible.

- Auto-relock-time is visible in Home Assistant, but as a sensor. No operation possible. This has suffix "s". Only two available values is 0 or 1, disabled or enabled. Not number of seconds. 

Second ( **with** fz.easycodetouch_action, **and** `e.action(['manual locked','zigbee unlocked','rfid tag unlocked','keypad unlocked'])` **and** fz.lock_operation_event : 
Herdsman debug log for fz.easycodetouch_action : https://pastebin.com/tswqksqL
Herdsman more specific : https://pastebin.com/eY7V4hag
Sensor values updated in Home Assistant as expected. 

Third ( **with** fz.easycodetouch_action, **and** `e.action(['manual locked','zigbee unlocked','rfid tag unlocked','keypad unlocked'])` **NO** fz.lock_operation_event : 
Same as the above, second. 
Herdsman log : https://pastebin.com/FQR6tgB1

Fourth ( **NO** fz.easycodetouch_action, **NO** `e.action(['manual locked','zigbee unlocked','rfid tag unlocked','keypad unlocked'])` **NO** fz.lock_operation_event : 
No action / lock operation source. 
Herdsman log for unlock / lock : https://pastebin.com/LPswWEQm

Koenkk, 2021-03-26:
@stolevegen 

- It's expected that you cannot control the sound volume and auto relock time from Home Assistant (because home assistant doesn't have the correct entities for this). You can control them via the Zigbee2MQTT frontend. 
- Update the merge request again, please apply the changes
  - The device sends malformed messages, therefore the `fz.lock_operation_event` won't work.
  - You should now get an action, if yes please provide me all the generated actions.

stolevegen, 2021-03-26:
> * It's expected that you cannot control the sound volume and auto relock time from Home Assistant (because home assistant doesn't have the correct entities for this). You can control them via the Zigbee2MQTT frontend.

I see. I'm not using the add-on. I have installed Z2M on Raspberry Pi, can I still use frontend? 

> * Update the merge request again, please apply the changes
>   
>   * The device sends malformed messages, therefore the `fz.lock_operation_event` won't work.
>   * You should now get an action, if yes please provide me all the generated actions.

WIll do.

stolevegen, 2021-03-26:
**Locking with Home Assistant/Zigbee**
`"action":"manual_lock","action_source":255,"action_user":0`
**Locking with Power Button on physical doorlock == manual_lock** 
`"action":"manual_lock","action_source":255`
Log for zigbee lock : https://pastebin.com/T7Xdj41z
Log for power button lock : https://pastebin.com/SC9XrQvj

**Unlocking with Home Assistant/Zigbee**
`"action":"manual_unlock","action_source":255`
Log : https://pastebin.com/6xXSgbtE

**Unlocking with keypad:**
`"action":"unlock","action_source":0,"action_source_name":"keypad"`
Log : https://pastebin.com/pwUCBkDR

**Unlocking with RFID tag:**
`"action":"unlock","action_source":3,"action_source_name":"rfid"`
Log: https://pastebin.com/LRzqp7py

Lock Home Assistant/Zigbee produces error: 
```
  zigbee-herdsman:adapter:zStack:adapter Data confirm error (0xf4ce3632a29609ab:62222,240,0) +130ms
  zigbee-herdsman:adapter:zStack:adapter Wait 2000ms +0ms
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xf4ce3632a29609ab:62222/11 (0,1,1) +2s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":62222,"destendpoint":11,"srcendpoint":1,"clusterid":257,"transid":34,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,145,11,10,0]}} +2s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,14,243,11,1,1,1,34,0,30,5,24,145,11,10,0,108] +2s
```
Unlock Home Assistant/Zigbee produces error:
```
  zigbee-herdsman:adapter:zStack:adapter Data confirm error (0xf4ce3632a29609ab:62222,240,0) +44ms
  zigbee-herdsman:adapter:zStack:adapter Wait 2000ms +0ms
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xf4ce3632a29609ab:62222/11 (0,1,1) +2s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":62222,"destendpoint":11,"srcendpoint":1,"clusterid":257,"transid":30,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,142,11,10,0]}} +2s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,14,243,11,1,1,1,30,0,30,5,24,142,11,10,0,79] +2s
```

GD-Dal, 2021-03-27:
Wow, things are happening!
But I'm confused; to test this, should I use the approach with a devices.js file and a fromZigbee.js file like @stolevegen does, or is it enough with an external converters file, like @Koenkk mentions here: https://github.com/Koenkk/zigbee2mqtt/issues/6551#issuecomment-808397423

stolevegen, 2021-03-27:
@Koenkk 
Regarding action. I think source code changes from index[3] to index[4] between (keypad+RFID) and (zigbee+powerbutton)

Unlock keypad will return [25,5,32,**0**,2,0,0,0,0,0,0] where index[3] == "0"
Unlock RFID tag will return [25,12,32,**3**,2,0,0,0,0,0,0] where index[3] == "3"

Unlock Zigbee will return [25,39,32,255,**14**,0,0,0,0,0,0] where index[4] == "14"
Lock powerbutton/zigbee will return [25,8,32,255,**13**,0,0,0,0,0,0] where index[4] == "13"

Koenkk, 2021-03-27:
@stolevegen
-  to enable the frontend see: https://www.zigbee2mqtt.io/information/frontend.html
- updated code again, actions should work now (used your code)

@GD-Dal only via devices.js, external converter config is a bit different, might be better to wait until it is merged (which is soon I think)

GD-Dal, 2021-03-28:
> @GD-Dal only via devices.js, external converter config is a bit different, might be better to wait until it is merged (which is soon I think)
Thanks, I'll just wait then.
The door look from the now 2 merged threads are probably the same, but the zigbee modules are probably not. Because mine is recognized as "easyCodeTouch_v1"
The cluster is not the same either:

[image]

stolevegen, 2021-03-28:
I have EasyCodeTouch which is also recognised as "easyCodeTouch_v1" with same clusters. 
So EasyCodeTouch and EasyFinger V1 seems to have same module.

stolevegen, 2021-03-28:
@Koenkk 
I've tested the updated code, through multiple reboot and some lock operation cycles. So far everything seems to work just fine :)

A couple of notes ; 

Auto Relock Time is described as number of seconds before locking, and has suffix "s". However it's setting is either enabled (1) or disabled (0), not the actual number of seconds. 

**In devices.js**
The line `e.action(['zigbee_unlock', 'manual_unlock', 'rfid_unlock', 'keypad_unlock'])],` is missing lock. 
I think should be `e.action(['zigbee_unlock', 'manual_lock', 'rfid_unlock', 'keypad_unlock'])],`
I get the correct values in Home Assistant. 
See values in fromZigbee.js below. 

**fromZigbee.js**
```
                13: 'manual_lock',
                 14: 'zigbee_unlock',
                 3: 'rfid_unlock',
                 0: 'keypad_unlock',
```

I set the name as `manual_lock` , but it actually covers both manual lock by push button on the doorlock, as well as lock by zigbee command, so there might be a more suitable name for it than manual.

As a note, when lock is operated, action will only be visible for max half a second, then it will be an empty string again. I'm not sure if it's possible to keep the value until next change? In Home Assistant I've used a workaround (an Automation with Input Select)

Koenkk, 2021-03-28:
@stolevegen 

- About the auto relock time, are you sure? Since this is what the ZCL specs says about it (could be that the device deviates from this)
[image]
- Fixed the action list and renamed `manual_lock` to `lock`
- The action disappearing immediately is expected when `homeassistant: true` is set in the `configuration.yaml`. This is to trigger an state change on the entity (otherwise you cannot use the entity for automations).

stolevegen, 2021-03-28:
Yes, there is an discrepancy between this zigbee module and ZCL spec. 
The manual states
```
Autolock
Enable = 0x01, Disable = 0x00 These are the only valid values. Reported when changed.
```
which is confirmed when tested. If value is 1, 10 or even 60, it will always relock after approx 7 seconds. 

>The action disappearing immediately is expected when homeassistant: true is set in the configuration.yaml. This is to trigger >an state change on the entity (otherwise you cannot use the entity for automations).

Ok, got it.

Koenkk, 2021-03-29:
@stolevegen could you share the data/database.db entry of this device?

Koenkk, 2021-03-29:
Both auto relock (is a binary now) and configure error should be fixed now.

stolevegen, 2021-03-30:
@Koenkk 
Sweet. 
I'm jumping back and forth between Z2M and deCONZ, but will test it soon.

stolevegen, 2021-04-01:
I get the following error when npm start : 

```
(node:14558) UnhandledPromiseRejectionWarning: Error: Cannot find module 'https-proxy-agent'
Require stack:
- /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js
- /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/ledvance.js
- /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/index.js
- /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js
- /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js
- /opt/zigbee2mqtt/lib/zigbee.js
- /opt/zigbee2mqtt/lib/controller.js
- /opt/zigbee2mqtt/index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:815:15)
    at Function.Module._load (internal/modules/cjs/loader.js:667:27)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:3:25)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
(node:14558) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:14558) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
pi@raspberrypi:/opt/zigbee2mqtt $ 
```

Koenkk, 2021-04-01:
Try executing: `/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/ && npm ci`

stolevegen, 2021-04-01:
Thanks. 
Settings sound volume and auto relock now works.
When changes are pulled it should be supported natively/out of the box, correct?

I got a small error at startup ; 
```
Zigbee2MQTT:error 2021-04-01 19:02:55: Failed to call 'Bridge' 'onMQTTConnected' (TypeError: this.herdsman.getPermitJoinTimeout is not a function
    at Zigbee.getPermitJoinTimeout (/opt/zigbee2mqtt/lib/zigbee.js:149:30)
    at Bridge.publishInfo (/opt/zigbee2mqtt/lib/extension/bridge.js:563:46)
    at async Bridge.onMQTTConnected (/opt/zigbee2mqtt/lib/extension/bridge.js:62:9)
    at async Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:382:21)
    at async Controller.start (/opt/zigbee2mqtt/lib/controller.js:163:9)
    at async start (/opt/zigbee2mqtt/index.js:43:5))
```

stolevegen, 2021-04-01:
full log : 

https://pastebin.com/EwqHMFW1

TheStigh, 2021-04-03:
@stolevegen, could you please share the documentation of this lock? You can contact me through PM. I assume your are in Norway as me?

Koenkk, 2021-04-04:
@stolevegen merged. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

https://github.com/Koenkk/zigbee2mqtt/issues/6551#issuecomment-812091441 looks like an incompatibility betwen zigbee-herdsman and zigbee2mqtt, if you update to the latest dev this should be solved.

stolevegen, 2021-04-04:
@TheStigh Yep, from Norway as well. Not sure how to send PM actually, but here is the manual :) 
The manual doesn't describe it, but to reset, hold the button on the zigbee module for 5-6-7 sec until it blinks rapidly. 
https://easyaccess.no/product/easycodetouch/
https://github.com/dresden-elektronik/deconz-rest-plugin/files/5921247/user.manual.pdf

@Koenkk  👍

TheStigh, 2021-04-04:
@stolevegen Kan du kontakte meg på stigh.aarstein@gmail.com eller på Telegram @TheStigh ?

GD-Dal, 2021-04-28:
Hi.
Just checking in.
Is this door lock fully implemented in next version of Z2MQTT now.
Can't wait to try it out :)

Thanks.

stolevegen, 2021-05-01:
@GD-Dal  I see easyaccess.js was added to zigbee-herdsman-converters/devices 5 days ago. It was working fine in dev last I tested. I've switched a lot between deCONZ and Z2M lately, and always get some issues somewhere, so would prefer not to switch again. @Koenkk

GD-Dal, 2021-05-02:
> 
> 
> @GD-Dal I see easyaccess.js was added to zigbee-herdsman-converters/devices 5 days ago. It was working fine in dev last I tested. I've switched a lot between deCONZ and Z2M lately, 
Nice.
But sadly it don't see to have been included in todays new version:
[image]

Any reason for switching between deCONZ and Z2M? Just testing?
My understanding has been that Z2M is the most developed of those 2, and with most devices supported. Maybe  that has shifted?

stolevegen, 2021-05-03:
I've been contributing to integrating on both platform, only reason for switching back and forth. 

I've been using both platforms, and happy with both. I'm currently on deCONZ only because my Conbee II have a lot better range than my cheap CC module. Now they have added support for Conbee II on Z2M I believe. 

It's easier to support new devices by yourself on Z2M.

TheStigh, 2021-05-03:
@stolevegen @GD-Dal  Just for your information - I have integrated the ID LOCK 150 with many possibilities available and it was supported officially from the May 1st release of Z2M. Though you probably don't have this lock (!) still I just wanted to inform you.

GD-Dal, 2021-05-03:
> I've been contributing to integrating on both platform, only reason for switching back and forth.
>
> I've been using both platforms, and happy with both. I'm currently on deCONZ only because my Conbee [...]

I see. but I have used Conbee II for a long time in Z2M already?
> 
> It's easier to support new devices by yourself on Z2M.
omg :) I would claim it is not easy at all in Z2M.. Can't imagine how hard it must be in dCONz :)

So what about our door lock module, named easyCodeTouch_v1 ?

One should think it was easy enough to just copy the easyaccess.js file in the repository and change the zigbeeModel, since it seems to be a module with similar features. And then add it to the system using external converters..
But it was not, it produces errors.

GD-Dal, 2021-05-03:
> 
> 
> @stolevegen @GD-Dal Just for your information - I have integrated the ID LOCK 150 with many possibilities available and it was supported officially from the May 1st release of Z2M. Though you probably don't have this lock (!) still I just wanted to inform you.

Gni det inn mener du? :) j/k

Nice! What possibilities and functions are we talking about? Do you have screenshots?
Are they using the same zigbee module as Easyaccess?

stolevegen, 2021-05-03:
@GD-Dal 

I can send you my devices.js, toZigbee.js, fromZigbee.js etc later if you want to test with my exact config. I had success with that, but haven't tested with latest commit/version.

TheStigh, 2021-05-03:
> Gni det inn mener du? :) j/k

Jepp :) send meg en epost på stigh.aarstein@gmail.com så skal du få screenshots

No, it is different. As far as I've learned, the firmware on ID LOCK are a lot less buggy compared to The Access ones, but what do I know. I love the ID LOCK 150!

GD-Dal, 2021-05-03:
> 
> 
> @GD-Dal
> 
> I can send you my devices.js, toZigbee.js, fromZigbee.js etc later if you want to test with my exact config. I had success with that, but haven't tested with latest commit/version.

That would be grand.

BTW: These are the errors I'm getting by using the easyaccess.js in the repository (the model name is changed of course):

`Failed to configure 'Utedør', attempt 1 (Error: ConfigureReporting 0xf4ce367919d08145/11 closuresDoorLock([{"attribute":"lockState","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received) at DeconzAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:607:23) at Generator.next (<anonymous>) at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:24:58))

Failed to configure 'Utedør', attempt 2 (TypeError: Cannot read property 'configureKey' of undefined at Configure.configure (/app/lib/extension/configure.js:138:69))

Failed to configure 'Utedør', attempt 3 (TypeError: Cannot read property 'configureKey' of undefined at Configure.configure (/app/lib/extension/configure.js:138:69))`

stolevegen, 2021-05-03:
@GD-Dal At own risk, but I would try adding `meta: {configureKey: 1},`  in easyaccess.js after line 15. Maybe it will fix error 2 and 3. The first one, I don't know, maybe it's specific to the use of conbee with z2m.

GD-Dal, 2021-06-03:
Hi
This integration did not make it's way into v 1.19.0-1 either.
Does it work properly for you.

It really bugs me that we are at the mercy of the developer when new models are supported if at all.
There should be a better way for all of use contribute, and speed things up. A database for example. like z-wave js

TheStigh, 2021-06-04:
> Hi
> This integration did not make it's way into v 1.19.0-1 either.
> Does it work properly for you.
>
> It really bugs me that we are at the mercy of the developer when new models are supported if at all. [...]

Just have to (sorry bud) - released a v2 of the ID Lock integration with 1.19.0. Now I do think this is the lock with most features available for HA: control volume, Enable/Disable Master PIN use, Enable/Disable RFID use, Enable/Disable Relock, Set 4 different lock modes, support Service Mode with random PIN generation for 1 x use or 24 hr use, Set/Get/Delete PINs.

If you were using Z2M with a regular device, I could have a look at your local converter. Send it to my email (you already have the address), also include a URL to YOUR lock from the manufacturers site. Do you also have the zigbee documentation available for this particular lock?

GD-Dal, 2021-07-03:
Sadly no support for this in 1.20.0-1 either.
Time to give up, with an (expensive) lesson learned:

Until some better tools or better documentation shows up, don't buy products that aren't already supported in Z2M.

I mean, Z2M is beautiful when it works, but when it don't, I'm completely helpless.

eirikd, 2021-07-15:
its supported in latest-dev now

closed 2021-07-15 by Koenkk

sonybansi, 2021-11-27:
> Looking at the pdf in [#6379 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/6379#issue-812608086) it seems they use standard zigbee stuff.
> 
> Try with the following external converter:
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const e = exposes.presets;
> 
> const device = {
>     zigbeeModel: ['EasyCode903G2.1'],
>     model: 'EasyCode903G2.1',
>     vendor: 'EasyAccess',
>     description: 'EasyFinger V2',
>     fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery],
>     toZigbee: [tz.lock],
>     meta: {configureKey: 1},
>     configure: async (device, coordinatorEndpoint, logger) => {
>         const endpoint = device.getEndpoint(1);
>         await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
>         await reporting.lockState(endpoint);
>         await reporting.batteryPercentageRemaining(endpoint);
>     },
>     exposes: [e.lock(), e.battery()],
> };
> 
> module.exports = device;
> ```
> 
> Next try to control it via the zigbee2mqtt frontend -> click on your device and go to the exposes tab.

Thanks , How can i solve the problem with battery state ?
`**Battery state not working** `

Lock/unlock is working.
Link Quality is working

[image]

[image]
[image]
[image]

TheStigh, 2021-11-27:
Thing is; there are so many versions in the market - they are sold through "white labeling" where they use EasyAccess brand but uses a 3.rd party zigbee module. I did integrate one model, but it was from a 3.rd party supplier. They had only implemented lock/unlock/lqi.

So for this manufacturer, it is not easy to make a proper integration working for "everybody".

If you want a well integrated lock with plenty of functionality from HA - there is only one: ID Lock 150.

sonybansi, 2022-03-23:
EasyAccess door loch unsupported again. What do I need to change now.

`Zigbee2MQTT:warn  2022-03-23 21:54:57: Received message from unsupported device with Zigbee model 'EasyCode903G2.1' and manufacturer name 'Datek Wireless'`

Mastiffen, 2022-04-10:
I have the same lock and the same problem. I had an old Zigbee only module, which was very unreliable. It stopped sending to the lock, but not receiving from it. So I could see in Hass when the lock was opened, but I couldn't open it from Hass, or from Zigbee2MQTT's control panel. Every few days I had to remove a battery to get it working again. So I received a new module, a combined Zigbee and BLE module. But now I have the same problem as @sonybansi. I have the latest update to Zigbee2MQTT, but it doesn't work.

@TheStigh I have two ID Lock 150's, been running those since they came, and I had the first model before them. I would prefer to use that, the problem is that they don't have a physical backup key for some strange reason. In the part of the house that I have the EasyAccess lock there is no other way in, so I'd have to smash a very expensive window (it's in an early 1800's house where the only windows that are accepted are the ones that are approved for use, and they cost) or break down the door to get in if the lock stops working completely.

Ninjacas08, 2022-05-02:
> Looking at the pdf in [#6379 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/6379#issue-812608086) it seems they use standard zigbee stuff.
>
> Try with the following external converter:
>
> ```js [...]

Where do I use this code and how? Please explain so I can understand (I am new to HA).

Mastiffen, 2022-05-02:
According to Sonybansi a few posts up it doesn't work anymore and he asks what should be changed to make it work. I was wondering about the same...

andreaseide80, 2022-05-03:
Managed to get our EasyCode Touch connected and "supported" by using the follwing external converter:

---
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const device = {
    zigbeeModel: ['EasyCodeTouch'],
    model: 'EasyCodeTouch-endpoint1',
    vendor: 'EasyAccess',
    description: 'EasyCodeTouch',
    fromZigbee: [fz.easycodetouch_action, fz.lock, fz.lock_operation_event, fz.battery],
    toZigbee: [tz.lock],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
        await reporting.lockState(endpoint);
        await reporting.batteryPercentageRemaining(endpoint);
    },
    exposes: [e.lock(), e.battery(), e.action(['manual locked','zigbee unlocked','rfid tag unlocked','keypad unlocked'])],
};

module.exports = device;
---

But we still get no battery or other status from the lock.
Have also tried device.getEndpoint(11); as noted in earlier posts.
Am I correct in understanding that model, vendor and description does not matter in connecting to the unit? But zigbeeModel is the required "correct" code to connect to it?

When we try to click the Exposed state "UNLOCK" or "LOCK" we get the following errors in the log:
Debug 2022-05-03 10:20:59Received MQTT message on 'zigbee2mqtt/Dør/set' with data '{"state":"UNLOCK"}'
Debug 2022-05-03 10:20:59Publishing 'set' 'state' to 'Dør'
Error 2022-05-03 10:21:20Publish 'set' 'state' to 'Dør' failed: 'Error: Command 0xf4ce36d1a66a2a2e/11 closuresDoorLock.unlockDoor({"pincodevalue":""}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 27260 - 11 - 3 - 257 - 1 after 10000ms)'
Debug 2022-05-03 10:21:20Error: Command 0xf4ce36d1a66a2a2e/11 closuresDoorLock.unlockDoor({"pincodevalue":""}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 27260 - 11 - 3 - 257 - 1 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7)

sonybansi, 2022-05-03:
Some months ago worked fine, but now ERROR 
```
Zigbee2MQTT:error 2022-05-03 21:22:17: Publish 'set' 'state' to 'Easy Access Lock' failed: 'Error: Command 0xxx000dxxxxxcb/1 closuresDoorLock.unlockDoor({"pincodevalue":""}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
```

sonybansi, 2022-05-03:
> @Sonybansi, there you are! I have asked you to contact me on XDA and every other forum I could find your user name... 

Sorry for inconvenience 😕 I was on vacation.

Mastiffen, 2022-05-03:
We all need a vacation every now and then. Maybe you could answer on XDA? You have no way of contact here on Github.

sonybansi, 2022-05-03:
> We all need a vacation every now and then. Maybe you could answer on XDA? You have no way of contact here on Github.

my mobile internet was mostly off during vacation.

Mastiffen, 2022-05-03:
That would have made me crazy... (Mind you, this part of the thread will probably be moderated out of excistence!) Oh, you'll find me as Mastiff on the Norwegian Hjemmeautomasjon forum too.

sonybansi, 2022-05-03:
> That would have made me crazy... (Mind you, this part of the thread will probably be moderated out of excistence!) Oh, you'll find me as Mastiff on the Norwegian Hjemmeautomasjon forum too.

Sorry again 😕

[57 more replies]