Koenkk/zigbee2mqtt #6552: HomeAssistant custom type

aureliosaraiva, 2021-03-03:
## What happened
When I change the status through the HomeAssistant interface of an entity that changed the type, a problem occurs.

Right after I turn on a lamp, it goes back to the state off, but the lamp remains on, but on HomeAssistant it shows that it is off.

The same does not happen with entities that have not had their type modified.

```
'0xec1bbdfffe31f09c': // Sonoff zigbee mini (not working)
    friendly_name: lamp_machine_house
    homeassistant:
      switch:
        type: light
        object_id: light
        name: lamp_machine_house
  '0xec1bbdfffe31f07b': // Sonoff zigbee mini (working)
    friendly_name: lamp_kiosk
```

When I define a type for the endpoint, the status does not reflect in HomeAssistant.

[image]

```
 MQTT publish: topic 'zigbee2mqtt/lamp_machine_house', payload '{"last_seen":"2021-03-03T12:35:54-03:00","linkquality":45,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/lamp_machine_house', payload '{"last_seen":"2021-03-03T12:36:06-03:00","linkquality":42,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/lamp_machine_house', payload '{"last_seen":"2021-03-03T12:38:16-03:00","linkquality":36,"state":"ON"}'
Info MQTT publish: topic 'zigbee2mqtt/lamp_machine_house', payload '{"last_seen":"2021-03-03T12:38:16-03:00","linkquality":45,"state":"ON"}'
```
## What did you expect to happen

I hope the status on zigbee2mqtt will reflect on HomeAssistant

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: zStack3x0
Adapter firmware version: 20201026

m0wlheld, 2021-03-04:
Is this a copy and paste error or is the indentation wrong? Device f09c (lamp_machine_house) should be at the same level as the second.

aureliosaraiva, 2021-03-04:
@m0wlheld it is only retreat and formation.

[image]

aureliosaraiva, 2021-03-04:
// log mqtt when I try turn on/off
```
zigbee2mqtt/lamp_kiosk/set
ON

zigbee2mqtt/lamp_kiosk
{"last_seen":"2021-03-04T17:04:50-03:00","linkquality":48,"state":"ON"}

zigbee2mqtt/lamp_kiosk
{"last_seen":"2021-03-04T17:04:50-03:00","linkquality":48,"state":"ON"}

zigbee2mqtt/lamp_machine_house/set
ON

zigbee2mqtt/lamp_machine_house
{"last_seen":"2021-03-04T17:06:55-03:00","linkquality":48,"state":"ON"}

zigbee2mqtt/lamp_machine_house
{"last_seen":"2021-03-04T17:06:55-03:00","linkquality":48,"state":"ON"}
```

``` 
// log zigbee2mqtt
Debug Received Zigbee message from 'lamp_machine_house', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Debug Received MQTT message on 'zigbee2mqtt/lamp_machine_house/set' with data 'ON'
Debug Received Zigbee message from 'lamp_machine_house', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/lamp_machine_house', payload '{"last_seen":"2021-03-04T17:18:42-03:00","linkquality":48,"state":"ON"}'

 Received Zigbee message from 'lamp_kiosk', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Debug Received MQTT message on 'zigbee2mqtt/lamp_kiosk/set' with data 'ON'
Debug Received Zigbee message from 'lamp_kiosk', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/lamp_kiosk', payload '{"last_seen":"2021-03-04T17:19:17-03:00","linkquality":48,"state":"ON"}'
``` 

```
// core.entity_registry
{
                "entity_id": "switch.lamp_kiosk",
                "config_entry_id": "3b961fe16837f02cf1e1ca7f3a34d269",
                "device_id": "adbfc2aa5df5b3147490b90c948f93dd",
                "area_id": null,
                "unique_id": "0xec1bbdfffe31f07b_switch_zigbee2mqtt",
                "platform": "mqtt",
                "name": null,
                "icon": null,
                "disabled_by": null,
                "capabilities": null,
                "supported_features": 0,
                "device_class": null,
                "unit_of_measurement": null,
                "original_name": "lamp_kiosk",
                "original_icon": null
            },
{
                "entity_id": "light.lamp_machine_house",
                "config_entry_id": "3b961fe16837f02cf1e1ca7f3a34d269",
                "device_id": "c9b3d15ca2a8cd1160cda209de594b60",
                "area_id": null,
                "unique_id": "0xec1bbdfffe31f09c_light_zigbee2mqtt",
                "platform": "mqtt",
                "name": null,
                "icon": null,
                "disabled_by": null,
                "capabilities": {},
                "supported_features": 0,
                "device_class": null,
                "unit_of_measurement": null,
                "original_name": "lamp_machine_house",
                "original_icon": null
            },
```

Mr-Groch, 2021-03-05:
Restart HA and report if then problem disapears. I think that this is the same as #6113 . After each z2m restart, switches exposed as lights in HA will be broken until HA restart.

aureliosaraiva, 2021-03-08:
I will restart everything to test.  but to me this problem seems to be in the z2m.

Mr-Groch, 2021-03-08:
> I will restart everything to test. but to me this problem seems to be in the z2m.

Try restart HA only. Or if you've restarted z2m, then HA

aureliosaraiva, 2021-03-14:
@Mr-Groch resolved! Thanks!

closed 2021-03-14 by aureliosaraiva