Koenkk/zigbee2mqtt #6554: IKEA E1702 control outlet OTA Update causing device to be unresponsive

Morphy99, 2021-03-03:
## What happened

Tried to update my E1704 outlet via OTA and using the ikea_ota_use_test_url: true which seems to complete however the device doesn't respond now:

```
info  2021-03-03 15:03:16: Update of 'Plug Lavalamp' at 100.00%
info  2021-03-03 15:03:16: MQTT publish: topic 'zb2mqtt/Plug Lavalamp', payload '{"last_seen":"2021-03-03T15:01:46+00:00","linkquality":76,"power_on_behavior":"off","state":"OFF","update":{"progress":100,"remaining":23,"state":"updating"},"update_available":false}'
info  2021-03-03 15:03:16: MQTT publish: topic 'zb2mqtt/bridge/log', payload '{"message":"Update of 'Plug Lavalamp' at 100.00%","meta":{"device":"Plug Lavalamp","progress":100,"status":"update_progress"},"type":"ota_update"}'
info  2021-03-03 15:03:39: Finished update of 'Plug Lavalamp'
info  2021-03-03 15:03:39: MQTT publish: topic 'zb2mqtt/Plug Lavalamp', payload '{"last_seen":"2021-03-03T15:01:46+00:00","linkquality":76,"power_on_behavior":"off","state":"OFF","update":{"state":"idle"},"update_available":false}'
info  2021-03-03 15:03:40: MQTT publish: topic 'zb2mqtt/bridge/devices', payload '[{"definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"11":{"bindings":[],"clusters":{"input":["ssIasAce"],"output":["ssIasZone","ssIasWd"]},"configured_reportings":[]},"110":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"12":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"13":{"bindings":[],"clusters":{"input":["genOta"],"output":[]},"configured_reportings":[]},"2":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"242":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"3":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"4":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"47":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"5":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"6":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]},"8":{"bindings":[],"clusters":{"input":[],"output":[]},"configured_reportings":[]}},"friendly_name":"Coordinator", CUT MESSAGE
info  2021-03-03 15:03:40: MQTT publish: topic 'zb2mqtt/bridge/log', payload '{"message":{"id":"Plug Lavalamp","transaction":"3pdim-5"},"meta":{"device":"Plug Lavalamp","from":{"dateCode":"20180528","softwareBuildID":"1.4.020"},"status":"update_succeeded","to":null},"type":"ota_update"}'
info  2021-03-03 15:03:40: MQTT publish: topic 'zb2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20180528","software_build_id":"1.4.020"},"id":"Plug Lavalamp","to":null},"status":"ok","transaction":"3pdim-5"}'
info  2021-03-03 15:03:40: MQTT publish: topic 'zb2mqtt/bridge/info', payload '{"commit":"99274c7","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":3600,"availability_whitelist":[],CUT MESSAGE

```
`error 2021-03-03 15:10:16: Publish 'set' 'state' to 'Plug Lavalamp' failed: 'Error: Command 0x000d6ffffe2b94dc/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 38698 - 1 - 53 - 6 - 11 after 10000ms)'`

I noticed it states status":"update_succeeded","to":null which suggests the issues with the battery powered devices not responding after an update is happening here too. 

Tried rebooting the outlet, restarting Z2M.

Factory resetting the outlet allows the device to join again and after force removing it I can see firmware is now updated.

## What did you expect to happen

The device to function normally after an update

## How to reproduce it (minimal and precise)

See above.

## Debug info
Zigbee2MQTT version: 1.17.1 commit: 99274c7
coordinator type
zStack12
coordinator revision
20201128
frontend version
0.3.23
Adapter hardware:  CC2530

Koenkk, 2021-03-03:
I'm not sure what we can do about this from the Zigbee2MQTT side. Looks like the bulb doesn't restart correctly after the OTA update?

Morphy99, 2021-03-03:
I'm not sure, maybe someone with an IKEA hub can verify if the firmware update works OK with that and if so what it's doing different?  

It's an outlet in this case, the bulbs appear to work OK. Restarting it has no effect, it loses the pairing I think so I have to re-join.

MattWestb, 2021-03-03:
The firmware is out for nearly  one yer and have working for most user without problem:
```
Release Version: 1.12.0
30th March 2020
New features and changes in Accessories:
Control Outlet 2.0.024.
Update Regarding the synchronization Feature.
```
Was you updating from one old ZLL version (1.XX) to Zigbee 3 (2.X) then it can you must deleting it and repairing it if Z2M is not reading the new endpoints and cluster OK.

For resetting use one paperclip and punching the reset button in the small hole for around 10 seconds. Some devices can being tricky getting reseted and i was using touch link for some.

I have hearing of some have problems that the relay is stop working but the zigbee module was OK but that was one hardware fault.

Morphy99, 2021-03-03:
> Was you updating from one old ZLL version (1.XX) to Zigbee 3 (2.X) then it can you must deleting it and repairing it if Z2M is not reading the new endpoints and cluster OK.
> 
Yep it was from v1.4.020 to the version in the test OTA v2.0.024. So it's normal for all devices to need to be re-paired going from v1 to v2? I'm not sure the bulbs do?

MattWestb, 2021-03-03:
Its depends how Z2M  is doing and i dont knowing how its reading the device. Its one complete new device with the same IEEE as the old one. 
I have only updating my outlets with IKEA GW and de(F)CONZ and IKEA is knowing how they is going around it for not braking all systems and  other is not (ZHA dont have that problem then the old is ZLL and the new is ZB3 standard = working out of the box).

Better deleting it and rejoining it for getting it "clean" in your system.

closed 2021-04-10 by github-actions[bot]