Koenkk/zigbee2mqtt #6555: Conbee II with livolo switches

woozar, 2021-03-03:
## What happened
I have paired some livolo switches, using the ConBee II stick. 
I can control the devices from my zigbee2mqtt dashboard but if I press the switch, the new state is not being reflected in the Dashboard. If I use my CC2531 instead, it is working properly. (Sadly I have too many devices for the cc2531)

## What did you expect to happen
I expect the status of the switch in the dashboard to change if i press the switch.

## How to reproduce it (minimal and precise)
1. Use conbee II.
2. Pair with livolo switch
3. Turn on/off the switch on the hardware

## Debug info
Zigbee2MQTT version: 1.18.1-1 (home assistant plugin)
Adapter hardware: Conbee II
Adapter firmware version: 2.09.03 / 23/12/2020
26660700

If you need any additional information (debug logs or whatever) please feel free to ask. I will gladly provide anything.

Koenkk, 2021-03-03:
Can you provide a sniff when pressing the button (you can take it with your CC2531: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

woozar, 2021-03-04:
Wow that reaction was fast. I will do it today after work. :)

woozar, 2021-03-08:
I was finally get the sniffer up and running. I hope the data will be of any help. If you need anything else, please let me know. I'm very happy to support you, where possible.

Koenkk, 2021-03-09:
thanks, can you provide the herdsman debug logging when clicking the button? Please make the log as short as possible, so start right before you click the button and stop immediately after that.

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

woozar, 2021-03-11:
First of all, I had to use a different switch today (same vendor and same model as the last time) because I did the sniffing with a friend and his switch and this time I used my device. My device is having this id: 0x00124b001802d9ef (0x66D9)

I am not sure what or if i did something wrong for capturing the logs. In the configuration of the home assistant addon it set:
* advanced.log_level = debug
* zigbee_herdsman_debug = true

There is a lot of log output in the log tab of the home assistant ui, but it looks like there is no output when pressing the button. I copied everything that happended in the specific time frame. 

`09:40:32.471Z zigbee-herdsman:deconz:adapter EP: 1
09:40:32.472Z zigbee-herdsman:deconz:adapter profile ID: 260
09:40:32.472Z zigbee-herdsman:deconz:adapter device ID: 5
09:40:32.472Z zigbee-herdsman:deconz:adapter input cluster: 0x0
09:40:32.472Z zigbee-herdsman:deconz:adapter input cluster: 0xa
09:40:32.472Z zigbee-herdsman:deconz:adapter input cluster: 0x19
09:40:32.472Z zigbee-herdsman:deconz:adapter output cluster: 0x1
09:40:32.472Z zigbee-herdsman:deconz:adapter output cluster: 0x20
09:40:32.472Z zigbee-herdsman:deconz:adapter output cluster: 0x500
09:40:32.473Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 49
09:40:39.435Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 50
09:40:39.439Z zigbee-herdsman:deconz:frameParser device state: 10100010
09:40:39.440Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
09:40:49.436Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 51
09:40:49.440Z zigbee-herdsman:deconz:frameParser device state: 10100010
09:40:49.441Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
09:40:59.433Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 52
09:40:59.438Z zigbee-herdsman:deconz:frameParser device state: 10100010
09:40:59.438Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1`

Koenkk, 2021-03-11:
@chrishae I need your help here. The issue with these devices is that they send malformed Zigbee messages: 

[image]

It seems that somehow the deconz driver doesn't emit these messages.

For the zstack adapter the messages is emitted as `raw` when parsing it as ZCL fails. https://github.com/Koenkk/zigbee-herdsman/blob/7a9078ce07af6ec8a38cc92e0647c9d40af6daa1/src/adapter/z-stack/adapter/zStackAdapter.ts#L748 . It looks that there is something similar for https://github.com/Koenkk/zigbee-herdsman/blob/7a9078ce07af6ec8a38cc92e0647c9d40af6daa1/src/adapter/deconz/adapter/deconzAdapter.ts#L1135 .

So I think these messages are not send from the deconz adapter to zigbee-herdsman at all, how to debug this? (if that is the case it means the same issue exist when using the conbee with ZHA or deconz rest)

Full message for future reference:
[image]

ChrisHae, 2021-03-12:
@Koenkk could it be the deconz adapter does not respond because the switch sends its commands to EP8?  There was also such an  issue with another switch. That would mean it is an issue within the ConBee2 firmware.

@woozar You have deCONZ installed, don't you? Then could you please open the deCONZ GUI and in the "Network Preferences" (button in the top middle above the network graph) open the tab "Endpoints". Then copy the settings of EP1 to the fields of EP2 with one exception: use 0x08 in the Endpoint field.

One warning: You will lose functionality for green power devices. Best copy and save the old values from the EP2 fields.

After that the switch should also work with z2mqtt. And if that really is the solution for your problem I have to trigger our firmware developer again to fix this in the ConBee2 firmware.

Koenkk, 2021-03-12:
@ChrisHae well spotted, that could indeed be the issue here!

woozar, 2021-03-13:
Awesome support so far. Thank you very much! I will give it a shot tomorrow and let you know! :)

woozar, 2021-04-13:
I am so sorry it took so long, but I finally managed to setup my raspi again to get the stick working and now I changed the settings as described and it works. I believe, that you already understand the issue pretty good, but if you need anything else, please feel free to ask. I will try my best to give quicker response the next time.
@Koenkk @ChrisHae

closed 2021-04-25 by woozar

ChrisHae, 2021-05-17:
@woozar could you test the new firmware that should fix this issue?
https://www.dropbox.com/s/8fh73071knq3y2a/deCONZ_ConBeeII_0x266c0700.bin.GCF?dl=0
here is a manual on how to update the firmware of the ConBee2: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

And when testing just make sure to revert the second Endpoint to the GP Endpoint or just undo the changes if you made any in the Endpoint Config of deCONZ.

woozar, 2021-05-17:
I will test it as soon as I have some time to spare. 

Thank you! :)

martinez1000, 2021-05-18:
Hi,
I can initially confirm this.

I, probably, as well as many others, struggle with setting the correct operation of LIVOLO TI0001.
Yesterday I updated deconz to version 6.9.0 and I have visible conbee II version as 2.11.5 / 5/9/2021.
Thanks to this, communication with livolo works both ways. Endpoint 2 is reset to the original values ​​(excluding those recommended for woozar - that is, a copy of EP1 to EP2).
I am using zigbee2mqtt version 1.18.3-1 and I only have 3 Tuya heads attached to the zigbee.
Unfortunately, so far I have observed a very strange phenomena - automatic switching "on" of the livolo switch (without being turned on by zigbee2mqtt or manually from the switch)
The zigbee2mqtt settings are:
  ext_pan_id_string: '33, 117, 141, 25, 0, 75, 18, 0 '
  pan_id: 6754
  channel: 26

Do you know this problem?
I will confirm in the evening whether it is still present.
Unfortunately, it also happened on CC2531 (I can only add here that the communication works from the beginning on both sides).

ChrisHae, 2021-05-18:
@martinez1000 So if I understand you right the livolo switch is turned on automatically? And it also happens with the CC2531. It sounds like it is another problem best placed in another github issue.

And back to the initial issue: when pressing a button on the switch does it produce any output in z2mqtt that indicates that a button was pressed? (with the provided Conbee2 firmware)

martinez1000, 2021-05-18:
Yes, the livolo turns on automatically without my help - I will open a new issue.

Regarding the two-way action. Unfortunately, update deconz did not resolve the problem of two-way communication. Maunal switching livolo is not seen in zigbee2mqtt. I do not know why, but the changes in deconz (Bringing EP2 back to the previous state) have not been saved. Summing up, I still have to copy the values of EP1 to EP2 (except for the endpoit field, because the switch probably still sends a message to EP8).

woozar, 2021-05-18:
I have had similar issues before zigbee was in place. Some switches (mostly 2 gang 2way) where switching randomly. Are you sure, that they are triggered by zigbee?

martinez1000, 2021-05-18:
I do not have an additional zigbee gate dedicated to livolo. I don't know what you mean that something else triggers livolo, due to the unique connection to livolo (channel 26, ext_pan_id) it seems unrealistic that something else triggers livolo than zigbee. What was the cause for you?

woozar, 2021-05-19:
I don't know what triggered it, but i sent back the switches to the merchant and had them replaced with ones that are not “2 way“ and they don't do it.

woozar, 2021-05-23:
@ChrisHae it seems I accidentally deleted the backup of the old values. 

Can I just reset the stick to defaults somehow?

ChrisHae, 2021-05-23:
Network Settings or Endpoint Settings? When you want to restore the old network settings open the Phoscon App (button in the top right corner of deCONZ) and there you go to Settings/Gateway/ page (top left button in the app). Then alt+left-click on the advanced button and you will come to a page where you can see and restore your last network settings.

If you want to know the default Endpoint settings - they are:

EP 0x01
profile id 0x0104
device id 0x0005
device ver. 0x01
in clusters 0x0000,0x000A,0x0019
out clusters 0x0001,0x0020,0x0500

EP 0xF2
profile id 0xA1E0
device id 0x0064
device ver. 0x01
in clusters 
out clusters 0x0021

ChrisHae, 2021-05-27:
Seems the FW does not fix the issue that commands from certain switches are forwarded to z2mqtt. But the developer is currently working on this. I will notice you when there are some news.

woozar, 2021-08-31:
@ChrisHae any updates?