Koenkk/zigbee2mqtt #6562: Wrong device type (router) for Xiaomi Aqara D1 (no neutral) models QBKG21LM and QBKG22LM

isorin, 2021-03-04:
## What happened
Xiaomi Aqara D1 (no neutral) models QBKG21LM (lumi.switch.b1lacn02) and QBKG22LM (lumi.switch.b2lacn02) appear in the database with device type "Router"

## What did you expect to happen
Device type should be "EndDevice".
The "no-neutral" models of Xiaomi Aqara D1 wall switches are end-devices not routers.

## How to reproduce it (minimal and precise)

## Debug info
{"id":14,"type":"Router","ieeeAddr":"0x00158d0005812857","nwkAddr":44503,"manufId":4151,"manufName":"LUMI","powerSource":"DC Source","modelId":"lumi.switch.b1lacn02","epList":[1,2,3,4,5,6,8],"endpoints":{"1":{"profId":260,"epId":1,"devId":6,"inClusterList":[0,3,1,2,25,10],"outClusterList":[0,10,25],"clusters":{"genBasic":{"attributes":{"65281":{"1":3300,"3":19,"5":29,"8":13850,"10":59418,"100":0,"101":0,"110":0,"111":255,"153":0,"155":0},"manufacturerName":"LUMI","powerSource":4,"zclVersion":0,"stackVersion":2,"hwVersion":54,"dateCode":"03-20-2019"}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[16,6,4,5],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"61440":117440615,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[16,6,4,5],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[18,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":260,"epId":5,"devId":0,"inClusterList":[18,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":260,"epId":6,"devId":0,"inClusterList":[18,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":83,"inClusterList":[12],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":54,"dateCode":"03-20-2019","zclVersion":0,"interviewCompleted":true,"meta":{},"lastSeen":1614848867927}

Koenkk, 2021-03-04:
How do you know that this is a end device and not a router? (asking because the device advertises itself as a router)

isorin, 2021-03-05:
Heh, heh - a Zigbee device that doesn't advertise proper data, something unheard of in the Zigbee world :)

Proofs:

1) It was stated in the announcement:
https://homekitnews.com/2020/02/29/updated-aqara-d1-switches-due-early-march/
"The neutral wire options will also be able to act as Zigbee signal repeaters. The no-neutral models will not be able to do this, however."

2) Stated here as well:
https://zigbee.blakadder.com/Xiaomi_QBKG21LM.html
"Doesn’t work as a router and doesn’t support power meter."

3) I tested them with a power recorder - they don't draw sufficient power (0.15-0.2 W/h) to act as a router / signal repeater.

Koenkk, 2021-03-06:
Thanks, fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-03-06 by Koenkk

isorin, 2021-03-06:
Great, thanks a million!