Koenkk/zigbee2mqtt #6564: New device - Danfoss Icon Master Controller Zigbee

TLongstride, 2021-03-04:
I need help for supporting the Danfoss Icon Master Controller 088U1141

After search, I find the Zigbee API from Danfoss. But I didn't understand how to add it from the explanation here : https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

## Information about the device + link

Danfoss Icon Master Controller 088U1141 : https://store.danfoss.com/fr/fr/Chauffages-et-chauffages-urbains/Plancher-chauffant-hydraulique/R%C3%A9gulateurs-d%27ambiance/R%C3%A9gulateurs-pour-plancher-chauffant%2C-Danfoss-Icon%2C-R%C3%A9gulateur-ma%C3%AEtre%2C-24-0-V%2C-Nombre-de-canaux%3A-10%2C-En-saillie/p/088U1141

API Zigbee Danfoss : https://drive.google.com/file/d/1n5MqOSaBiPYZAXqRsyglrGSwGzOSjZUA/view

TLongstride, 2021-03-06:
I found that Deconz supporte it with the v2.10.0-beta came few days before !

Maybe we can do it for Z2M too ! Thanks

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4273
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4423

nagydavid, 2021-04-13:
I am willing to give a shot to integrate it, but I will need some guidance.

Koenkk, 2021-04-14:
@nagydavid I'm not aware of any other tutorial, but what is missing from the support new devices one?

closed 2021-04-22 by github-actions[bot]