Koenkk/zigbee2mqtt #6570: Unsupported device with Zigbee model 'TS011F'

Kinnoon, 2021-03-05:
I bought TuYa 2 gang wall outlet (switch) (model : TS011F_2_gang_wall)
from support device page
[image]
[image]

I try to connect TuYa 2 gang wall outlet via Zigbee2mqtt but log show that unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_raviyuvk'

log :>
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_raviyuvk'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

I need help with support for new device
I try to do it myself, but I'm beginner in programing and no experiment with scripts T T

Thank you

Koenkk, 2021-03-06:
Added, changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-03-06 by Koenkk

Aaike, 2022-04-10:
@Koenkk I recently bought a power socket with the same model and manufacturer but it is also supposed to support energy monitoring. Can the energy monitoring be added through an external converter ?

Aaike, 2022-04-12:
[](https://dd.lnwfile.com/_/dd/_raw/nx/nv/io.jpg)@Koenkk  here it is

```javascript
{"id":4,"type":"Router","ieeeAddr":"0x50325ffffe405720","nwkAddr":1165,"manufId":4098,"manufName":"_TZ3000_raviyuvk","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,6,1794,2820],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_raviyuvk","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":0}},"seMetering":{"attributes":{"currentSummDelivered":[0,11],"divisor":100,"multiplier":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":232,"rmsCurrent":30,"activePower":3}},"genPowerCfg":{"attributes":{}},"genDeviceTempCfg":{"attributes":{}},"genIdentify":{"attributes":{}},"genGroups":{"attributes":{"nameSupport":0}},"genScenes":{"attributes":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOffSwitchCfg":{"attributes":{}},"genLevelCtrl":{"attributes":{}},"genTime":{"attributes":{}},"genAnalogInput":{"attributes":{}},"genAnalogValue":{"attributes":{}},"genBinaryInput":{"attributes":{}},"genOta":{"attributes":{}},"lightingBallastCfg":{"attributes":{}},"haApplianceIdentification":{"attributes":{}},"haMeterIdentification":{"attributes":{}},"manuSpecificTuya_3":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002390d46a","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b002390d46a","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b002390d46a","endpointID":1},{"cluster":2816,"type":"endpoint","deviceIeeeAddress":"0x00124b002390d46a","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002390d46a","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b002390d46a","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":81,"inClusterList":[4,5,6,1794,2820],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"genPowerCfg":{"attributes":{}},"manuSpecificTuya_2":{"attributes":{}},"manuSpecificTuya_3":{"attributes":{}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":232,"rmsCurrent":30,"activePower":3}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002390d46a","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1887835227},"lastSeen":1649709882285,"defaultSendRequestWhen":"immediate"}

```

---

NOTE:  i have been trying to get this to work in the meantime.
I am very new to all this but here is what i have managed to find found:

```javascript
async function onEventMeasurementPoll(type, data, device, options) {
  const endpoint = device.getEndpoint(1);
  const endpoint2 = device.getEndpoint(2);
  if (type === 'stop') {
    clearInterval(globalStore.getValue(device, 'interval'));
    globalStore.clearValue(device, 'interval');
  } else if (!globalStore.hasValue(device, 'interval')) {
    const seconds =
      options && options.measurement_poll_interval
        ? options.measurement_poll_interval
        : 60;
    if (seconds === -1) return;
    const interval = setInterval(async () => {
      try {
        await endpoint.read('haElectricalMeasurement', [
          'rmsVoltage',
          'rmsCurrent',
          'activePower',
        ]);
        await endpoint2.read('haElectricalMeasurement', [
          'rmsVoltage',
          'rmsCurrent',
          'activePower',
        ]);
      } catch (error) {
        /* Do nothing*/
      }
    }, seconds * 1000);
    globalStore.putValue(device, 'interval', interval);
  }
}

module.exports = [
  {
    fingerprint: [
      { modelID: 'TS011F', manufacturerName: '_TZ3000_mvn6jl7x' },
      { modelID: 'TS011F', manufacturerName: '_TZ3000_raviyuvk' },
      { modelID: 'TS011F', manufacturerName: '_TYZB01_hlla45kx' },
      { modelID: 'TS011F', manufacturerName: '_TZ3000_92qd4sqa' },
      { modelID: 'TS011F', manufacturerName: '_TZ3000_zwaadvus' },
    ],
    model: 'TS011F_2_gang_wall',
    vendor: 'TuYa',
    description: '2 gang wall outlet (with power monitoring)',
    ota: ota.zigbeeOTA,
    toZigbee: extend.switch().toZigbee.concat([
      tz.moes_power_on_behavior,
      tz.tuya_backlight_mode,
      tz.tuya_switch_power_outage_memory,
      tz.ts011f_plug_indicator_mode,
      tz.ts011f_plug_child_mode,
      tz.tuya_data_point_test,
    ]),
    fromZigbee: extend.switch().fromZigbee.concat([
      fz.moes_power_on_behavior,
      fz.tuya_backlight_mode,
      fz.metering,
[... 142 lines cut ...]

and here are readings when the fan is connected to it

```javascript
{
    "child_lock": "UNLOCK",
    "current": 0.19,
    "energy": 0.69,
    "indicator_mode": "off",
    "linkquality": 90,
    "power": 43,
    "power_outage_memory": "on",
    "state": "ON",
    "voltage": 231,
    "update": {
        "state": null
    },
    "update_available": null
}
```

The readings from the smart plug are correct.

So... all in all i think when polling the numbers the wall socket just returns wrong readings... ?
Or is there still hope for the wall socket ?

Koenkk, 2022-04-13:
@Aaike it looks to me the socket doesn't support reading for the sockets separately, the reading of endpoint 1 seems to be the combination of both (?)

chukaonline, 2023-01-01:
This converter worked for me until update 1.28.3-1. I can no longer read power, current, and voltage details even though I can see in the state that is available but not exposed. Any fix for the converter will be helpful

`const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {
  ...require('zigbee-herdsman-converters/converters/fromZigbee'),
  legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee,
};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const extend = require('zigbee-herdsman-converters/lib/extend');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

async function onEventMeasurementPoll(type, data, device, options) {
  const endpoint = device.getEndpoint(1);
  const endpoint2 = device.getEndpoint(2);
  if (type === 'stop') {
    clearInterval(globalStore.getValue(device, 'interval'));
    globalStore.clearValue(device, 'interval');
  } else if (!globalStore.hasValue(device, 'interval')) {
    const seconds =
      options && options.measurement_poll_interval
        ? options.measurement_poll_interval
        : 60;
    if (seconds === -1) return;
    const interval = setInterval(async () => {
      try {
        await endpoint.read('haElectricalMeasurement', [
          'rmsVoltage',
          'rmsCurrent',
          'activePower',
        ]);
        await endpoint2.read('haElectricalMeasurement', [
          'rmsVoltage',
          'rmsCurrent',
          'activePower',
        ]);
      } catch (error) {
        /* Do nothing*/
      }
    }, seconds * 1000);
    globalStore.putValue(device, 'interval', interval);
  }
}

module.exports = [
  {
    fingerprint: [
      { modelID: 'TS011F', manufacturerName: '_TZ3000_mvn6jl7x' },
      { modelID: 'TS011F', manufacturerName: '_TZ3000_raviyuvk' },
      { modelID: 'TS011F', manufacturerName: '_TYZB01_hlla45kx' },
      { modelID: 'TS011F', manufacturerName: '_TZ3000_92qd4sqa' },
      { modelID: 'TS011F', manufacturerName: '_TZ3000_zwaadvus' },
    ],
    model: 'TS011F_2_gang_wall',
    vendor: 'TuYa',
    description: '2 gang wall outlet (with power monitoring)',
    ota: ota.zigbeeOTA,
    toZigbee: extend.switch().toZigbee.concat([
      tz.moes_power_on_behavior,
      tz.tuya_backlight_mode,
      tz.tuya_switch_power_outage_memory,
      tz.ts011f_plug_indicator_mode,
      tz.ts011f_plug_child_mode,
      tz.tuya_data_point_test,
    ]),
    fromZigbee: extend.switch().fromZigbee.concat([
      fz.moes_power_on_behavior,
      fz.tuya_backlight_mode,
      fz.metering,
      fz.ignore_basic_report,
      fz.tuya_data_point_dump,
      fz.electrical_measurement,
      fz.tuya_switch_power_outage_memory,
      fz.ts011f_plug_indicator_mode,
      fz.ts011f_plug_child_mode,
    ]),
    configure: async (device, coordinatorEndpoint, logger) => {
      const endpoint = device.getEndpoint(1);
      await reporting.bind(endpoint, coordinatorEndpoint, [
        'genOnOff',
        'haElectricalMeasurement',
        'seMetering',
      ]);

      // NOTE: Tuya smart plugs manufactured after Q4 2021 do not support reporting
      //   await reporting.rmsVoltage(endpoint, { change: 5 });
      //   await reporting.rmsCurrentPhB(endpoint, { change: 50 });
      //   await reporting.activePower(endpoint, { change: 10 });
      //   await reporting.currentSummDelivered(endpoint);

      endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
        acCurrentDivisor: 1000,
        acCurrentMultiplier: 1,
      });
      endpoint.saveClusterAttributeKeyValue('seMetering', {
        divisor: 100,
        multiplier: 1,
      });
      device.save();
    },
    options: [exposes.options.measurement_poll_interval()],
    onEvent: onEventMeasurementPoll,
    exposes: [
      e.switch().withEndpoint('l1'),
      e.switch().withEndpoint('l2'),
      e.power().withEndpoint('l1'),
      e.power().withEndpoint('l2'),    
      e.current().withEndpoint('l1'),
      e.current().withEndpoint('l2'),
      e.voltage().withEndpoint('l1'),
      e.voltage().withEndpoint('l2'),      
        exposes.enum('power_on_behavior', ea.ALL, ['on', 'off', 'previous']),
        exposes.enum('backlight_mode', ea.ALL, ['LOW', 'MEDIUM', 'HIGH']),

      // Enabling these result in home assisant errors
      // () => new Numeric('power_l1', access.STATE).withUnit('W').withDescription('Instantaneous measured power L1'),
      // () => new Numeric('voltage_l1', access.STATE).withUnit('V').withDescription('Measured electrical potential value L1'),
      // () => new Numeric('current_l1', access.STATE).withUnit('C').withDescription('Instantaneous measured electrical current L1'),
      //   () => new Numeric('power_l2', access.STATE).withUnit('W').withDescription('Instantaneous measured power L2'),
      //   () => new Numeric('voltage_l2', access.STATE).withUnit('V').withDescription('Measured electrical potential value L2'),
      //   () => new Numeric('current_l2', access.STATE).withUnit('C').withDescription('Instantaneous measured electrical current L2'),

      e.energy(),
      exposes
        .enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
        .withDescription('Recover state after power outage'),
      exposes
        .enum('indicator_mode', ea.ALL, ['off', 'off/on', 'on/off', 'on'])
        .withDescription('Plug LED indicator mode'),
      e.child_lock(),
    ],
    whiteLabel: [{ vendor: 'ClickSmart+', model: 'CMA30036' }],
    endpoint: (device) => {
      return { l1: 1, l2: 2 };
    },
    meta: { multiEndpoint: true },
  },
];`