Koenkk/zigbee2mqtt #6578: MEM_ERROR after using GreenPower Switch (Philips Hue)

s1rd4v3, 2021-03-05:
## What happened
Zigbee2mqtt stops working after I receive the error `UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')` and having paired a GreenPower Zigbee switch and clicking sometimes on it.

## What did you expect to happen
Zigbee2mqtt doesn't hang after using a GreenPower switch.

## How to reproduce it (minimal and precise)
1. I added a Philips Hue Light Bulb
2. I paired a Friends of Hue switch  (via https://www.zigbee2mqtt.io/devices/GreenPower_On_Off_Switch.html#pairing-friends-of-hue-switches-enocean-ptm-216z-module-based) which runs over the GreenPower Zigbee. Pairing worked as expected
3. the first few clicks are working as expected and i can see the clicks in the MQTT logs and in my HASS instance
4. After several clicks on the switch, i get the above mentioned error and I am unable to start zigbee2mqtt again. I then have to change my network key in order to have it up and running again (of course i then also need to rebind all the devices each time). Even the backup, which was made before adding the GreenPower switch, can't be used to have it up again.
5. trying to restart it manually afterwards. the error is `Error: SREQ '--> ZDO - activeEpReq - {"dstaddr":0,"nwkaddrofinterest":0}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')`

## Debug info
Zigbee2MQTT version: v1.18.1
Adapter hardware: CC2538 HAT
Adapter firmware version: 20200211

Koenkk, 2021-03-06:
Looks like a firmware problem, @0x3EC  can you check this?

0x3EC, 2021-03-06:
@s1rd4v3 
_Adapter firmware version: 20200211_
Can you upgrade to 20201010 and check again?

s1rd4v3, 2021-03-07:
Thanks for investigating! Today i've upgraded the firmware to 20201010 and still get the exact same error. Full error log can be seen here: https://pastebin.com/64RRt0Lv

s1rd4v3, 2021-03-09:
Here you can find the corresponding [coordinator_backup.json](https://pastebin.com/Ghdrb5KA) file.

Secondly you can find here another [coordinator_backup.json](https://pastebin.com/yeL15UPt) file from a backup with which i had the same error as reported above but has `81` as __MaxChildren__ and __MaxRouters__. 
I've also tested the same CC2538 HAT coordinator on another Pi (Zero instead of 4) just now with a fresh zigbee2mqtt setup but I got the same error as mentioned above after using the GreenPower Switch. (Had `81` MaxChildren and MaxRouters). So not sure if this is the main issue in my case.

For testing purpose I've added a CC2531 to my Pi 4. With this adapter, my setup with the GreenPower Switch is working perfectly fine.

0x3EC, 2021-03-10:
@s1rd4v3 
Unfortunately, there may be a situation when data in the backup file does not correspond to data saved in NV_MEM. This can happen when migrating from a different stick or from a different firmware.
To eliminate this situation, follow these steps. After that, I will be sure that your stick is configured correctly and that any errors you find are actually problems you found.  

Do this:

```
1. Stop z2m
2. Delete z2m backups
3. Clean the NV_MEM script https://github.com/Koenkk/zigbee2mqtt/blob/master/scripts/zStackEraseAllNvMem.js
4. Cold restart stick 2538
5. Start z2m
6. Set up a new network, add devices.
7. Stop z2m and check that the NIB table is 116 bytes. Check if there are any problems

Use the firmware for 2538 20201010.
```

s1rd4v3, 2021-03-11:
OK, did all of your steps and here's the [log](https://pastebin.com/arDcjafX) and the [coordinator_backup.json](https://pastebin.com/signup). Outcome is still the same: zigbee2mqtt freezes and won't let me control or restart the instance anymore after some clicks on the GreenPower switch with the error I've mentioned in the initial post. Let me know if I can do something else to identify the issue.

s1rd4v3, 2021-03-25:
Hey @0x3EC. Any news on this issue?

0x3EC, 2021-03-29:
> Hey @0x3EC. Any news on this issue?
Hi, @s1rd4v3 
In my spare time, I work on your problem. The difficulty is that I could not repeat this malfunction.

s1rd4v3, 2021-03-31:
Hmm ok. Whenever i can provide you with more informations that might help you, just let me know... Thanks a lot for your effort!

closed 2021-05-09 by github-actions[bot]

Timminater, 2021-05-25:
Same issue here. After a few button presses Zigbee2MQTT/CC2538 seems to crash.

Zigbee2MQTT version: 1.18.1 commit: 1.18.1
Coordinator type: zStack30x
Coordinator revision: 20201010
Frontend version: 0.3.76

amazing4u, 2021-07-03:
Same problem here using IOBROKER + ZIGBEE 1.56 Adapter with CC2538 + CC2592 MODKAMRU