Koenkk/zigbee2mqtt #6579: Firmware version not updated after device OTA update (Legrand)

jd1900, 2021-03-05:
## What happened
I'm trying to add Legrand devices OTA support.
However, after upgrading a device, the "firmware version" showed in the OTA tab in the GUI is not updated.
The device is properly updated, as when I try to check for new firmware, it returns the proper version, but even after a few reboots and turning off phisically the devices, they don't change at all.

## What did you expect to happen
Change to update the new firmware version.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: CC26X2R1

Koenkk, 2021-03-06:
> The device is properly updated,

How you do you know this?

jd1900, 2021-03-06:
> > The device is properly updated,
> 
> How you do you know this?

Because it has new functionallity from the update. 
And when changed to the original gateway, it says it is updated.

Koenkk, 2021-03-06:
After removing and repairing, does the new version show up? If not they probably didn't updated the firmware version in the fw. (which can happen since this is not used to determine if a newer fw version is available)

jd1900, 2021-03-07:
Yes, it does.

But in the test I have repaired in the official gateway, so it was not a perfect test.

closed 2021-04-15 by github-actions[bot]