Koenkk/zigbee2mqtt #6581: Please add support for LEDVANCE AC10402

fkh-bims, 2021-03-05:
I use 

{
        zigbeeModel: ['Gardenpole Mini RGBW Z3'],
        model: 'AC10402',
        vendor: 'LEDVANCE',
        description: 'Smart+ mini gardenpole RGBW',
        extend: preset.ledvance.light_onoff_brightness_colortemp_color(),
        ota: ota.ledvance,
    },

in ...\zigbee-herdsman-converters\devices.js

## Information about the device + link
[https://www.ledvance.co.uk/tech/products/index.jsp?productId=ZMP_3573690&classificationId=GPS01_3576325](url)

[https://dammedia.ledvance.info/media/resource/hires/asset-11766924/SMART-GARDENPOLE-EXTENSION_SMART-GARDENPOLE-MINI-EXTENSION.pdf](url)

## data/database.db entry of the device
{"id":26,"type":"Router","ieeeAddr":"0xf0d1b8000014c841","nwkAddr":14961,"manufId":4489,"manufName":"LEDVANCE","powerSource":"Mains (single phase)","modelId":"Gardenpole Mini RGBW Z3","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,2821,4096,64527],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Gardenpole Mini RGBW Z3","manufacturerName":"LEDVANCE","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"Feb 27 2019\u0000\u0000\u0000\u0000\u0000","swBuildId":"00103103"}},"genLevelCtrl":{"attributes":{"currentLevel":1}},"genOnOff":{"attributes":{"onOff":0}},"lightingColorCtrl":{"attributes":{"currentHue":254,"currentSaturation":254,"currentX":11947,"currentY":42008,"colorTemperature":500}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff048487","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00212effff048487","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00212effff048487","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":300,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":3,"maxRepIntval":300,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"Feb 27 2019\u0000\u0000\u0000\u0000\u0000","swBuildId":"00103103","zclVersion":2,"interviewCompleted":true,"meta":{"reporting":1,"configured":2},"lastSeen":1614961144384}

sjorge, 2021-03-05:
I made a PR, but we're missing some info needed for the color temp expose :(

Can you add 
```
   {
        zigbeeModel: ['Gardenpole Mini RGBW Z3'],
        model: '4058075208353',
        vendor: 'LEDVANCE',
        description: 'SMART+ Gardenpole Multicolour',
        extend: preset.ledvance.light_onoff_brightness_colortemp_color(),
        ota: ota.ledvance,
    },
```

To your devices.js and repair the bulb? And then post the new databases.db entry, then it should have the info.

fkh-bims, 2021-03-06:
I hope I did the right thing. (add not replace)

`{"id":26,"type":"Router","ieeeAddr":"0xf0d1b8000014c841","nwkAddr":14961,"manufId":4489,"manufName":"LEDVANCE","powerSource":"Mains (single phase)","modelId":"Gardenpole Mini RGBW Z3","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,2821,4096,64527],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Gardenpole Mini RGBW Z3","manufacturerName":"LEDVANCE","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"Feb 27 2019\u0000\u0000\u0000\u0000\u0000","swBuildId":"00103103"}},"genLevelCtrl":{"attributes":{"currentLevel":1}},"genOnOff":{"attributes":{"onOff":0}},"lightingColorCtrl":{"attributes":{"currentHue":254,"currentSaturation":254,"currentX":11947,"currentY":42008,"colorTemperature":500,"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":526}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff048487","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00212effff048487","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00212effff048487","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":300,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":3,"maxRepIntval":300,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"Feb 27 2019\u0000\u0000\u0000\u0000\u0000","swBuildId":"00103103","zclVersion":2,"interviewCompleted":true,"meta":{"reporting":1,"configured":2},"lastSeen":1615014164974}`

sjorge, 2021-03-06:
Yep you did ! ```"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":526``` is what I needed, I'll update the PR.

closed 2021-04-02 by fkh-bims