Koenkk/zigbee2mqtt #6585: xy color in light state when changing color_temp

Mr-Groch, 2021-03-05:
## What happened
When I publish "color_temp" property in "set" message, then xy color will be added to light state. This will be exposed to Home Assistant. This will brake hs color only lights in Home Assistant (like OSRAM AC03645) - when xy color appers in state, then when I try to change color in Home assistant - it will change physically, but displayed color in HA will not be updated ever

## What did you expect to happen
There should be no xy color property added when changing only color_temp

## How to reproduce it (minimal and precise)
Publish for example {"color_temp":350} to set topic of light and observe in light state that xy color will appear

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: CC26X2R1 zzh!
Adapter firmware version: 20210120

sjorge, 2021-03-05:
I think once I fix both TODO's here https://github.com/Koenkk/zigbee-herdsman-converters/pull/2283#issuecomment-785212562 that should be fixed right?

But we've never settled weather stripping the non-active color_mode is desired behavior, I personally think that would make sense.

Mr-Groch, 2021-04-05:
Nie stałe, problem with adding color mode that bulb doesn't support still exists (like adding xy color for only hs color bulb when changing only ct)

AChiTenshi, 2021-04-23:
Any word on this?  I am currently experiencing the issue in #6672 where the color is not getting updated in the state after reconnecting the bulb.  Brightness updates and I get the message: Failed to read state of 'Bulb 3' after reconnect

Neither color_temp nor color get updated.  And I'm not sure if the color_mode should but its "xy"

closed 2021-05-31 by github-actions[bot]