Koenkk/zigbee2mqtt #6586: hue_move with negative values doesn't work

Mr-Groch, 2021-03-05:
## What happened
I've buyed recently IKEA E1524/E1810 remote, and wanted to set hue_move in both directions when holding left/right buttons. I've noticed that hue_move with positive values works, but with negative - nothing happens.

## What did you expect to happen
hue_move should work both ways with positive and negative values

## How to reproduce it (minimal and precise)
Publish for example {"hue_move": -40} to light set topic

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: CC26X2R1 zzh!
Adapter firmware version: 20210120

Koenkk, 2021-03-06:
What bulb?

Mr-Groch, 2021-03-06:
> What bulb?

[Osram LIGHTIFY LED CLA60 E27 RGBW (AC03645)](https://www.zigbee2mqtt.io/devices/AC03645.html#osram-ac03645)

Koenkk, 2021-03-06:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/6587#issuecomment-791959436

closed 2021-03-06 by Koenkk