Koenkk/zigbee2mqtt #6587: color_temp_move and color_temp_step with positive values doesn't work

Mr-Groch, 2021-03-05:
## What happened
I've buyed recently IKEA E1524/E1810 remote, and wanted to set color_temp_move or color_temp_step in both directions when pressing left/right buttons. I've noticed that color_temp_move or color_temp_step with negative values works, but with positive - nothing happens.

## What did you expect to happen
color_temp_move and color_temp_step  should work both ways with positive and negative values

## How to reproduce it (minimal and precise)
Publish for example {"color_temp_move ": 40} or  {"color_temp_step ": 40} to light set topic and observe nothing happens

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: CC26X2R1 zzh!
Adapter firmware version: 20210120

Koenkk, 2021-03-06:
What bulb?

Mr-Groch, 2021-03-06:
> What bulb?

[Osram LIGHTIFY LED CLA60 E27 RGBW (AC03645)](https://www.zigbee2mqtt.io/devices/AC03645.html#osram-ac03645)

Koenkk, 2021-03-06:
- I've just checked {"hue_move": -40} with a Hue bulb and it works correctly, can you try with another bulb, maybe the OSRAM firmware doesn't support it
- Color temp move and stepping also works with the Hue bulb. But note that there is a small typo in your examples`{"color_temp_move ": 40}` -> `{"color_temp_move": 40}` (trailing space)

Mr-Groch, 2021-03-06:
I've checked other light - [OSRAM Led strip ](https://www.zigbee2mqtt.io/devices/4052899926110.html#osram-4052899926110) and {"hue_move": -40} works, {"color_temp_move": 40} also works...

So yes - it must be something with that bulb... It's not typo mistake in my mqtt message - this space appeared while pasting into issue...

I've checked again my bulb - still color_temp_move or color_temp_step  with positive values doesn't work... The same messages worked for LED strip...

It can be software hacked somewhat to make it work?

Koenkk, 2021-03-07:
It's a special Zigbee command, I recommend to make some scripting to just set the color temp instead.

Mr-Groch, 2021-03-07:
> It's a special Zigbee command, I recommend to make some scripting to just set the color temp instead.

I will, thanks for help

closed 2021-03-07 by Mr-Groch