Koenkk/zigbee2mqtt #6588: zigbee2mqtt stopped working after updated to HA 2021.3.1

waternsw-chrischen, 2021-03-05:
## What happened
Previously connected devices stopped response and cannot control. Temperature and binary sensors do not update.
The update also brought the supervisor version 2021.03.4. I cannot downgrade the version of supervisor to 2021.02.x
## What did you expect to happen
Connected devices reporting their states and being controlled through mqtt 
## How to reproduce it (minimal and precise)
Since the update, it does not work anymore.
## Debug info
Zigbee2MQTT version:1.17.1-1

debug info only listed all devices and the previous mqtt massage. No further information posted.

ced1442, 2021-03-05:
Same for me, but I'm not sure it come from zigbee2mqtt because logs are ok. The communication with my devices is ok.

m0wlheld, 2021-03-06:
HA Container 2021.3.1 works with 1.18.1. This is not a Zigbee2mqtt issue.

juslex, 2021-03-07:
Mosquitto 5.1.1 have a bug. Check it.

christoph-luebbe, 2021-03-08:
Yes this is the same with Zwave2mqtt and the newer ZwaveJS2mqtt. They all suffer from mosquitto 5.1.1

closed 2021-04-16 by github-actions[bot]