Koenkk/zigbee2mqtt #6594: TS0601_thermostat unknown

jeppper, 2021-03-06:
## What happened
I paired a new TS0601_thermostat but it says Manufacturer "UNKNOWN"

[image]

On the package it says Model HY368. Could that be the problem?

## What did you expect to happen
It to pair successfully
## How to reproduce it (minimal and precise)
Re pair
## Debug info
Zigbee2MQTT version: 1.17.1
Adapter hardware: CC2538
Adapter firmware version:

insipiens, 2021-03-06:
Did it give you the Id (assuming you’re running logs at debug level). You could post the log snippet

jeppper, 2021-03-06:
I got this from the .db. Is the ID included here or must I enable debug logging?
{"id":81,"type":"EndDevice","ieeeAddr":"0x847127fffe1f2f59","nwkAddr":35222,"manufId":4098,"manufName":"_TZE200_ywdxldoj","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ywdxldoj","powerSource":3,"zclVersion":3,"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1615026918474}

insipiens, 2021-03-06:
Looks like this already is being looked at, see here. Maybe they have put in a PR?

https://github.com/Koenkk/zigbee-herdsman-converters/issues/2184#issue-797672053

jeppper, 2021-03-06:
Okay but must I do something or will there be a solution you think?
The other thread is a few monts old

insipiens, 2021-03-06:
Best way will be to try adding it yourself. 

As it probably uses the same converters you might just need to test adding it to devices.js

have a read of this page https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

I’d copy the entry for the existing hy368 and adjust accordingly

jeppper, 2021-03-06:
I am not familiar with adding new devices but I will have a look at it.

skion, 2021-03-13:
@jeppper FWIW I have a TS0601_thermostat which is recognised in version 1.18.1, try upgrading?

jeppper, 2021-03-13:
Thanks. You know if yours is a HY368 model?

jeppper, 2021-03-17:
After updating to 1.18.1 mine is also recognized now and I am able to control it.

skion, 2021-03-17:
@jeppper I do wonder if yours are also very bad at periodically reporting temperatures, and whether that is a bug somewhere... LEt me know if you see the same!

jeppper, 2021-03-17:
@skion I don´t plan on getting the temperature from this. I have a seperate sensor (Xiaomi) in the room for this. I just want to control sent setpoints and shut it off when a window contact is open.

insipiens, 2021-03-17:
Glad to hear its now working!

@skion  As to temperature gathering, depends what is meant by bad. Most sensors report on a change only

Here's a graph of on of mine radiator devices, a Moes HY369:

[image]
The blue line is the internal temperature sensor in the device. Red is its set target, and yellow is a seperate Xiaomi device in the same room.

insipiens, 2021-03-17:
There's a discussion thread here which a few of us use to discuss these Tuya type radiator valves, their limitations and ideas around how to use them

https://github.com/Koenkk/zigbee2mqtt/discussions/5748

jeppper, 2021-03-18:
@insipiens Thanks for the link to the other thread.
Do you know if it is possible to use an external room temp. sensor (Xiaomi) instead of the internal, but still use the internal PI control?

insipiens, 2021-03-18:
Yes, of course. There are probably a few ways of doing it. You should post in the discussion for more ideas

Just be aware doing so will increase your heating bills - I only do this in my office because I'm in here 12 hours a day ;-)

The automation runs every 5 minutes and sets the calibration = "external sensor" - "your target"

for example then target = 15, ext sensor =13, set calibration to -2 so radiator thinks its 2 degrees colder, repeat every 5 minutes to adjust calibration again accordingly.

In this automation office _sensor is the external xiaomi sensor and the djradtemp (code snippet below automation) is an mqtt sensor. The service publishes an MQTT message back to the radiator setting a new calibration.

```
- id: '999999999999999999'
  alias: calibrate_djradiators
  description: 'calibrate Radiator'
  trigger:
    platform: time_pattern
    minutes: "/5"
  condition:
    condition: template
    value_template: "{{ (states('sensor.office_sensor_temperature')|float - states('sensor.djradtemp')|float)|round(1 , 'half') != 0 }}"
  action:
  - service: mqtt.publish
    data_template:
      topic: zigbee2mqtt/Radiator_DJ/set/local_temperature_calibration
      payload_template: "{{ ( states('sensor.office_sensor_temperature')|float -                                                                            $
  mode: restart

```

Read internal radiator sensor from MQTT
```

  - platform: mqtt
    name: djradtemp
    state_topic: "zigbee2mqtt/Radiator_DJ"
    unit_of_measurement: '°C'
    value_template: '{{ value_json.local_temperature }}'

```

`

jeppper, 2021-03-19:
@insipiens Okay thanks :-)

skion, 2021-03-20:
> @skion As to temperature gathering, depends what is meant by bad. Most sensors report on a change only

@insipiens Here's what I'm seeing:

[image]

- The bottom graph is a working SEA801 TRV which reports temperature every N mins
- The top  graph is a TS0601_thermostat TRV which only reports temperature only on state changes

Is this expected behaviour?

closed 2021-04-28 by github-actions[bot]

kds69, 2022-01-28:
Strangely for my HY369 no _select.xxxx_force_ entity has been created in Homeassistant, whereas the attribute is reported.
This is fine with my HY368 types they all show _select.xxxx_force_.

Any idea at what level things went wrong during discovery?  HA or zigbeemqtt??