Koenkk/zigbee2mqtt #6600: Please add support for 'TS0215A' and manufacturer name '_TZ3000_fsiepnrh'

steejvan, 2021-03-06:
Hello Koen,

Could you please add support for 'TS0215A' and manufacturer name '_TZ3000_fsiepnrh'. 

It looks like the same device as the Woox one, except this one is from Nedis (ZBRC10WT)
I've tried to add the device to devices.js as your support page describes (copy from the Woox version) but for some reason it's still displayed as unsupported.

I'm running the latest version on Unbuntu under Docker.
but I'm not sure I've written back the devices.js back to docker (I've edited it under SSH though)

--
devices.js:

    {
        fingerprint: [{modelID: 'TS0215A', manufacturerName: '_TZ3000_p6ju8myv'}],
        model: 'TS0215A_remote',
        vendor: 'TuYa',
        description: 'Security remote control',
        fromZigbee: [fz.command_arm, fz.command_emergency, fz.battery],
        exposes: [e.battery(), e.action(['disarm', 'arm_day_zones', 'arm_night_zones', 'arm_all_zones', 'invalid_code', 'emergenc>
        toZigbee: [],
        whiteLabel: [{vendor: 'Woox', model: 'R7054'}],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genTime', 'genBasic', 'ssIasAce', 'ssIasZone']);
        },
    },
 {
        fingerprint: [{modelID: 'TS0215A', manufacturerName: '_TZ3000_fsiepnrh'}],
        model: 'TS0215A_remote',
        vendor: 'TuYa',
        description: '4 button remote control',
        fromZigbee: [fz.command_arm, fz.command_emergency, fz.battery],
        exposes: [e.battery(), e.action(['1_single', '1_double', '1_hold', '2_single', '2_double', '2_hold',
            '3_single', '3_double', '3_hold', '4_single', '4_double', '4_hold'])],
        toZigbee: [],
        whiteLabel: [{vendor: 'Nedis', model: 'ZBRC10WT'}],
        },
    },

## Information about the device + link
https://nedis.com/en-us/product/smart-living/smart-home/lighting/550735173/smart-remote-contoller-zigbee-4-buttons-battery-included-white
..

## data/database.db entry of the device
..

Koenkk, 2021-03-06:
Added, 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-03-06 by Koenkk

steejvan, 2021-03-06:
Thanks for this quick response!!

branislavbudzak, 2022-05-13:
Hi, I switched to dev branch, deleted this device, added it again and even reconfigure the device, but it still shows up as TS0215A_remote even though it is Nedis ZBRC10WT . What should I do next to make it work? @Koenkk @steejvan

Thank you for your help